MichaIng/DietPi #2983: Homeassistant failing hash check

JJGO, 2019-07-12:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.25.3
- Distro version | stretch
- Kernel version | 4.19.57-v7+
- SBC device | RPi 3b+

## Details

I tried to install homeassistant using `dietpi-software`. The service does not crash but the software is not functional. 

Checking `/mnt/dietpi_userdata/homeassistant/home-assistant.log` returns the following 

    2019-07-11 21:20:07 WARNING (Recorder) [homeassistant.components.recorder] Ended unfinished session (id=3 from 2019-07-11 23:24:28.760002)
    2019-07-11 21:30:35 ERROR (SyncWorker_13) [homeassistant.util.package] Unable to install package home-assistant-frontend==20190626.0: ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
        home-assistant-frontend==20190626.0 from https://www.piwheels.org/simple/home-assistant-frontend/home_assistant_frontend-20190626.0-py3-none-any.whl#sha256=42e4d7d91d8d0f1217f251e15b257857e356a994b997c6e3d2a0d360d2e6258d (from -c /home/homeassistant/.pyenv/versions/3.6.3/envs/homeassistant-3.6.3/lib/python3.6/site-packages/homeassistant/package_constraints.txt (line 13)):
            Expected sha256 42e4d7d91d8d0f1217f251e15b257857e356a994b997c6e3d2a0d360d2e6258d
                 Got        6b03907a548d4f239a99b2172397e05eb2d40a8286b41a73824db400879c029a
    2019-07-11 21:30:35 ERROR (MainThread) [homeassistant.requirements] Not initializing frontend because could not install requirement home-assistant-frontend==20190626.0
    2019-07-11 21:30:35 ERROR (MainThread) [homeassistant.setup] Setup failed for frontend: Could not install all requirements.
    2019-07-11 21:30:35 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of logbook. Setup failed for dependencies: frontend
    2019-07-11 21:30:35 ERROR (MainThread) [homeassistant.setup] Setup failed for logbook: Could not set up all dependencies.
    2019-07-11 21:30:35 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of map. Setup failed for dependencies: frontend
    2019-07-11 21:30:35 ERROR (MainThread) [homeassistant.setup] Setup failed for map: Could not set up all dependencies.
    2019-07-11 21:30:35 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of default_config. Setup failed for dependencies: frontend, logbook, map
    2019-07-11 21:30:35 ERROR (MainThread) [homeassistant.setup] Setup failed for default_config: Could not set up all dependencies.

I looked at some other issues and it might be related to the fact that it uses `pyenv`. I have an already exisiting installation of `pyenv` and a 3.7 env. However, this is not under the `dietpi`user but under the `jjgo` user.

MichaIng, 2019-07-13:
@JJGO 
Many thanks for your report.

The idea behind pyenv is that it creates an individual Python environment for each instance, so there should not be any conflicts.

Let me do a test install to see if this is a general issue with the latest HA version `20190626.0`.

JJGO, 2019-07-17:
I got it working.

Plain reinstall did  help partially but the device still had issues related to database corruption.

I was able to fix everything by doing the following

    rm -rf /mnt/dietpi_userdata/homeassistant
    rm -rf /home/homeassistant/pyenv
    dietpi-software reinstall 157

I had no previous configuration, that is why I deleted the `dietpi_userdata` folder anyways.

closed 2019-07-17 by JJGO

MichaIng, 2019-07-17:
@JJGO 
Okay great to hear.