MichaIng/DietPi #2984: DietPi-Process_Tool | Manually added service is removed if you exclude it from DietPi-Services control

AtkLordOverAll, 2019-07-13:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.25.3
- Distro version | Stretch
- Kernel version | 4.19.42-v7+
- SBC device | RPi 1b
- Power supply used | Official
- SDcard used | EG: SanDisk Ultra

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi?
Installation is about as fresh as it gets, only Pi-hole and Unbound installed/running.

#### Steps to reproduce
(I'm going to use `unbound` as my example)
1. Add `unbound` to services list on DietPi-Process_Tool
2. Go to `unbounds`'s single service options, and try to set `Service Control -> Include/Exclude` to excluded
3. Get kicked back to DietPi-Process_Tool, and `unbound` is no longer in the services list

#### Expected behaviour
Setting changed from "included" to "excluded"

#### Actual behaviour
Get booted back to main menu, process' entry has been removed, no settings have been applied

#### Extra details
Have only tested this with unbound as it's the only service that's not a part of DietPi-Software that I use
This is my first bug report; sorry if I fudged something up

MichaIng, 2019-07-14:
@AtkLordOverAll 
Many thanks for your report.

This is actually the expected behaviour:
- Including services, that are not known to DietPi-Services by default, and excluding services that are handled by default, use the same include/exclude file: `/DietPi/dietpi/.dietpi-services_include_exclude`
- It does IMO not make much sense to add a custom service there and then add a second entry which excludes it, while removing both entries has the same effect?
- But while writing I think I get the intention: Allow to set CPU/IO scheduling settings for services while not having them controlled by DietPi-Services start/stop during updates/installs etc? This makes sense for `unbound` which of course cannot be stopped during maintenance tasks since this would break name resolving, but you might want to change its scheduling settings.

Basically it makes sense to have two lists separated, one for automated service control (start/stop/restart etc.) and one for applying scheduling options. I am thinking about the following:
- By default having the list like now, showing only services that are handled by default and those which were manually included.
- Replacing the `Add` option with something like: `Full list : Show all services available on this system`
- Selecting the above will go through all systemd+sysvinit service locations to create a complete list. Selecting any of those services then allows settings/options as before, scheduling settings and also including the service for automated start/stop control during updates/installs, which will also add it to the initial list.
- The list will be quite long, since that is a huge amount of systemd services available by default. Not sure if we should filter it a bid, e.g. remove services that are already part of the initial list (known by DietPi or included by user), and/or remove all inactive services? Or perhaps even more elegant, having a filter menu to allow selecting filter for masked and/or disabled and/or inactive services. Even a search button could be added.
_____________________
#### Workaround for now:
- When adding `unbound`, adjusting scheduling options, and excluding it again, it will not be shown in the list, but scheduling options will stay. You can use `htop` e.g. after reboot or such, to verify that Nice and priority levels are still as chosen before.
- Scheduling options are saved/set in: `/etc/systemd/system/unbound.service.d/dietpi-process_tool.conf`
- This is a systemd drop-in config that is applied outside of DietPi tools after set once.

AtkLordOverAll, 2019-07-14:
I did wonder if it was expected behaviour, and yes you have correctly guessed why I still wanted to be able to use the utility while having it excluded - access to nice and I/O scheduling.
Shall make use of the workaround, thank you! It is worth noting that when you follow the workaround, you will get an error when exiting the DietPi-Process_Tool (as it tries to set attributes of a null service), you are safe to ignore it, you just need to reboot / restart the service to see the effects

MichaIng, 2019-07-14:
@AtkLordOverAll 
> It is worth noting that when you follow the workaround, you will get an error when exiting the DietPi-Process_Tool (as it tries to set attributes of a null service), you are safe to ignore it, you just need to reboot / restart the service to see the effects

Ah that is true. When excluding a service, that was manually included before, it is removed from the list by unsetting it's array entries. However when scheduling settings were changed, its index is still marked to trigger a restart. Can be easily solved by leaving the array when excluding the service. This even allows to still edit settings of the service within the current DietPi-Services session.