MichaIng/DietPi #2985: sed error in DietPi banner

UNiXMIT, 2019-07-15:
Since updating to v6.25.3 I am seeing the following error in the banner:

sed: couldn't open temporary file /DietPi/dietpi/sedf86Tcn: Permission denied

I don't see that file in the /DietPi/dietpi directory and it's not something I'm generating manually. It's always a different filename - sed*****
It seems to be caused by my script which is run by dietpi-banner to display some information I want in the banner. This is the script:

#!/bin/bash
 
curl -s wttr.in/Andover,UK?format="4";
echo " - Root Disk Space : $(df -h / | grep / | awk '{ print $4}') remaining - $(df -h / | grep / | awk '{ print $5}') full";
echo " - UNiX Disk Space : $(df -h /mnt/unix | grep / | awk '{ print $4}') remaining - $(df -h /mnt/unix | grep / | awk '{ print $5}') full";
echo " - Backup Disk Space : $(df -h /mnt/backup | grep / | awk '{ print $4}') remaining - $(df -h /mnt/backup | grep / | awk '{ print $5}') full";
echo " - Memory : $(free -tmh | head -n +2 | tail -1 | awk '{ print $3}')";
echo " - Swap : $(free -tmh | head -n +3 | tail -1 | awk '{ print $3}')";

In dietpi-banner it's launched by:
sh /home/unix/banner.sh

This didn't happen with the previous version. Any ideas?
Thanks

------
v6.25.3
Debian Stretch
Linux DietPi 3.10.105-bsp-1.2-ayufan-118 #1 SMP PREEMPT Sun Oct 15 21:40:57 UTC 2017 aarch64 GNU/Linux
Pine64

MichaIng, 2019-07-15:
@UNiXMIT 
Many thanks for your report.
Can you please do:
```
rm /tmp/.dietpi_motd
/DietPi/dietpi/func/dietpi-banner 1
```

UNiXMIT, 2019-07-15:
Thanks so much @MichaIng that's fixed it.

closed 2019-07-15 by UNiXMIT