MichaIng/DietPi #2986: Can't connect to https://raspbian.raspberrypi.org/raspbian while I'm connected

Proxyl3ss, 2019-07-16:
#### Details:
- Date           | Tue 16 Jul 11:31:00 BST 2019
- Bug report     | N/A
- DietPi version | v6.23.0 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | #1219 SMP Tue May 14 21:20:58 BST 2019
- Distro         | stretch (index=4)
- Command        | Connection test: http://raspbian.raspberrypi.org/raspbian
- Exit code      | 4
- Software title | DietPi-Login

#### Steps to reproduce:

1. Flash SD card
2. Connect Ethernet and power it up
3. SSH into it.

#### Expected behaviour:

- It should continue downloading the updates because it has internet connection but it doesn't

#### Actual behaviour:

- Shows an error that it cannot connect to the raspberrypi server while I can sudo apt-get update

#### Extra details:

- I'm running a 2GB SD Card with dietpi because it's lightweight.

#### Additional logs:
```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2019-07-16 11:29:31--  http://raspbian.raspberrypi.org/raspbian
Resolving raspbian.raspberrypi.org (raspbian.raspberrypi.org)... failed: Connection timed out.
wget: unable to resolve host address ‘raspbian.raspberrypi.org’
```

MichaIng, 2019-07-16:
@Proxyl3ss 
Many thanks for your report.

> Resolving ... timed out.

Your DNS request took more than 5 seconds, which timed out the connection test. You can increase the timeout via: `dietpi-config` > `Network Options: Misc` > `G_CHECK_URL Timeout`

But since a DNS request should not take more than ~0.5 seconds, you should investigate why it takes so long in your case:
- `time wget --spider https://dietpi.com` # Check other URL
- `cat /etc/resolv.conf` # Check active nameserver
- `echo 'nameserver 8.8.8.8' > /etc/resolv.conf && time wget --spider https://dietpi.com` # Test with Google DNS to compare resolving time.

Do you use Pi-hole or unbound or any other local DNS server?
Do you use DHCP or static IP?

Proxyl3ss, 2019-07-16:
I just increased the timeout to 10 second and it worked! Thank you for your
time.

Proxyl3ss, 2019-07-16:
I also faced another problem. I regularly proceeded with the set up but when it goes to the point that it says Failed connection attempt to ```https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt (2/3), retrying...``` it just doesn't connect again. I increased the timeout time again and it doesn't apply to that download. It just used the default 5 seconds for some reason

Proxyl3ss, 2019-07-16:
I am not using a static IP and im not using pihole. Also testing with Google dns doesn't change anything.

MichaIng, 2019-07-16:
@Proxyl3ss 
Hmm strange. You changed the setting right from the error prompt?

Can you paste:
```
grep -m1 '^[[:blank:]]*CONFIG_G_CHECK_URL_TIMEOUT=' /DietPi/dietpi.txt | sed 's/^[^=]*=//'
```
- This value is used for every connection check, no sure how once 10 seconds can be used and 5 seconds the next time.

You can also override the default within the script:
```
sed -i 's/local timeout=5/local timeout=10/' /DietPi/dietpi/func/dietpi-globals
```

Proxyl3ss, 2019-07-16:
I somehow tried logging in to it right now and it is working. Sorry for
wasting your time and thank you once again for your help.

MichaIng, 2019-07-17:
Okay I mark this as closed then. Feel free to reopen if required.

closed 2019-07-17 by MichaIng