MichaIng/DietPi #2987: Fuzon.co.uk is down

crazyquark, 2019-07-16:
The host fuzon.co.uk seems to be down again.

ponasromas, 2019-07-16:
Exact error:

```
Err:9 http://fuzon.co.uk/meveric all Release
  Unable to connect to fuzon.co.uk:http:
Err:10 http://fuzon.co.uk/meveric stretch Release
  Unable to connect to fuzon.co.uk:http:
Reading package lists... Done
E: The repository 'http://fuzon.co.uk/meveric all Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'http://fuzon.co.uk/meveric stretch Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

MichaIng, 2019-07-16:
Confirmed, I am informing @Fourdee .

Until then you can use an alternative mirror:
```
sed -Ei 's@http://fuzon.co.uk@https://oph.mdrjr.net@' /etc/apt/sources.list.d/meveric*
G_AGUP
```

Fourdee, 2019-07-17:
@MichaIng 
fuzon.co.uk is back up. Apologies. It uses the old payment system in MVS which does not support auto renewals.

closed 2019-07-17 by Fourdee