MichaIng/DietPi #2990: ODROID XU4 incorrect hardware version

ponasromas, 2019-07-17:
### Creating a bug report/issue

#### Required Information
- DietPi version 
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=25
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='Fourdee

- Distro version
stretch

- Kernel version 
Linux DietPi 4.14.66+ #1 SMP PREEMPT Thu Aug 23 05:59:33 UTC 2018 armv7l GNU/Linux

- SBC device
Odroid XU3/**XU4**/HC1/HC2 (armv7l)

#### Steps to reproduce
1. Install latest ODROID XU4 DietPI OS
2. Perform `cat /proc/cpuinfo | grep Hardware`
3. You will get XU3 even if hardware is XU4

#### Expected behaviour
O original Ubuntu 18.04.1 it shows correct hardware version for ODROID XU4

MichaIng, 2019-07-17:
@ponasromas 
Thanks for your report.

Not sure if this is related to the kernel or bootloader or how it is invoked. At least the boot.ini explicitly checks for the model: https://github.com/MichaIng/DietPi/blob/master/boot_xu4.ini#L124

If you find time, could you try with a fresh Meveric image: https://forum.odroid.com/viewtopic.php?f=96&t=30552#p219509

ponasromas, 2019-07-18:
Can't test it, just setuped production environment :( . Does the hardware detection can impact something meaningful such as performance etc.?

MichaIng, 2019-07-18:
@ponasromas
Not that I'm aware of. All share the same CPU + GPU with same default clocks.

Technetium1, 2020-02-09:
@MichaIng `cat /proc/cpuinfo | grep Hardware`
on above linked image:
`Hardware        : ODROID-XU3`

on current Old Stable - DietPi_v6.14_OdroidXU4-ARMv7-Stretch:
`Hardware        : ODROID-XU3`

on Testing - DietPi_OdroidXU4-ARMv7-Buster [2020-02-04 01:11]:
`Hardware        : ODROID-XU3`

I got an error trying to start up the current Testing, I will open a new issue for it. I'm happy to contribute anything that might help here.

MichaIng, 2020-02-11:
@Technetium1 
Since the image works very well otherwise on XU3 + XU4 + HC1 + HC2 + MC1, this should be not an issue. Probably the kernel for Ubuntu was slightly adjusted compared to the one maintained by Meveric.

Could you please check: `cat /proc/device-tree/model`

Technetium1, 2020-02-13:
@MichaIng Testing last modified `2020-02-09 23:12`:
`Hardkernel Odroid XU4` 🎊

MichaIng, 2020-02-13:
@Technetium1 
Strange, I actually only fixed some missing ca-certificates/curl issue with the last upload, no kernel changes 🤔: https://github.com/MichaIng/DietPi/issues/3376#issuecomment-583901295
However great if shows the correct hardware now 👍.

MichaIng, 2020-02-13:
I mark this as closed then. Also good to know the new image works fine in general, so we can move this to stable soon.

Technetium1, 2020-02-14:
@MichaIng Just to reiterate, it's only `/proc/device-tree/model` that shows XU4 correctly, but I do not believe it matters much. I think this can now be closed safely now!

MichaIng, 2020-02-14:
@Technetium1 
Ah of course. Yes so that means that as far as the kernel supports it, it detects the different models correctly, uses the right dtb etc. A differentiation within `/proc/cpuinfo` seems to not have been implemented. But yeah it is visual only and does not matter, simply because the SoC of all boards is the same anyway.

We could report this to Meveric, if this is something he can implement in his kernel packages, else we can forward to hardkernel where it most likely will not be handled at all (no priority): https://forum.odroid.com/viewtopic.php?f=96&t=30552

closed 2020-02-14 by MichaIng