MichaIng/DietPi #2991: Dietpi without internet?

zenmaster24, 2019-07-18:
Hi,

I am filing this ias a bug, but I dont know if it is.

Is it possible to use DietPi *without* an internet connection? Just by itself, connected to the lan.
When booting, it fails with an error message about being unable to spider a file online - i dont need internet connecitivity all the time, I just want to run it to see how it goes.

Using Raspberry Pi 2, with an ethernet connection if that matters.

Cheers!

Fourdee, 2019-07-18:
@zenmaster24 

Hi,

Yes, firstly disable boot wait for network:
- ```dietpi-config```
- networking options: Misc
- Boot net wait. Set to ```disabled```

Secondly (if above still fails URL test) we need to find out what exactly is triggering this in our code. 
Ideally, if you could take a screenshot or the URL its checking, we can track it down.

MichaIng, 2019-07-18:
And you as well need to disable network time sync. dietpi-config advanced options.

Basically we need an offline mode to automatically skip all network request attempts: https://github.com/MichaIng/DietPi/issues/2915

zenmaster24, 2019-07-19:
Hi,

Thanks for the pointers - I did the following in **dietpi.txt**

1. Set `CONFIG_BOOT_WAIT_FOR_NETWORK` to `0`
2. Set `CONFIG_NTP_MODE` to `0`.

Would that be enough to do the above tasks mentioned above?
After doing this and booting with nothing connected but the TV, I still get the error.

The relevant paraphrased parts of the error are :

```
- Command | Connection test: http://raspbian.raspberrypi.org/raspbian
- Exit code | 4

Log file contents:
Spider mode enabled. Check if remote file exists.
Resolving raspbian.raspberrypi.org... failed
wget: unable to resolve host address raspbian.raspberrypi.org

[FAILED] DierPi-Login | Unable to continue, DietPit-Login will now terminate
root@DietPi:~#
```

Thanks :)

MichaIng, 2019-07-19:
@zenmaster24 
Ah so it's the first boot.

As long as there is no strong reason to keep it disconnected from internet, I suggest you at least connect it for the first boot (or first login), to allow auto-update to the newest version and finishing the firstrun setup.

However if you want to skip for now: `echo 2 > /DietPi/dietpi/.install_stage`
As fast as you have internet, revert: `echo 0 > /DietPi/dietpi/.install_stage`
Then login again to apply firstrun setup steps.

zenmaster24, 2019-07-19:
Thanks @MichaIng - will give it a try!

MichaIng, 2019-07-26:
I mark this as closed, referring to: https://github.com/MichaIng/DietPi/issues/2915
Let's collect all info there about which steps are required to have DietPi running nicely without internet access.

closed 2019-07-26 by MichaIng