MichaIng/DietPi #2992: Subsonic Jukebox mode issue / no sound output on 3mm port on pi 3

stownplayer, 2019-07-18:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=25
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
stretchcat /etc/debian_version

- Kernel version | `uname -a`
Linux SubsonicHome 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l GNU/Linux

- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
RPi 3 Model B+ (armv7l)

- Power supply used | (EG: 5V 1A RAVpower)
pi store brand - 5v 1amp
- SDcard used | (EG: SanDisk ultra)
Sandisk extreme

#### Additional Information (if applicable)
- Software title |  Subsonic
- Was the software title installed freshly or updated/migrated? Installed in January
- Can this issue be replicated on a fresh installation of DietPi? Yes
   71e51554-7012-4383-b8d9-f9b91d180a1c

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. ... Install and configure subsonic. No special setup. Audio and server work flawless once music shares are added and scanned. When playing music from an app like Play:sub and selecting jukebox mode, which should output sound on the pi3, there is no sound. 
2. ...

#### Expected behaviour
 There should be sound output on the 3mm jack when I play the app in jukebox mode.
- ...

#### Actual behaviour
 No sound
- ...

#### Extra details

sound works on the pi using aplay and testing a wave file. No issues. The jukebox mode is the only issue. I've also tested hifiberry dac and get the same results. System sound with aplay works but Subsonic won't output sound. 

The issue seems related to Java sound settings. See the link below. Seems the java settings in the subsonic settings does not use the correct audio device. I would test further but I'm unable to get an accurate list of java audio devices in the format "NVidia [plughw:0,3]" for example. This seems to be at the root of the issue and I don't know if that is an issue that can be solved by the underlying os or if it's simply a Subsonic issue. 
http://forum.subsonic.org/forum/viewtop ... java+sound
- ...

MichaIng, 2019-07-18:
@stownplayer
Many thanks for your report. We will investigate.

MichaIng, 2019-07-19:
Confirmed issue that transcoding settings for Java are hardcoded, thus the transcoder settings applied in Subsonic web UI are without any effect: http://forum.subsonic.org/forum/viewtopic.php?f=2&t=18279

If this hardcoded transcoding settings do not work (on my test VM system it does not work as well), Jukebox mode simply does not work, instead each attempt to play music spawns an additional ffmpeg instance that freezes and can only by stopped via SIGKILL... Not very elegant, another reason why I would never use Java... besides the 300 MiB RAM usage just for a web interface...

I am testing now with Airsonic, perhaps they solved this.

MichaIng, 2019-07-19:
Okay Airsonic docs provide useful information about how to force Java use a certain sound device: https://airsonic.github.io/docs/jukebox/

Testing to add to the command (`hw:0,0` is the valid sound device):
```
-Djavax.sound.sampled.Clip=#PCH [plughw:0,0] -Djavax.sound.sampled.Port=#Port PCH [hw:0] -Djavax.sound.sampled.SourceDataLine=#PCH [plughw:0,0] -Djavax.sound.sampled.TargetDataLine=#PCH [plughw:0,0]
```

MichaIng, 2019-07-20:
@stownplayer 
To come back to the actual issue with Subsonic, what you can try:
```
root@VM-Stretch:/tmp# cat << _EOF_ > audioDevList.java
import java.io.*;
import javax.sound.sampled.*;

public class audioDevList {
    public static void main(String args[]) {
        Mixer.Info[] mixerInfo =
            AudioSystem.getMixerInfo();
            System.out.println("Available mixers:");
            for(int cnt = 0; cnt < mixerInfo.length;cnt++) {
                System.out.println(mixerInfo[cnt].getName());
        }
    }
}
_EOF_
root@VM-Stretch:/tmp# javac audioDevList.java
root@VM-Stretch:/tmp# java audioDevList
```
- This hopefully lists all available sound devices on your RPi that are available for Java.

Then to apply these for all Java applications (including Subsonic):
```
mkdir -p /usr/lib/jvm/java-1.8-openjdk/jre/lib
cat << _EOF_ > /usr/lib/jvm/java-1.8-openjdk/jre/lib/sound.properties
javax.sound.sampled.Clip=#PCH [plughw:1,0]
javax.sound.sampled.Port=#Port PCH [hw:1]
javax.sound.sampled.SourceDataLine=#PCH [plughw:1,0]
javax.sound.sampled.TargetDataLine=#PCH [plughw:1,0]
_EOF_
```
- Something like this, with the values matching the actual correct sound device(s) from the list.

Ref: https://airsonic.github.io/docs/jukebox/

MichaIng, 2019-07-20:
Okay:
- The transcoding settings in the web UI are for streaming only, thus they have no effect on Jukebox mode, where nothing is streamed anyway (to the web client).
- Java uses internal audio settings for audio device and transcoding.
- Headless JRE package is not sufficient to allow Java direkt audio access, so `G_AGI openjdk-8-jre` is required to allow Jukebox mode.
- https://airsonic.github.io/docs/jukebox/ explains how to list available sound devices for Java and configure it to use a certain one by default. The script needs to be compiled (JDK required, installed with Java by default on DietPi currently): https://github.com/MichaIng/DietPi/issues/2992#issuecomment-513466665
- But this does not affect the Java internal transcoder command options, which seems to be hardcoded as well: http://forum.subsonic.org/forum/viewtopic.php?f=2&t=18279
  - However perhaps something can be done about this, I did not fully check the code and the link does not include further details.
  - Also if it is hardcoded, it should work in most cases, perhaps local audio setup needs to be adjusted.
- Airsonic has two Jukebox modes, "Java Jukebox" allows to choose exactly from the sound devices that the above scripts as available for Java. So I guess the custom settings are not required for this mode.
- However both Jukebox modes fail in my case, basically with the same error:
```
Normal Jukebox mode:
ERROR --- o.a.p.s.JukeboxLegacySubsonicService     : Error in jukebox: javax.sound.sampled.LineUnavailableException
Java Jukebox mode:
Exception in thread "com.github.biconou.AudioPlayer.JavaPlayer.Tplayer" java.lang.RuntimeException: java.lang.RuntimeException: javax.sound.sampled.LineUnavailableException: line with format PCM_SIGNED 44100.0 Hz, 16 bit, stereo, 4 bytes/frame, little-endian not supported.
```
- Both share: `javax.sound.sampled.LineUnavailableException`, web search does not really give useful info. I tried with different sound cards, `sound.properties` (Java) settings, ALSA settings, permissions (run as root) etc.
- I also tested Subsonic and the exactly same error occurs.

Perhaps also the VM and virtual sound cards it the issue, could be tested on physical device.

stownplayer, 2019-07-21:
> @stownplayer
> To come back to the actual issue with Subsonic, what you can try:
>
> ```
> root@VM-Stretch:/tmp# cat << _EOF_ > audioDevList.java
> import java.io.*;
> import javax.sound.sampled.*;
>
> public [...]

I get "PulseAudio Mixer" when I run audioDevList.Java.  Should I proceed testing although I don't get the same audio dev list format like you listed?

MichaIng, 2019-07-21:
@stownplayer 
> I get "PulseAudio Mixer" when I run audioDevList.Java.

As I found later, install the non-headless JRE package to get the full device list: `G_AGI openjdk-8-jre`

stownplayer, 2019-07-21:
> @stownplayer
> 
> > I get "PulseAudio Mixer" when I run audioDevList.Java.
> 
> As I found later, install the non-headless JRE package to get the full device list: `G_AGI openjdk-8-jre`

Is that safe to do on a "production" install. I use that music server quite a lot.

MichaIng, 2019-07-21:
@stownplayer 
Jep, it will not affect any running service/process and not replay any files anyway. It is an addition to the headless Java JRE package that includes a GUI for desktops systems e.g. but obviously as well enables direct access to audio devices for Java.

MichaIng, 2019-07-21:
Testing on RPi with onboard 3.5mm sound enabled. aplay works well:
- _Sidenote: Raspbian Buster repo ships all OpenJDK version: 8, 9, 10 and 11, so we could allow Subsonic + Airsonic install there._
```
root@DietPi:/tmp# java audioDevList
Available mixers:
PulseAudio Mixer
ALSA [default]
ALSA [plughw:0,0]
ALSA [plughw:0,1]
Port ALSA [hw:0]
```
Same error:
```
ERROR JukeboxService - Error in jukebox: javax.sound.sampled.LineUnavailableException
```

@stownplayer 
Ah the sound properties file is located differently on armhf and RPi:
```
root@DietPi:/tmp# ls -Al /usr/lib/jvm/
total 8
lrwxrwxrwx 1 root root   20 Mar 29 14:54 java-1.8.0-openjdk-armhf -> java-8-openjdk-armhf
-rw-r--r-- 1 root root 2712 Mar 29 14:54 .java-1.8.0-openjdk-armhf.jinfo
drwxr-xr-x 7 root root 4096 Jul 21 17:44 java-8-openjdk-armhf
root@DietPi:/tmp# ls -Al /usr/lib/jvm/java-8-openjdk-armhf/jre/lib/sound.properties
lrwxrwxrwx 1 root root 36 Mar 29 14:54 /usr/lib/jvm/java-8-openjdk-armhf/jre/lib/sound.properties -> /etc/java-8-openjdk/sound.properties
```
So the actual file is located at: `/etc/java-8-openjdk/sound.properties`
By default PulseAudio is indeed configured. There are commented lines containing "DirectAudioDevice", will try it with them.

stownplayer, 2019-07-21:
So the confirmed settings for Java are the list below with everything else commented out in the file /etc/java-8-openjdk/sound.properties? I didn't have any luck with this. Perhaps I am misunderstanding what you are doing. 

javax.sound.sampled.Clip=com.sun.media.sound.DirectAudioDeviceProvider
javax.sound.sampled.Port=com.sun.media.sound.PortMixerProvider
javax.sound.sampled.SourceDataLine=com.sun.media.sound.DirectAudioDeviceProvider
javax.sound.sampled.TargetDataLine=com.sun.media.sound.DirectAudioDeviceProvider

MichaIng, 2019-07-21:
@stownplayer 
First of all please remove the other file, it seems to be without effect anyway, but just to be failsafe:
```
/usr/lib/jvm/java-1.8-openjdk/jre/lib/sound.properties
rmdir -p --ignore-fail-on-non-empty /usr/lib/jvm/java-1.8-openjdk/jre/lib
```

Okay your `/etc/java-8-openjdk/sound.properties` then exactly looks like mine: Everything commented but those 4 lines.

`G_AGI openjdk-8-jre` done?

And Subsonic restarted via:
```
killall -w java
systemctl restart subsonic
```
?

MichaIng, 2019-07-21:
Finally I got it working on VM as well with:
```
javax.sound.sampled.Clip=#I82801AAICH [plughw:0,0]
javax.sound.sampled.Port=#Port I82801AAICH [hw:0]
javax.sound.sampled.SourceDataLine=#I82801AAICH [plughw:0,0]
javax.sound.sampled.TargetDataLine=#I82801AAICH [plughw:0,0]
```

While testing Airsonic on RPi concurrently... It took about 1.5 hours for first web UI access and login. Now waiting again lets see how long to access settings tab the fist time 🙄. Need to forward this to Airsonic devs to have a look. One could say "Who cares, it's just first start + first access each page", but no user expects this and will simply interpret it as failed after ~5 minutes no access success, thus drop the program. At least I would do...

MichaIng, 2019-07-21:
This worked as well on RPi2:
```
javax.sound.sampled.Clip=#ALSA [plughw:0,0]
javax.sound.sampled.Port=#Port ALSA [hw:0]
javax.sound.sampled.SourceDataLine=#ALSA [plughw:0,0]
javax.sound.sampled.TargetDataLine=#ALSA [plughw:0,0]
```

stownplayer, 2019-07-22:
can you advise on best way to install G_AGI openjdk-8-jre? I don't see it in the diet-pi software list.  I'll test further this evening if I can.

stownplayer, 2019-07-22:
> can you advise on best way to install G_AGI openjdk-8-jre?  I'll test further this evening if I can.

I got he dietpi error handler: G_AGI openjdk-8-jre Exit code: 100
Failed to fetch. Http://respbian.raspberrypi.org/raspbian/pool/main/o/openjdk-8/openjdk-8-jre-.........................

stownplayer, 2019-07-22:
> > can you advise on best way to install G_AGI openjdk-8-jre?  I'll test further this evening if I can.
> 
> I got he dietpi error handler: G_AGI openjdk-8-jre Exit code: 100
> Failed to fetch. [Http://respbian.raspberrypi.org/raspbian/pool/main/o/openjdk-8/openjdk-8-jre-](Http://respbian.raspberrypi.org/raspbian/pool/main/o/openjdk-8/openjdk-8-jre-).........................

Actually, looks like apt-get update fixed the issue. I'll continue testing but I'm not local to the device. I'll follow up when I'm home later.

MichaIng, 2019-07-22:
@stownplayer 
> Actually, looks like apt-get update fixed the issue.

Jep "Failed to fetch." usually means that the APT list is outdated, so `apt update` or `G_AGUP` fixes it.

stownplayer, 2019-07-22:
> @stownplayer
> First of all please remove the other file, it seems to be without effect anyway, but just to be failsafe:
> 
> ```
> /usr/lib/jvm/java-1.8-openjdk/jre/lib/sound.properties
> rmdir -p --ignore-fail-on-non-empty /usr/lib/jvm/java-1.8-openjdk/jre/lib
> ```
> 
> Okay your `/etc/java-8-openjdk/sound.properties` then exactly looks like mine: Everything commented but those 4 lines.
> 
> `G_AGI openjdk-8-jre` done?
> 
> And Subsonic restarted via:
> 
> ```
> killall -w java
> systemctl restart subsonic
> ```
> 
> ?
> 
> If so, could you verify the last error log entry in `/var/subsonic/subsonic.log` looking like this:
> 
> ```
> [2019-07-21 17:52:47,065] ERROR JukeboxService - Error in jukebox: javax.sound.sampled.LineUnavailableException: line with format PCM_SIGNED 44100.0 Hz, 16 bit, stereo, 4 bytes/frame, big-endian not supported.
> javax.sound.sampled.LineUnavailableException: line with format PCM_SIGNED 44100.0 Hz, 16 bit, stereo, 4 bytes/frame, big-endian not supported.
>         at com.sun.media.sound.DirectAudioDevice$DirectDL.implOpen(DirectAudioDevice.java:513)
> ```
> 
> * This is what I still get on VM. On my RPi2 it works, but on VM it does not.

Although, I get nothing listed under Available mixers when I run audiodevlist after following your directions and restarting java and subsonic. I got the "OK" after installing G_AGI openjdk-8-jre but other than that I have no idea if it's correct. I didn't find any log entries that look like you posted.

stownplayer, 2019-07-22:
ok, something broke subsonic. Working on restoring a backup.  :(

stownplayer, 2019-07-23:
> ok, something broke subsonic. Working on restoring a backup. :(

ok, everything back running. I'll attempt to test again after I update diet-pi and backup the most recent version. I evidently rolled back to 7/11.......

MichaIng, 2019-07-23:
@stownplayer 
After `G_AGI openjdk-8-jre`
The following should definitely list other audio devices besides PulseMixer:
```
cd /tmp
cat << _EOF_ > audioDevList.java
import java.io.*;
import javax.sound.sampled.*;

public class audioDevList {
    public static void main(String args[]) {
        Mixer.Info[] mixerInfo =
            AudioSystem.getMixerInfo();
            System.out.println("Available mixers:");
            for(int cnt = 0; cnt < mixerInfo.length;cnt++) {
                System.out.println(mixerInfo[cnt].getName());
        }
    }
}
_EOF_
javac audioDevList.java
java audioDevList
```
- If it does not, then any changes in `/etc/java-8-openjdk/sound.properties` will most likely not work.
________
Another question:
- When you run `systemctl stop subsonic`, does the Subsonic process is really closed in your case? You can check via `htop`.
- In my case, on x86_64 machines and VMs, it works as expected, but on my RPi (at least Raspbian Buster based image) it has no effect. I have to run `killall -w java` to kill it and start again via `systemctl restart subsonic`. Otherwise any Java settings changes will have no effect, the service always needs to be restarted.

stownplayer, 2019-07-23:
> @stownplayer
> After `G_AGI openjdk-8-jre`
> The following should definitely list other audio devices besides PulseMixer:
> 
> ```
> cd /tmp
> cat << _EOF_ > audioDevList.java
> import java.io.*;
> import javax.sound.sampled.*;
> 
> public class audioDevList {
>     public static void main(String args[]) {
>         Mixer.Info[] mixerInfo =
>             AudioSystem.getMixerInfo();
>             System.out.println("Available mixers:");
>             for(int cnt = 0; cnt < mixerInfo.length;cnt++) {
>                 System.out.println(mixerInfo[cnt].getName());
>         }
>     }
> }
> _EOF_
> javac audioDevList.java
> java audioDevList
> ```
> 
> * If it does not, then any changes in `/etc/java-8-openjdk/sound.properties` will most likely not work.
> 
> Another question:
> 
> * When you run `systemctl stop subsonic`, does the Subsonic process is really closed in your case? You can check via `htop`.
> * In my case, on x86_64 machines and VMs, it works as expected, but on my RPi (at least Raspbian Buster based image) it has no effect. I have to run `killall -w java` to kill it and start again via `systemctl restart subsonic`. Otherwise any Java settings changes will have no effect, the service always needs to be restarted.
> 
> #### EDIT: Ah I remember the related issue: #2787
> * Possibly the issue is that DietPi-Software installs Java 11 on Buster by default and probably Subsonic runs as Java 11 then, which is not officially supported.
> * However not an issue since we do not support Subsonic+Airsonic on Buster for that reason. When enabling support for RPi Buster (where Java 8 is available via APT repo), then Java 8 only needs to be installed anyway.
> * However would be great if you could verify that on your Stretch image, `systemctl` handles the service correctly, just to be failsafe.

Ok, audioDevlist shows

root@SubsonicHome:~# java audioDevList
Available mixers:
PulseAudio Mixer
ALSA [default]
ALSA [plughw:0,0]
ALSA [plughw:0,1]
Port ALSA [hw:0]

systemctl stop subsonic does indeed stop that process - Confirmed
I'll confirm more when I can get time tonight.

MichaIng, 2019-07-23:
@stownplayer 
Okay so far so good. When you find time, test the following `/etc/java-8-openjdk/sound.properties` content:
- First
```
javax.sound.sampled.Clip=com.sun.media.sound.DirectAudioDeviceProvider
javax.sound.sampled.Port=com.sun.media.sound.PortMixerProvider
javax.sound.sampled.SourceDataLine=com.sun.media.sound.DirectAudioDeviceProvider
javax.sound.sampled.TargetDataLine=com.sun.media.sound.DirectAudioDeviceProvider
```
- Second
```
javax.sound.sampled.Clip=#ALSA [plughw:0,0]
javax.sound.sampled.Port=#Port ALSA [hw:0]
javax.sound.sampled.SourceDataLine=#ALSA [plughw:0,0]
javax.sound.sampled.TargetDataLine=#ALSA [plughw:0,0]
```

As said, the first worked for me on RPi only, while the second worked on RPi and VM. RPi with onboard 3.5mm sound enabled.

I hope one works for you as well, then we can add this info to the Subsonic (and Airsonic) online docs first and when I find time I add this as optional choice prompt to DietPi-Software installs, when being called interactively.

Interesting that Java makes it so complicated instead of just allowing to use the default soundcard configured with ALSA 🙄.

stownplayer, 2019-07-23:
Verified working! Very nice. Jukebox mode via Play:sub and web gui working perfectly through the 3mm jack. Now I'll have to hook the hifiberry dac up and see if I can get that working too. I would imagine the Java Audiodevlist will show it once I hook it up and I've set it as the default device.? Then it's just a matter of adding that info to the Java sound properties? I'll post the output in a little bit. 

By the way the working setting for my pi in the java sound properties was the first one. 

javax.sound.sampled.Clip=com.sun.media.sound.DirectAudioDeviceProvider
javax.sound.sampled.Port=com.sun.media.sound.PortMixerProvider
javax.sound.sampled.SourceDataLine=com.sun.media.sound.DirectAudioDeviceProvider
javax.sound.sampled.TargetDataLine=com.sun.media.sound.DirectAudioDeviceProvider

stownplayer, 2019-07-24:
ok. I plugged up the Hifiberry DAC and set it as the sound device in dietpi-config. Rebooted and aplay worked to play a sound file. Jukebox mode on Subsonic does not work with the Hifiberry. 

Java audiodevices lists:

root@SubsonicHome:~# java audioDevList
Available mixers:
PulseAudio Mixer
sndrpihifiberry [plughw:0,0]
Port sndrpihifiberry [hw:0]

stownplayer, 2019-07-24:
> ok. I plugged up the Hifiberry DAC and set it as the sound device in dietpi-config. Rebooted and aplay worked to play a sound file. Jukebox mode on Subsonic does not work with the Hifiberry.
>
> Java [...]

Also, new issue. I had restored diet pi but it was on a previous version. Verified working last night in jukebox mode . I ran the update this morning and planned on doing the backup but I decided to test and Jukebox mode is not working after the update. Any ideas on that on? The sound properties and audiodevlist have not changed. Nor has the default audio device in dietpi-config.

MichaIng, 2019-07-24:
@stownplayer 
A question aside: Why did you need to restore to a previous version? It should not cause issues, but there are only rare situations where it is required, e.g. when important system files have been corrupted or something. In very most cases it is easier to e.g. un/reinstall a faulty software title or revert steps you did (generally when doing system configurations, leave some notes what you did, so you can revert 😉).

Please recheck all the steps we did above. Reapplying the sound device via dietpi-config, restart Subsonic or reboot could help.

For `Hifiberry DAC`, if these `DirectAudioDeviceProvider` defaults of sound.properties do not work, I would try this:
```
javax.sound.sampled.Clip=#sndrpihifiberry [plughw:0,0]
javax.sound.sampled.Port=#Port sndrpihifiberry [hw:0]
javax.sound.sampled.SourceDataLine=#sndrpihifiberry [plughw:0,0]
javax.sound.sampled.TargetDataLine=#sndrpihifiberry [plughw:0,0]
```
I hope you get the system:
- There is always one "Port" entry that only contains the sound card (e.g. `hw:0`). This needs to be applied to the `javax.sound.sampled.Port` setting. Simply the whole line with leading `#`.
- And then there is at least one other related entry with same soundcard and additionally defining a specific device of this card, e.g. `plughw:0,0`, but could be additionally one with `plughw:0,1`. These different devices can be e.g. different output types/plugs of the same soundcard, analogue vs digital or optical and such. So one of these need to be applied to the other three settings, as well whole line with leading `#`.
_(For adjusting a sample rate, it always needs to be a `plughw` entry. The `plug` module allows e.g. downsampling, compared to the bare `hw` module. And this is required for the default ffmpeg call of Java.)_

stownplayer, 2019-07-24:
I restored from the dietpi-backup because I had no idea what caused subsonic to break. To be honest, that was easier than waiting days to figure out or resolve the issue. Remember this is well used music server so I really didn't want it down. The latest backup just happened to be right before I had upgraded diet-pi.  

I have reset the audio device to rpi-bcm2835-3.5mm. I rebooted and I have restarted subsonic. No luck unfortunately. I'll dig back through all the steps and see if I can find anything. 

Thanks

MichaIng, 2019-07-24:
@stownplayer 
If only Subsonic broke, I suggest you try:
```
dietpi-software uninstall 34
dietpi-software install 34
```
🙂

Do e.g. `aplay`, `123mpg`, `123ogg` and such work well?

From which DietPi version did you update?

stownplayer, 2019-07-24:
> @stownplayer
> If only Subsonic broke, I suggest you try:
> 
> ```
> dietpi-software uninstall 34
> dietpi-software install 34
> ```
> 
> 🙂
> 
> Do e.g. `aplay`, `123mpg`, `123ogg` and such work well?
> 
> From which DietPi version did you update?

The only concern would be all my Subsonic playlists, starred songs, folders etc. From past experience re-installs tend to loose all that or have to rebuild it. I many years of information I would prefer to not loose. What will the reinstall do for that info?

I upgraded from the last version. I think 6.25.2 or something like that.

MichaIng, 2019-07-24:
@stownplayer 
Ah yeah settings will be lost, that's true. However sometimes exactly this is what solves issues. I am currently not 100% sure but I believe playlist (besides music files) are all stored in /mnt/dietpi_userdata/Music, right? So those can be re-important into Subsonic relatively easy.

`dietpi-software reinstall 34` will otherwise re-apply the package install (with current version) without affecting existing settings.

I just checked the last update. What we do on RPi is reinstalling ffmpeg to use the default version from Raspbian repo, which has been updated to fully support hardware acceleration etc. After restoring the backup you could go through the steps manually to see which part might break Subsonic:
```
G_AGUP
G_AGI --reinstall ffmpeg
```
```
apt-mark auto libx264 libmp3lame libfdk-aac
G_AGA
```
To include general APT package updates:
```
apt upgrade
```
Since the above includes a kernel+firmware update, load it by reboot:
```
reboot
```
After each above, restart Subsonic and check if it still works.
You don't use MPD, do you?
Aside form that I do not see any step that could affect audio behaviour.
___________
But perhaps better would be to investigate what exactly fails after the update. Perhaps the DietPi-wise update is not even the reason, but an unrelated APT package update or the reboot itself, e.g. enabling a new firmware/kernel version or such (added tests above).
What exactly breaks? Is the web UI not reachable or some element not working? Does the service start itself fail?
Steps to debug:
```
systemctl status subsonic
journalctl -u subsonic
cat /var/subsonic/subsonic.log
```

stownplayer, 2019-07-24:
> @stownplayer
> Ah yeah settings will be lost, that's true. However sometimes exactly this is what solves issues. I am currently not 100% sure but I believe playlist (besides music files) are all stored in /mnt/dietpi_userdata/Music, right? So those can be re-important into Subsonic relatively easy.
> 
> `dietpi-software reinstall 34` will otherwise re-apply the package install (with current version) without affecting existing settings.
> 
> I just checked the last update. What we do on RPi is reinstalling ffmpeg to use the default version from Raspbian repo, which has been updated to fully support hardware acceleration etc. However this should be actually done automatically by regular APT upgrades, so this step is only to be failsafe.
> However after restoring the backup you could go through the steps manually to see which part might break Subsonic:
> 
> ```
> G_AGUP
> G_AGI --reinstall ffmpeg
> ```
> 
> ```
> apt-mark auto libx264 libmp3lame libfdk-aac
> G_AGA
> ```
> 
> To include general APT package updates:
> 
> ```
> apt upgrade
> ```
> 
> Since the above includes a kernel+firmware update, load it by reboot:
> 
> ```
> reboot
> ```
> 
> After each above, restart Subsonic and check if it still works.
> You don't use MPD, do you?
> Aside form that I do not see any step that could affect audio behaviour.
> 
> But perhaps better would be to investigate what exactly fails after the update. Perhaps the DietPi-wise update is not even the reason, but an unrelated APT package update or the reboot itself, e.g. enabling a new firmware/kernel version or such (added tests above).
> What exactly breaks? Is the web UI not reachable or some element not working? Does the service start itself fail?
> Steps to debug:
> 
> ```
> systemctl status subsonic
> journalctl -u subsonic
> cat /var/subsonic/subsonic.log
> ```

Reinstalling ffmpeg did the trick and the hifiberry is working in jukebox mode as well after I update the system card and java settings. I'll continue testing. Thank you!

MichaIng, 2019-07-24:
@stownplayer 
Ah nice. Yeah in the past we installed a self-compiled ffmpeg binary/package to enable hardware acceleration and provide a newer version. However Raspbian has updated it, thus this is not required anymore and further might conflict with new RPi firmware. Hence the reinstall on DietPi-Update.

Yeah Subsonic uses an internal database, so no regular SQLite or MariaDB that could be easily dumped and re-imported. I guess it would be possible to copy the related file to the new instance, but better no mess with this, e.g. if some other settings (provided/saved not within the database) need to match the database or such.

I hope the mentioned Java sound settings for HifiBerry work, so we have some generic steps to get Jukebox mode working with most soundcards.

stownplayer, 2019-07-24:
> @stownplayer
> Ah nice. Yeah in the past we installed a self-compiled ffmpeg binary/package to enable hardware acceleration and provide a newer version. However Raspbian has updated it, thus this is [...]

Hifiberry is working. It's the Hifiberry DAC+ by the way. I am Testing with an RCA to 3mm cable at the moment but working fine. Thanks for all the help

MichaIng, 2019-07-24:
@stownplayer 
Perfect, many thanks for testing and patience.