MichaIng/DietPi #2994: openmediavault-rrdtoolgraph in cronjob even no omv is installed

kt1024, 2019-07-19:
### Creating a bug report/issue
openmediavault-rrdtoolgraph in cronjob even no omv is installed. (never run omv or install command)

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
```
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=25
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='Fourdee'
```

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
9.9
- Kernel version | `uname -a`
Linux DietPi 4.14.0 #82 SMP Fri Dec 8 14:33:14 CST 2017 aarch64 GNU/Linux
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
NanoPi NEO 2 (aarch64)

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi?
Yes, have nano pi neo2 SBC, both have same job.

#### Steps to reproduce

```
root@DietPi:/etc/cron.d# cat openmediavault-rrdtoolgraph 
# Create graphs every 15 minutes
# m h dom mon dow user    command
*/15 * * * * root /usr/sbin/omv-mkgraph >/dev/null 2>&1
```

#### Expected behaviour

- remove _openmediavault-rrdtoolgraph_ from /etc/cron.d/ in source image, coz omv is not pre-installed & is not supported in dietpi.
I'v remove it manuualy.

#### Actual behaviour

 ```
root@DietPi:/etc/cron.d# cat openmediavault-rrdtoolgraph 
# Create graphs every 15 minutes
# m h dom mon dow user    command
*/15 * * * * root /usr/sbin/omv-mkgraph >/dev/null 2>&1
root@DietPi:/etc/cron.d# journalctl | grep omv
Jul 16 21:30:01 DietPi CRON[2612]: (root) CMD (/usr/sbin/omv-mkgraph >/dev/null 2>&1)
Jul 16 21:45:01 DietPi CRON[3310]: (root) CMD (/usr/sbin/omv-mkgraph >/dev/null 2>&1)
Jul 16 22:00:01 DietPi CRON[4015]: (root) CMD (/usr/sbin/omv-mkgraph >/dev/null 2>&1)
Jul 16 22:15:01 DietPi CRON[4748]: (root) CMD (/usr/sbin/omv-mkgraph >/dev/null 2>&1)
Jul 16 22:30:01 DietPi CRON[5445]: (root) CMD (/usr/sbin/omv-mkgraph >/dev/null 2>&1)

```
#### Extra details

- ...

MichaIng, 2019-07-19:
@kt1024
Many thanks for your report.

Looks like a missed leftover from the pre-image we used. I will add a patch with v6.26 to remove this cron job if the related binary is not available.

MichaIng, 2019-07-21:
Okay resolved:
- For new images: https://github.com/MichaIng/DietPi/blob/2675e5fcf4b5cb305a4cdac4b06a6e72f86cd7fa/PREP_SYSTEM_FOR_DIETPI.sh#L1054-L1056
- For existing systems: https://github.com/MichaIng/DietPi/blob/bd2f262515f645dabe70ce6a95ae12ddf219a8a5/dietpi/patch_file#L2085-L2087
- Changelog: https://github.com/MichaIng/DietPi/commit/76728b1bf84bc2d7c20fb6d06b96587b74276a8c

The reason why they are not removed is that they are not part of the current OMV packages anymore: https://packages.openmediavault.org/public/pool/main/o/openmediavault/openmediavault_5.0.4-1_all.deb
- The binary got renamed and the cron job probably replaced by a web UI trigger, as it was responsible to update a web UI graphic.
- No idea why those were not removed by upgrading the packages or left on the image at all, however v6.26 will solve it on all images.

Very good find, many thanks for reporting 👍!

kt1024, 2019-07-24:
found 2 user may be should not be created by default, in /etc/passwd

`
openmediavault-webgui:x:999:997:openmediavault WebGUI Management:/home/openmediavault-webgui:/usr/sbin/nologin
admin:x:998:100:openmediavault WebGUI Administrator:/home/admin:/usr/sbin/nologin
`

MichaIng, 2019-07-24:
@kt1024 
Another good stop, they will be removed as well.

Long story short, OMV images are no good basis for DietPi, I guess it is hard so find some since FriendlyELEC ships the Ubuntu-based FriendlyCore. Perhaps OMV is included in their NAS image (http://download.friendlyarm.com/nanopineo2), not sure. Just found ARMbian as another alternative, which comes with Buster + Linux 4.19.

closed 2019-09-02 by MichaIng