MichaIng/DietPi #2995: DietPi-PREP | Setting up LVM dietpi does not boot

LexiconCode, 2019-07-19:
### Creating a bug report/issue

#### Required Information
Hardware: ODROID-H2
OS: debian-10.0.0-amd64-netinst.iso
Setup: DietPi-PREP

#### Steps to reproduce

1. Startup install Debian 10.0.0
2.  partition entire disc using LVM
[image]
3. Run and complete DietPi PREP 
4. Reboot
5 system is left in a non-bootable state in (initramfs)

#### Expected behaviour

- Should boot int dietpi

#### Actual behaviour
Gave up waiting for suspend/resume device
Gave up waiting for root file system device. Common problems:
- Boot Args(cat /proc/cmdline)
  -check root delay =(did system wait long enough?)
- missing modules (cat /proc/modules; ls /dec)
Alert! /dev/mapper/debian--g--root does not exist.. Dropping to shell!

[/quote]

#### Extra information 
LVM does work, I boot all my ODROIDs off of LVM as rootfs partitions.
But LVM can be a bit tricky itself and sometimes you need to manually add the modules or command lines to run it to the initramfs to work correctly.

Here are some steps you might want to follow:

[*]make sure lvm2 is installed
[*][c]sed -i "/modprobe -q dm-mod/alvm vgchange -ay" /usr/share/initramfs-tools/scripts/local-top/lvm2[/c] (this will add vgchange -ay to the initramfs script which apparently is not loaded correctly)
[*][code]
update-initramfs -d -k `uname -r`
update-initramfs -c -k `uname -r`
update-initramfs -u -k `uname -r`
[/code]seems ridiculous but I had it sometimes that an update of the initramfs did not update the scripts inside, so I normally make "very sure" it is
[*]make sure /boot/boot.ini points to the LVM device by path not UUID for example: /dev/hdd/root or /dev/mapper/hdd-root or whatever you named your LVM volume and group
[*]make sure you set the correct UUID in /etc/fstab 

More info to come

MichaIng, 2019-07-21:
@LexiconCode 
Many thanks for your report.

Never tried it with LVM actually, will do a test install with VM (which should behave the same).

LexiconCode, 2019-12-11:
Just checking in to see if there's news.

MichaIng, 2019-12-12:
Sorry, it was forgotten by me. I assigned v6.28 milestone to remind myself.

LexiconCode, 2020-01-21:
> 
> 
> Sorry, it was forgotten by me. I assigned v6.28 milestone to remind myself.

Thanks for looking into it!  I believe this will allow Urbackup client to backup the server.

MichaIng, 2020-03-18:
@LexiconCode 
I think I found it. From fresh VM with LVM setup I purged `lvm2` page. Reboot worked well. I purged some more packages, most likely not related, then though about the initramfs. Ran `update-initramfs -u` and reboot failed. I need to verify:
```
apt purge lvm2
apt autoremove --purge
update-initramfs -u
reboot
```
If the above breaks VM boot replicable, then we found it. Then we need to check for an LVM mount identifier, basically the source path shows it. Then add `lvm2` package to dependencies. Since this setup includes a swap partition, this would be another question, to re-add the swap partition to /etc/fstab as well.

However I will not be able to add this with v6.29.

MichaIng, 2020-05-18:
LVM support: https://github.com/MichaIng/DietPi/issues/1285#issuecomment-629822409
- + Likely `initramfs-tools` is required instead of `tiny-initramfs`, which is currently installed when choosing VM.
swap partition: https://github.com/MichaIng/DietPi/issues/1285#issuecomment-630098552

@rondadon 
Totally forgot that we have an open request about that 😄.

MichaIng, 2020-05-20:
Additional info: https://github.com/MichaIng/DietPi/issues/1285#issuecomment-630510940

LexiconCode, 2020-08-17:
This is a bit all spread out what's the correct approach?

MichaIng, 2020-08-17:
After DietPi-PREP has finished but before rebooting, do:
```
apt update
apt install lvm2 initramfs-tools
apt autopurge
sed -i '\|[[:blank:]]/[[:blank:]]|s|UUID="[^"]*"|/dev/mapper/vg00-lv01|"' /etc/fstab
update-initramfs -u
```
If there is not (swap) partition present in front of the root partition, `vg00-lv01` must be replaced with `vg00-lv00`.

For the swap partition:
```
/boot/dietpi/func/dietpi-set_swapfile 0
mkswap /dev/mapper/vg00-lv00
swapon /dev/mapper/vg00-lv00
echo '/dev/mapper/vg00-lv00 none swap sw' >> /etc/fstab
```

But check back the drive sources: `blkid`

LexiconCode, 2020-08-20:
Your instructions worked well!