MichaIng/DietPi #2996: udate failed

Paulisper, 2019-07-20:
Hi,
here are some blue lines.
Sorry for that.
Best regards
Jean

#### Details:
- Date           | Sat 20 Jul 13:06:49 BST 2019
- Bug report     | N/A
- DietPi version | v6.22.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi B+ (armv6l) (index=1)
- Kernel version | #1200 Tue Feb 12 20:11:02 GMT 2019
- Distro         | stretch (index=4)
- Command        | G_AGI rsync
- Exit code      | 100
- Software title | DietPi-Backup

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/r/rsync/rs                  ync_3.1.2-1+deb9u1_armhf.deb  404  Not Found
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-mis                  sing?
```

--------------------------------------------------------------------

[FAILED] DietPi-Backup | Unable to continue, DietPi-Backup will now terminate.
[FAILED] DietPi-Backup An issue has occurred

MichaIng, 2019-07-21:
@Paulisper 
Indeed this version is outdated, APT lists need to be updated.
Please do:
```
apt update
dietpi-backup
```
- DietPi-Backup should now be able to install rsync.

MichaIng, 2019-07-26:
The function that is called by dietpi-backup to assure rsync is installed now updates the APT lists automatically, if any package needs to be installed: https://github.com/MichaIng/DietPi/commit/05b644d59a1d655a7db15dec99e6f3db895148f1
Changelog: https://github.com/MichaIng/DietPi/commit/94f8519a48622c28499002a1f881e7ddd5da8bd1

However the case is very rare, solution for pre-v6.26 provided above. I mark this issue as closed, however feel free to reopen if required.

closed 2019-07-26 by MichaIng