MichaIng/DietPi #2997: Wireguard missing from DietPi list in new Pi Zero installation?

johnnyt83, 2019-07-21:
Dear Michael,

I am missing Wireguard on my new DietPi (Pi Zero). How can I install that package? PiHole installation was simple and easy, but now I want to add WireGuard to use my PiHole on the go.

Best wishes.

MichaIng, 2019-07-21:
@johnnyt83 
Many thanks for your question.

Actually WireGuard was never available for ARMv6 systems (RPi 1 and Zero). It is installed from the Debian Sid repository and their armhf branch is only ARMv7 compatible. Only complete source build from scratch would be thinkable. but this would need to be redone after every kernel upgrade.

IMO we should wait for a port to Debian Buster (which will then be available on Raspbian Buster as well, short after), before enabling it for ARMv6 RPis.

johnnyt83, 2019-07-21:
I would like to retest, just guide me gently :)

auch auf deutsch, wenn du magst.

johnnyt83, 2019-07-21:
I already have a working DietPi setup from this morning and Pi-Hole is running just fine. Now I would like to add WireGuard (for dummies).

(already followed this tutorial (https://www.reddit.com/r/pihole/comments/bnihyz/guide_how_to_install_wireguard_on_a_raspberry_pi/) on my other Pi Zero and managed to get it working (VPN at least), but no internet access when connected to WG.

MichaIng, 2019-07-21:
@johnnyt83 
I was mistaken, indeed the WireGuard binaries are architecture specific: https://packages.debian.org/sid/wireguard-tools
You guide supports this, as it suggests a source build for ARMv6.

Here is the WireGuard server config we use:
```
cd /etc/wireguard
cat << _EOF_ > wg0.conf
[Interface]
Address = 10.9.0.1/24
PrivateKey = $(<server_privatekey)
ListenPort = 51820

PostUp = sysctl net.ipv4.conf.%i.forwarding=1 net.ipv4.conf.\$(sed -n 3p /DietPi/dietpi/.network).forwarding=1
PostUp = sysctl net.ipv6.conf.\$(sed -n 3p /DietPi/dietpi/.network).accept_ra=2
PostUp = sysctl net.ipv6.conf.%i.forwarding=1 net.ipv6.conf.\$(sed -n 3p /DietPi/dietpi/.network).forwarding=1
PostUp = iptables -A FORWARD -i %i -j ACCEPT; iptables -t nat -A POSTROUTING -o \$(sed -n 3p /DietPi/dietpi/.network) -j MASQUERADE
PostUp = ip6tables -A FORWARD -i %i -j ACCEPT; ip6tables -t nat -A POSTROUTING -o \$(sed -n 3p /DietPi/dietpi/.network) -j MASQUERADE
PostDown = iptables -D FORWARD -i %i -j ACCEPT; iptables -t nat -D POSTROUTING -o \$(sed -n 3p /DietPi/dietpi/.network) -j MASQUERADE
PostDown = ip6tables -D FORWARD -i %i -j ACCEPT; ip6tables -t nat -D POSTROUTING -o \$(sed -n 3p /DietPi/dietpi/.network) -j MASQUERADE

# Client 1
[Peer]
PublicKey = $(<client1_publickey)
AllowedIPs = 10.9.0.2/32

# Client 2
#[Peer]
#PublicKey = XXXX
#AllowedIPs = 10.9.0.3/32
```
- This includes all IP forwarding rules in the service startup. I also added the keys above, if you created them with the same names as the guide you linked. Otherwise add manually.

The client config look exactly like ours besides, we use netmask /24 on the client as well, not 100% sure currently about the practical difference however: `Address = 10.9.0.2/24`
And as DNS nameserver it is simplest to use the WireGuard network IP itself, since this for sure will never change, regardless of local network IP: `DNS = 10.9.0.1`

Do you use a mobile client or another server or desktop?
You could run the `wg` command on server (and in case client) to check if the connection has indeed successfully established, handshake is done etc.

johnnyt83, 2019-07-21:
only "wg" does not return anything on my zero unfortunately.
I copied your server config and added my keys.

just to make sure I did not screw up that last simple bit, here is my client1.conf:

[Interface]
Address = 10.9.0.2/32
DNS = 192.168.1.4
PrivateKey = XXXX

[Peer]
PublicKey = YYYY
Endpoint = myDDNS:forwardedPort
AllowedIPs = 0.0.0.0/0, ::/0
#PersistentkeepAlive = 60

MichaIng, 2019-07-21:
@johnnyt83 
> only "wg" does not return anything on my zero unfortunately.

Hmm, at least the server values should always show up. Is the interface up actually? `ip a`

johnnyt83, 2019-07-21:
this is what it gives me: 

https://abload.de/img/wiregzuegjj9.png

MichaIng, 2019-07-21:
@johnnyt83 
Okay the interface is not loaded, thus the service was not started or failed. Please run:
```
sudo systemctl start wg-quick@wg0
journalctl -u wg-quick@wg0
```
- If the second command shows --- no entries --- try this one: `journalctl -u wg-quick`
I am not 100% sure how to journalctl handles instantiated services 😉.

johnnyt83, 2019-07-21:
even the first command did not go through positive:

https://abload.de/img/wiregzuovkw6.png

MichaIng, 2019-07-21:
@johnnyt83 
Could you paste the full line that starts with `/usr/bin/wg-quick: eval: line 248: syntax error near...`.

johnnyt83, 2019-07-21:
of course:

https://abload.de/img/wiregzu22bsj0j.png

MichaIng, 2019-07-21:
Ah found it, it's about the PreUp/PostDown scripts.

You are indeed on an DietPi system, aren't you? Because user+hostname from your screens look like a Raspbian image. DietPi comes without user `pi` and with `hostname` DietPi by default. Because the WG server config I pasted requires a DietPi file to be present.

On non DietPi, switch the following in wg0.conf:
All `$(sed -n 3p /DietPi/dietpi/.network)` by `eth0` (your main network device).

johnnyt83, 2019-07-21:
my test pi zero is a raspbian (pi hole prebuilt), yes.
that was what I was playing around with today, because I could not even find it in DietPi and the tutorial was for a simple raspi.

johnnyt83, 2019-07-21:
UPDATE: seems to work when testing local WiFi now (speed is of course less than without Wireguard), but it refuses to work on mobile network connection.

MichaIng, 2019-07-21:
@johnnyt83 
The tutorial will work on DietPi as well because the core system is Raspbian as well 😉.
Great that it works now.

For mobile connection you need to assure that UDP port 51820 is forwarded from your router to the Pi. Ah from your posts above it looks like it is, but assure that it is UDP and not (only) TCP.

Does `wg` on the server list the client and show a handshake with matching timestamp when you connect and try to access something from the client?

johnnyt83, 2019-07-21:
"Unable to access interface wg0: Operation not permitted"

Port is forwarded (UPD and TCP) and is the same as in the config file.

MichaIng, 2019-07-21:
@johnnyt83 
I guess you need root permissions to access the WireGuard CLI: `sudo wg`

johnnyt83, 2019-07-21:
looks good for local:

https://abload.de/img/wiregzu22jlkcr.png

mobile is still the last hurdle to overcome.

MichaIng, 2019-07-21:
@johnnyt83 
One thing to try on the client config:
```
Address = 10.9.0.2/24
DNS = 10.9.0.1
```
- So netmask from 32 to 24 and use WireGuard interface IP instead of local network IP.

And with Pi-hole you have a webserver installed that can be used to test general access from external network:
- Forward TCP port 80 to the Pi.
- Try to access from mobile, via mobile internet, without VPN connection of course, via browser `yourDDNS/admin`, which should be the Pi-hole admin page. If this works well, then at least from local network side the VPN requests should arrive as well.

johnnyt83, 2019-07-21:
netmask 24 seems to work fine (dunno what it does though) and DNS is fine, too. Thanks so much. Can I follow the same tutorial on DietPi?

MichaIng, 2019-07-21:
@johnnyt83 
Ah great, so there was a reason why we use netmask 24. To be true I am also not 100% sure about the practical difference.

Usually netmasks define the IP range of the network, 32 specifies one specific IP, while 24 specifies all IPs from 10.9.0.1 - 10.9.0.255, 16 would be 10.9.0.1 - 10.9.255.255 etc, so again one variable octet more etc.
Of course the client has one specific IP, like the server as well, but it seems usual that those "Address" entries should contain the netmask the defines the whole IP range of the WireGuard network.

johnnyt83, 2019-07-21:
thanks for the help! whenever I need a secure connection now I can enable this "free" VPN to my home network. astonishingly the pi zero peaks at around 25 mbit/s which is fast enough for my use cases.

MichaIng, 2019-07-21:
@johnnyt83 
> astonishingly the pi zero peaks at around 25 mbit/s which is fast enough for my use cases.

Jep I guess mobile internet will be the bottle neck in most cases.

Okay I mark this as closed then. As you proved the source build to work quite well, and as this aside from binaries seem to create systemd unit (service file) and config dirs etc identical to the Debian package, we could actually add this to DietPi-Software as well.

How long did the source build take (running the `make` command)?

johnnyt83, 2019-07-21:
3 minutes, I guess. would be great to find it in the DietPi software list, soon.

MichaIng, 2019-07-21:
@johnnyt83 
Wow that is faster than I though. Proves that WireGuard holds its promises to be small and efficient code-wise.

So
```
apt install raspberrypi-kernel-headers libelf-dev libmnl-dev build-essential git
git clone https://git.zx2c4.com/WireGuard
cd WireGuard/
cd src/
make
make install
```
worked without error?
We would skip Git and instead download the code manually (`wget https://git.zx2c4.com/WireGuard/snapshot/WireGuard-master.tar.xz`) to tmp dir and remove sources after install.

Ah the steps match official instructions: https://www.wireguard.com/install/
- Besides `pkg-conf` mentioned there as additional tool chain package 🤔.

johnnyt83, 2019-07-21:
will try your mini-tutorial tomorrow on the DietPi.

johnnyt83, 2019-07-22:
btw: is it possible to get my key files to the new DietPi so that I can have the same configuration there?

best wishes!

MichaIng, 2019-07-22:
@johnnyt83 
Jep, you can simply copy `/etc/wireguard` over to the new system and skip the related key creation and config steps.

johnnyt83, 2019-07-22:
that would be great. I already tried via FTP, but it seems that these folders are protected. how would you do it?

johnnyt83, 2019-07-22:
"sudo su -c /usr/lib/openssh/sftp-server" did the trick on WinSCP!

MichaIng, 2019-07-22:
@johnnyt83 
Otherwise copying to USB (flash) drive? However both works. Jep root permissions required to access system dirs.

johnnyt83, 2019-07-22:
I have the wireguard folder now from my old Pi and try to get it onto DietPi. so I installed ProFTP.

but even after a reboot I cannot connect via WInSCP to put the folder in the right place. It gives me this:
https://abload.de/img/errorwireguardfolder8vk86.png
trying to login with root/dietpi on SFTP.

johnnyt83, 2019-07-22:
I only managed to copy it here:

https://abload.de/img/wireguard222pxkbj.png

johnnyt83, 2019-07-22:
I just rebuilt everything from scratch. maybe this is the culprit here:

https://abload.de/img/depmodn8j3x.png

MichaIng, 2019-07-24:
@johnnyt83 
Perhaps there has been some interference with a recent firmware update.
Please try the following:
```
G_AGI raspberrypi-kernel raspberrypi-kernel-headers libmnl-dev libelf-dev build-essential pkg-config
G_AGDUG
reboot
# After reboot, so new kernel 4.19.58 is loaded and matches the headers
cd /etc/wireguard/WireGuard/src
make
make install
```

johnnyt83, 2019-07-25:
"root@DietPi:/etc/wireguard/WireGuard/src# make install
  INSTALL /etc/wireguard/WireGuard/src/wireguard.ko
  DEPMOD  4.19.57+
Warning: modules_install: missing 'System.map' file. Skipping depmod.
depmod -a"

this is what I got in return :(

johnnyt83, 2019-07-25:
and of course all the other things that I wanted to put in "spoiler" tags that I cannot find here:

"
removed '/usr/bin/wg'
'wg' -> '/usr/bin/wg'
removed '/usr/share/man/man8/wg.8'
'man/wg.8' -> '/usr/share/man/man8/wg.8'
removed '/usr/share/bash-completion/completions/wg'
'completion/wg.bash-completion' -> '/usr/share/bash-completion/completions/wg'
removed '/usr/bin/wg-quick'
'wg-quick/linux.bash' -> '/usr/bin/wg-quick'
removed '/usr/share/man/man8/wg-quick.8'
'man/wg-quick.8' -> '/usr/share/man/man8/wg-quick.8'
removed '/usr/share/bash-completion/completions/wg-quick'
'completion/wg-quick.bash-completion' -> '/usr/share/bash-completion/completions/wg-quick'
removed '/lib/systemd/system/wg-quick@.service'
'systemd/wg-quick@.service' -> '/lib/systemd/system/wg-quick@.service'
"

MichaIng, 2019-07-25:
@johnnyt83 
You are still on old kernel 4.19.57, or at least WireGuard tries to compile against it.

`uname -a` shows 4.19.57 or 4.19.58?

Perhaps also some configs have been already created that do not match anymore, so you might need to remove the WireGuard sources and re-download.

Otherwise please reassure that you did the mentioned G_AG* commands and a reboot afterwards.

johnnyt83, 2019-07-25:
root@DietPi:~# uname -a
Linux DietPi 4.19.57+ #1244 Thu Jul 4 18:42:50 BST 2019 armv6l GNU/Linux

But I set it up/ran an update just a few days ago. Everything was up to date.

johnnyt83, 2019-07-25:
when I start from scratch now with dietPi, is wireguard in the software list now?

MichaIng, 2019-07-25:
The new update was also a few days ago. It is necessary that active kernel and installed headers match. 

Nope it's not yet in the install list. If it works I might be able to add it the next days to dev code.

johnnyt83, 2019-07-25:
I would really like to fix that, but it seems I am up to date:

https://abload.de/img/dietpijvkfs.png

MichaIng, 2019-07-25:
Its not about dietpi version but about the kernel version. Please run the commands I mentioned above to update the APT packages which include kernel.

johnnyt83, 2019-07-25:
did it again (all the commands above), one by one was executed OK. and I am stuck on 
"root@DietPi:/etc/wireguard/WireGuard/src# make install
  INSTALL /etc/wireguard/WireGuard/src/wireguard.ko
  DEPMOD  4.19.57+
Warning: modules_install: missing 'System.map' file. Skipping depmod.
depmod -a"

really weird.

johnnyt83, 2019-07-25:
can I install a remote tool on my dietPi so you could login for yourself?

MichaIng, 2019-07-25:
@johnnyt83
Yeah probably easier. I have two other ideas:
- Removing and re-downloading the sources to remove some probably created configs that do not match
- Creating a symlink from `/lib/modules/<version>/build` to `/lib/modules/<version>/source` or the other way round. I recognised that on some systems only one is present while build tools search for kernel sources in the other one. Although error message should be different then.

Yeah if you want to grant remote access, SSH would be easiest, as I am not too familiar with remot3.it or such.
Or do you have a desktop installed? Then VNC would be an alternative that allows you to track what I am doing.
In case of SSH, when opening it to www, assure that root and dietpi users have a strong password and at best do not forward port 22 to 22 but some random port to 22, which is not usual target of brute-force attempts.

Send login credentials to micha@dietpi.com

johnnyt83, 2019-07-25:
Yes, SSH sounds fine. So I only need to change the password for users root and dietpi and forward a random port to my dietpi, right?

MichaIng, 2019-07-25:
@johnnyt83 
Jep, you can change both passwords (for dietpi and root users together) via `dietpi-config` > `Security Options`, which as well allows you to change the default password for (most) new software installs.

MichaIng, 2019-07-25:
Okay compilation from source works well, following the Debian steps: https://www.wireguard.com/install/
Of course raspberrypi-kernel + raspberrypi-kernel-headers instead of linux-image-*.

Sources should be kept on the machine. Since no DKMS is installed, after kernel updates WireGuard needs to be recompiled:
```
cd /path/to/WireGuard/src
make
make install
```
- Where to store?
- `/etc/wireguard/WireGuard/src` is not thee classic location for this, on the other hand then everything is in one place.
- Else `/opt` or `/usr/local/` or `/mnt/dietpi_userdata` to have it on external drive easily?

Another idea:
- Create `/etc/kernel/postinst.d/dietpi-update` on install which contains the WireGuard module compile steps: `dpkg-reconfigure wireguard-dkms` on RPi2/3/4 and `cd ..; make; make install` on RPi1/Zero
- Currently on all most SBCs one needs to manually re-compile the WireGuard module after every kernel upgrade.

pqhf5kd, 2019-08-09:
I'm also trying to install WireGuard on a pi2 and while I can follow basic CLI instructions, I'm no expert. Can you just clarify this please:

> ### Notes to self
> Okay compilation from source works well, following the Debian steps: https://www.wireguard.com/install/
> **Of course raspberrypi-kernel + raspberrypi-kernel-headers instead of linux-image-*.**
>

MichaIng, 2019-08-10:
@pqhf5kd 
Follow these steps: https://github.com/MichaIng/DietPi/issues/2997#issuecomment-513595547

> raspberrypi-kernel raspberrypi-kernel-headers

These are the APT packages for the RPi kernel and it's sources/headers. required to build kernel modules, like WireGuard is one. The official build instructions assume those be named like `linux-image-*` and `linux-headers-*` which is true for most systems but not for RPi.

qymab, 2019-09-03:
> @johnnyt83
> Perhaps there has been some interference with a recent firmware update, so that the kernel headers do not match the currently loaded kernel anymore.
> Please try the following:
> 
> ```
> G_AGI raspberrypi-kernel raspberrypi-kernel-headers libmnl-dev libelf-dev build-essential pkg-config
> G_AGDUG
> reboot
> # After reboot, so new kernel 4.19.58 is loaded and matches the headers
> cd /etc/wireguard/WireGuard/src
> make
> make install
> ```

Using this working great in my RPI 1, thanks.

MichaIng, 2019-09-21:
@johnnyt83 
Great news, the source build for ARMv6 RPis (1+Zero) is not required anymore. WireGuard packages have been added to the Bullseye (new testing) repo, thus are available from Raspbian as well. I switched the source repo to Bullseye on all devices, since this can be considered more stable then Sid, and enabled support for RPi ARMv6 by this: https://github.com/MichaIng/DietPi/commit/e62ba9f6399b151b6dbeeba786d042d3e13a73db

Another minor enhancement: Now, when kernel packages are upgraded on RPi, the WireGuard kernel module is rebuild automatically as well.

closed 2019-10-15 by MichaIng