MichaIng/DietPi #2999: Update Gogs version to latest v.11

LazyLama, 2019-07-22:
Hi All,
  I was wondering if you all could update the version of Gogs. Currently you have v0.9.141 which is pretty old as it was released back on Feb 11, 2017 [https://github.com/gogs/gogs/releases/tag/v0.9.141].

MichaIng, 2019-07-22:
@LazyLama 
Many thanks for your report.

At least for RPi2+ and x86_64 we could use the pre-compiled binaries. For RPi1/Zero those do not seem to work (https://github.com/gogs/gogs/issues/3415), thus someone needs to compile it. I lack an ARMv6 RPi, so am not able to do so, without some nasty cross-compiling.

MichaIng, 2019-07-25:
PR up: https://github.com/MichaIng/DietPi/pull/3006

MichaIng, 2019-07-26:
@LazyLama 
Okay PR is done. If you want to test/update:
```
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=gogs' /DietPi/dietpi.txt
dietpi-update 1
dietpi-software reinstall 49
```
- Currently only updates ARMv7 and x86_64, since one needs to compile for ARMv6 first. Which arch you use?
- On ARMv7 we install the pre-compiled RPi2 binaries, which work fine on RPi3 and should work on all ARMv7 systems. However test to be failsafe is outstanding. If an issue occurs, reverting is easy.
- Reinstall will preserve existing settings and repos of course.

Shall we include the Gogs reinstall (update) in `dietpi-update` automatically, or leave existing installs untouched, what you think?
Regardless I will add update infos to our online docs, once v6.26 is released.

LazyLama, 2019-07-31:
Hi MichaIng,
 I ran the provided update commands and got a few errors. Everything seemed to install fine tho. Just posting them here in case you need them. I'll do more testing, but so far so good. Oh, and TY.
 
 Device model : RPi 3 Model B (armv7l)
 
 DietPi-Pre-patch
─────────────────────────────────────────────────────
 Mode: Applying pre-patches

./pre-patch_file: line 141: ((: < 26 : syntax error: operand expected (error token is "< 26 ")
[  OK  ] Successfully applied pre-patches

[image]

DietPi-Update
─────────────────────────────────────────────────────
 Mode: Running incremental patching

[ INFO ] DietPi-Update | Current version : v6.25.3 (branch: MichaIng/master)
[ INFO ] DietPi-Update | Latest version  : v6.26.0 (branch: MichaIng/dev)
[ INFO ] DietPi-Update | Patching 6.25 to 6.26
[ SUB2 ] DietPi-Set_hardware > rpi-camera (disable)
[  OK  ] DietPi-Set_hardware | Desired setting in /DietPi/config.txt was already set: start_x=0
[  OK  ] rpi-camera disable | Completed
[  OK  ] DietPi-Patchfile | Added setting dtoverlay=disable-bt to end of file /DietPi/config.txt
DTOVERLAY[error]: Failed to open '/boot/overlays/disable-bt.dtbo'
* Failed to read '/boot/overlays/disable-bt.dtbo'
[  OK  ] DietPi-Update | Patching to v6.26.0 completed 

[image]

MichaIng, 2019-07-31:
@LazyLama 
Many thanks for reporting, jep the first is a little typo from yesterday night, I did not yet test. Just fixed: https://github.com/MichaIng/DietPi/commit/79ab1d9bab4062d277c06e2b24da5f7150d61aaf

The second is strange indeed, since even my RPi2 without actual Bluetooth, is able to load this overlay. It is to disable onboard Bluetooth, which we did by blacklisting the related modules only, previously, but not add the related (relatively new) dtoverlay that should further reduce hardware load and releases serial device (ttyAMA0).

First of all the overlay exist on your RPi, right? `dtoverlay -h disable-bt`
Kernel is up-to-date? `uname -a`

LazyLama, 2019-07-31:
[image]

MichaIng, 2019-07-31:
@LazyLama
Ah your loaded kernel is not yet up to date. I guess the update upgraded the packages (or any previous APT upgrade) but you did not yet reboot?

Okay that is a general minor issue when we load overlays during updates as the loaded kernel might not match the installed one. However this is not a real issue since those are loaded on next boot then via added config.txt entry.
Perhaps we should hide the possible error to not confuse users or compare loaded and installed kernel versions before attempting.

LazyLama, 2019-07-31:
I just rebooted, now I see the new kernel

[image]

LazyLama, 2019-07-31:
I'd say you can close this one out. Gogs is working as it should. Thanks again.

MichaIng, 2019-07-31:
@LazyLama
Good about Gogs.

Ah I know now why the overlay does not load. It just got renamed fron rpi3-disable-bt to disable-bt since it applies to RPi4 now as well. But that name change was part of firmware with kernel 4.19.58 which is available on Buster only currently. So it will not work on Stretch images until the next firmware release. However we will leave it inside for now as it doesn't hurt until then.

PR #3006 (2019-07-30): DietPi-Software | Gogs: Install latest pre-compiled binaries from GitHub for ARMv7 + x86_64
**Status**: Testing
- [x] DietPi-Patch | Remove `/etc/__MACOSX` if present, coming from our archives
- [x] Changelog
- [ ] Testing on non-RPi ARMv7 to be failsafe
- [x] Test reinstall from pre-v6.26 instance to assure InnoDB 4byte support does not break existing database
- [ ] Compile and use ARMv6 binaries

Fixes #2999

**Commit list/description**:
+ DietPi-Software | Gogs: Install latest pre-compiled binaries from GitHub for ARMv7 + x86_64. For ARMv6 (RPi1/Zero) no binaries are available yet
+ DietPi- [... cut]
closed 2019-10-15 by MichaIng