MichaIng/DietPi #3001: Small collaboration for another project [Arthemis]

Fray2000, 2019-07-22:
Hi,
Just wanted to know if i could use and advertise your software for this project :

The Arthemis project is ment for students or persons with limited possibilities to deploy home informatic infrastructure and modernise a home on the cheap.

It's main objectives are to have a compact home server that can offer many features in a beautiful packaging. The tageted use is to have one computer that does tasks that would be usually managed by many different devices such as a router or a a NAS. It's use of mainstream components make the Arthemis system very effective, afforable and very flexible. Many devices can be added to the overall system. It's open source source makes it's improvements totally free and the server have a much longer lifetime.

The use of a Raspberry pi makes it a perfect fit for such use. If possible, the suport of other boards will be added with time. The removal of the SD Card by using the USB boot feature added in the latest Raspberry's makes the computer much more durable since more durable memories than SD cards can be used for running the system. 

It's low power consomption and adjustable CPU frequency makes it able to enter economy mode that will be used for the power efficiency of the entirety system. The system is also more ecologycal thanks to the use of only one board and it's flexibility promisses a longer lifetime than comperable devices.

Some modifications will be done on the original Dietpi software to make it more flexible and programmer friendly without compromising it's functions.

The projects offers: 

- A 3D printed base that supports 2.5 drives and the Raspberry Pi 2 / 3 (all versions included with B and B+ varients) 

- software ?

https://github.com/Fray2000/Arthemis

Sorry if it was a bit long but thank you !

MichaIng, 2019-07-24:
@Fray2000 
Many thanks for your question and idea.

What you describe sounds exactly what DietPi aims for, running multiple software/tasks on a single machine instead of having multiple single-purpose devices. So this should fit well.
So yeah you are of course free to use DietPi sources as long as you keep `/boot/dietpi-README.md`, which contains the GPLv2 license info as well.

> Some modifications will be done on the original Dietpi software to make it more flexible and programmer friendly without compromising it's functions.

As we aim to make DietPi more flexible and compatible as well, I would be happy if you'd contribute your ideas directly to the DietPi project, where applicable. I'd support you as much as I can and it would probably reduce your code maintenance/updates when DietPi changes.

If applicable, if Arthemis will include some own software, tools, GUI or such, we could also include it into default DietPi-Software code as install option. E.g. Amiberry is available via dedicated pre-installed image, tailored for Amiga Emulator use, but one can install it on a raw DietPi system as well.

MichaIng, 2021-05-02:
Looping through old issues, I'm marking this as closed. Feel free to reopen when Arthemis is re-activated and/or I can be of any help.

closed 2021-05-02 by MichaIng