MichaIng/DietPi #3002: Failing migration to RPi4

JJGO, 2019-07-23:
#### Required Information
- DietPi version | 6.25.3
- Distro version | stretch
- Kernel version | 4.19.57-v7+
- SBC device | RPi 3b+/4

I have been running an RPi 3b+ with DietPi and want to migrate it to RPi 4.  

To transfer the data I did the following

1. Made a dietpi-backup to an USB in the 3b+
2. Installed the experimental image from [here](https://github.com/MichaIng/DietPi/issues/2935#issuecomment-511226948) to the RPi 4
3. Restored using dietpi-backup from the USB drive.

However it fails (which could be the expected behaviour, I do not really now). What is the easiest way to migrate to the RPi 4 from a given installation?

MichaIng, 2019-07-24:
@JJGO 
Many thanks for your report.

Raspbian Stretch is not compatible with RPi4. So you cannot restore the Stretch backup, which turns the Buster system back into Stretch.
You could "apt full-upgrade" the Stretch image to Buster, but such upgrades can cause follow-up issues.

I suggest you copy back only those parts of the backup, that you really need, e.g. `/mnt/dietpi_userdata` content and other single config files based on what you have installed.

JJGO, 2019-07-24:
Thanks for the reply

If I were to upgrade to using `apt full-upgrade` the RPi 3, solve the issues that may appear and then restore that backup in the RPi4, could it work?

I agree a fresh install and config is preferable, but given the amount of installed programs it would be quite time consuming.

MichaIng, 2019-07-24:
@JJGO 
Probably. Actually easier would be the test to simply plug the SDcard into the RPi4 after the upgrade (assuming you have the backup).
Also after the upgrade and reboot carefully check that all software is still working. We do distro version specific install steps for many software titles, some of them have own Buster APT repos as well and such (check /etc/apt/sources.list.d/ for additional lists).

Which software titles do you have installed btw? I could do a quick review about which might be adjusted to match Buster-specific install.

JJGO, 2019-07-25:
This is the output I get from `sudo dietpi-software list | grep " =2 " | cut -d " " -f6`

    openssh
    samba
    ffmpeg:
    java:
    build-essentials:
    git
    vim:
    plex
    syncthing:
    rpi-monitor:
    fail2ban:
    lesp:
    ubooquity:
    nginx:
    sqlite:
    php:
    certbot:
    samba:
    rsyslog:
    dietpi-ramlog:
    openssh
    rtorrent:
    nfs
    tautulli:
    home
    unrar:
    wireguard:

MichaIng, 2020-01-01:
@JJGO 
Sorry that I forgot about this issue. From what I see, Buster upgrade should work, but I would definitely do a full system backup before head. I mark this issue as closed. Feel free to reopen if required.

closed 2020-01-01 by MichaIng