MichaIng/DietPi #3003: IP Forwarding not working?

johnnyt83, 2019-07-23:
Although it IS enabled in sysctl.conf

https://abload.de/img/ipforwarding91jgs.png

it tells me 
"root@DietPi:~# sysctl net.ipv4.ip_forward
net.ipv4.ip_forward = 0
"
after reboot. Is this a bug?

MichaIng, 2019-07-24:
@johnnyt83 
Many thanks for your report.

There have been reports about `/etc/sysctl.conf` being ignored since a symlink from `/etc/sysctl.d`/ was missing.
Please assure that this is the case:
```
2019-07-24 14:22:37 root@micha:/tmp# l /etc/sysctl.d/99-sysctl.conf
lrwxrwxrwx 1 root root 14 Jun  6 15:45 /etc/sysctl.d/99-sysctl.conf -> ../sysctl.conf
```
Otherwise:
```
ln -s ../sysctl.conf /etc/sysctl.d/99-sysctl.conf
```

But generally I would never use `/etc/sysctl.conf`. Just leave this untouched, so APT can update defaults in case. Instead create an own drop-in config file, e.g. `/etc/sysctl.d/ip_forwarding.conf` and put the setting there.

And, as long as you configure WireGuard with the `wg0.conf` I provided, you don't need the above setting at all. The `wg0.conf` enables IP forwarding targeted for the required network interfaces as PostUp commands.

johnnyt83, 2019-07-25:
I hope you can do something with this info here:

"/src# ls -Al /etc/sysctl.d/99-sysctl.conf
-rw-r--r-- 1 root root 2683 Jul 21 09:54 /etc/sysctl.d/99-sysctl.conf
root@DietPi:/etc/wireguard/WireGuard/src# /etc/sysctl.d/99-sysctl.conf -> ../sysctl.conf
bash: /etc/sysctl.d/99-sysctl.conf: Permission denied
root@DietPi:/etc/wireguard/WireGuard/src#"

MichaIng, 2019-07-25:
@johnnyt83 
`/etc/sysctl.d/99-sysctl.conf` is a file, that is strange... No idea what created this, especially since we recreated the symlink some version ago as already some other users reported it... Must be some faulty program installer.

Please do the following:
```
mv /etc/sysctl.d/99-sysctl.conf /etc/sysctl.conf
ln -s ../sysctl.conf /etc/sysctl.d/99-sysctl.conf
```
Then you can edit `/etc/sysctl.conf` and those will be loaded after reboot or sysctl reloading.
But it is as said not required with our WireGuard config, as we tied the settings to the WireGuard start there. The advantage is that nothing is forwarded if WireGuard has not been started and the other way round, if something resets the settings, WireGuard start will re-enable forwarding automatically.

Something I can fix as well when dealing with the WireGuard build process from the other issue 😉.

Joulinar, 2020-08-03:
@MichaIng 
sorry for coming back on this after such a long time but I noticed some strange behaviour regarding thesylink.

1. checked the image, the link was there :heavy_check_mark:
2. flashed this image to SD card and mounted it on a different DietPi box, the link was there :heavy_check_mark:
3. pluged this SD card into a RPi and perform initial setup, the link was missing :x:

not sure what happen on the initial run but the link got removed

MichaIng, 2020-08-05:
Hmm, I'll also try this out, probably some bad package upgrade. I was always wondering why users report this (rarely but) regularly.

Current stable (32-bit) RPi image?

Joulinar, 2020-08-05:
Both, 32bit as well as 64bit

**From my current 64bit**

```
root@DietPi3:/etc/sysctl.d# ls -la
total 32
drwxr-xr-x  2 root root 4096 Aug  5 13:28 .
drwxr-xr-x 63 root root 4096 Aug  5 18:59 ..
-rw-r--r--  1 root root   51 Aug  5 13:26 98-rpi.conf
-rw-r--r--  1 root root 2351 Aug  5 13:26 99-sysctl.conf
-rw-r--r--  1 root root  639 May 31  2018 README.sysctl
-rw-r--r--  1 root root   70 Aug  5 13:26 dietpi-disable_ipv6.conf
-rw-r--r--  1 root root  220 Aug  5 13:26 dietpi.conf
-rw-r--r--  1 root root  324 Aug  5 13:26 protect-links.conf
root@DietPi3:/etc/sysctl.d#
```

Which is strange, as my current RPi4B prod system has the link still available. But I did not update for a while. Still running `4.19.118-v7l+`
```
root@DietPi4:/etc/sysctl.d# ls -la
#insgesamt 28
drwxr-xr-x  2 root root 4096 Jun  8 20:54 .
drwxr-xr-x 78 root root 4096 Aug  3 21:25 ..
-rw-r--r--  1 root root   51 Sep 26  2019 98-rpi.conf
lrwxrwxrwx  1 root root   14 Mai 11 09:06 99-sysctl.conf -> ../sysctl.conf
-rw-r--r--  1 root root  220 Jul  7 23:42 dietpi.conf
-rw-r--r--  1 root root   70 Sep 26  2019 dietpi-disable_ipv6.conf
-rw-r--r--  1 root root  324 Sep 26  2019 protect-links.conf
-rw-r--r--  1 root root  639 Mai 31  2018 README.sysctl
root@DietPi4:/etc/sysctl.d#
```

MichaIng, 2020-08-05:
Confirmed, it is already the case right after first login, without DietPi-Update or DietPi-Software.

MichaIng, 2020-08-05:
Fixed: https://github.com/MichaIng/DietPi/blob/dev/dietpi/func/dietpi-set_hardware

closed 2020-08-31 by MichaIng