MichaIng/DietPi #3004: Openauto for rpi3

krishnasen2401, 2019-07-24:
Please openauto for raspberry pi 3 for automatic installation from software list

Project link
https://github.com/f1xpl/openauto/blob/development/Readme.md

MichaIng, 2019-07-24:
@krishnasen2401 
Many thanks for your request.

Probably something I should ask the devs then, but perhaps you know:
- If it is already or planned to become RPi4 + Raspbian Buster compatible, as we soon switch our RPi image the Buster basis.
- If it is possible to configure it using plain ALSA audio configuration instead of relying on PulseAudio. DietPi configures and uses plain ALSA and PulseAudio will conflict with our configs and some software installs.

MichaIng, 2021-09-24:
We decided to cleanup open issues a bit and close those which will obviously not be addressed soon. If anyone wants to start working on implementing this software, please feel free to to a start, and we'll help where we can.

Here you find instructions about how to implement a new software title to `dietpi-software`: https://github.com/MichaIng/DietPi/wiki/How-to-add-a-new-software-title

closed 2021-09-24 by MichaIng