MichaIng/DietPi #3008: Raspi Load manager flawed?!

johnnyt83, 2019-07-26:
Hi everybody,
although there is no visible load, the display jumps to values that are not comprehensible.
Example: At the end the current load is low, but the 1min display jumps from 0.19 to 0.26 just like that?

For further investigation I recorded a video here. It might be possible (very likely) this is not even DietPi related and I am only missing something here.
I really appreciate any help.
Take a look if you want:     https://youtu.be/SSTttpiuB0w

MichaIng, 2019-07-26:
@johnnyt83 
Many thanks for your report.

I think the default tick rate (2 seconds) is too low to see every e.g. DNS (Pi-hole), webserver and PHP request properly raising the current load.
Please try `htop -d 5` =>0.5 seconds tick rate, perhaps you will see more short-time peaks of certain processes, especially Pi-hole in your care I guess.

MichaIng, 2019-08-11:
I mark this issue as closed. Is as well a lower tick rate does not show some CPU usage peaks to explain the minutely average, then please forward this issue to htop dev: https://github.com/hishamhm/htop/issues

closed 2019-08-11 by MichaIng