MichaIng/DietPi #3010: Concurrent execution of DietPi-Run_NTPD detected

Sopor, 2019-07-26:
I'm running nginx on DietPi but all the log files suddenly change to be 0 bytes. I also noticed that all the log files for the system also are 0 bytes.

I then run `dietpi-software` (to change the log issue), but it did have some problems with the NTP so i follow the instructions, but now i get this after selecting my gateway as an NTP server:

[image]

Could you tell me how i make the log files to be persistent? This seems to not be an easy task to do :(

Sopor, 2019-07-26:
I manage to change it on another DietPi and when it work as it should it wasn't so hard :)

MichaIng, 2019-07-26:
@Sopor 
Jep logfiles are cleared every every hour due to RAMlog, otherwise RAM would fill up unlimited.
So you need to disable/uninstall DietPi-RAMlog via `dietpi-software`. Choose "Custom", as rsyslog is actually a very obsolete thing on modern Debian. systemd-journald is active anyway which stores all system logs well sortable and readable via `journalctl` in RAM. rsyslog just doubles this, making all syslogs a more-than-double effort.
I will change this to remove the rsyslog option completely and instead allow to make journald boot persistent, which is quite easy as well.

About the NTPD error:
Jep I think I know what causes this loop:
- When you run into a time sync error, you can open `dietpi-config` to change settings.
- But when you change time sync settings, the same script is called again.
- What we need to do is from within dietpi-config, check if the script already runs and in case skip the additional call.

But in your particular case you managed to solve everything now?

Sopor, 2019-07-26:
I still not manage to get it to work
```
[ INFO ] DietPi-Run_NTPD | NTPD: Waiting for completion of systemd-timesyncd (10/10)
```

I run pfsense as a router and i have an NTP server running... I'm not sure what it want?

MichaIng, 2019-07-26:
@Sopor 
Please paste: `journalctl -u systemd-timesyncd`

Sopor, 2019-07-27:
```
$ sudo journalctl -u systemd-timesyncd
-- Logs begin at Thu 2016-11-03 18:16:42 CET, end at Sat 2019-07-27 14:23:13 CEST. --
Jun 08 00:17:14 DietPi systemd[1]: Starting Network Time Synchronization...
Jun 08 00:17:14 DietPi systemd[1]: Started Network Time Synchronization.
Jun 08 00:23:23 DietPi systemd-timesyncd[662]: Synchronized to time server 192.36.143.130:123 (0.debian.pool.ntp.org).
Jun 08 00:23:24 DietPi systemd[1]: Stopping Network Time Synchronization...
Jun 08 00:23:24 DietPi systemd[1]: Stopped Network Time Synchronization.
```

MichaIng, 2019-07-27:
@Sopor 
> Jun 08 00:17:14 DietPi systemd[1]: Started Network Time Synchronization.
Jun 08 00:23:23 DietPi systemd-timesyncd[662]: Synchronized to time server 192.36.143.130:123 (0.debian.pool.ntp.org).

It looks like it took 6 minutes to sync time...
It should be ~1 second.
It syncs with `0.debian.pool.ntp.org` currently, which is default on Debian but not optional. "Gateway" (local router) should be best in very most cases.

All instances of run_ntpd are killed, right? `killall -w run_ntpd`

> and i have an NTP server running

What you mean by this? An NTP server in the router of on the DietPi machine? Or the "ntp" package installed?

Sopor, 2019-07-27:
Instead of syncing to `time.windows.com` i sync my computers to my pfsense.

MichaIng, 2019-07-27:
@Sopor 
Yeah this makes sense, for the DietPi system as well. When you choose "Gateway" in dietpi-config > Network Options: Misc > NTP mirror it should do this as well, if the router is the default internet gateway for the network.
Switching this and then running `journalctl -u systemd-timesyncd` should show:
```
systemd-timesyncd[662]: Synchronized to time server <ip_of_your_router>
```
as one of the last lines then.

MichaIng, 2019-09-03:
PR up: https://github.com/MichaIng/DietPi/pull/3095
Some other enhancements are and need to be done with this, however concurrent execution from both sides are resolved.

closed 2019-10-15 by MichaIng

codebeat-nl, 2021-05-16:
@MichaIng Thanks! `journalctl -u systemd-timesyncd` did the trick.

MichaIng, 2021-05-16:
`journalctl -u systemd-timesyncd` only prints the services logs but does not change something 😉. So I guess the time it took to search and run the command was the solution, as the concurrent script finished in the meantime 🙂.

PR #3095 (2019-09-05): Time sync | Avoid concurrent DietPi-Run_NTPD executions
**Status**: Ready
- [x] DietPi-Set_software
- [x] DietPi-Run_NTPD
- [x] Changelog

Fixes: #3010

**Commit list/description**:
+ DietPi-Config | Do not call DietPi-Run_NTPD after selecting new time sync mirror or mode, if it is already running, to avoid concurrent execution, if DietPi-Config was originally called from DietPi-Run_NTPD on error.
+ DietPi-Config | Call DietPi-Run_NTPD non-interactively after selecting new time sync mirror or mode, to avoid its internal menu on error, which would allow to call a concurrent DietPi-Config or ove [... cut]