MichaIng/DietPi #3011: Set APT cache back as it was before

Sopor, 2019-07-26:
Now when you have disabled the APT cache i wondering how i set it back as it was before?

APT cache    : [Disabled]
APT lists    : [On disk]
APT archives : [On disk]

I have no idea what i should choose :(

MichaIng, 2019-07-26:
@Sopor 
Set APT cache to disk is the Debian/Raspbian default. Enabling the cache speeds up certain APT calls (the reading package lists... part) indeed, but every apt update will write (and keep) ~60 MiB data to the disk, which is quite heavy for SDcards.
An alternative is to set it to RAM, which makes APT calls even faster and does not cause any SDcard writes. But it consumes 60 MiB RAM then, which can be an issue on low RAM devices or when you have many services up.

Ah there is the "Reset" option which recovers system defaults, glad I added this, should be self-explaining?
This was also state on DietPi before v6.25. I just felt that SDcard corruption is really one of the most usual case for very critical errors and data loss on SBCs. And since one usually runs APT update before every package install, this happens quite often.

Perhaps we should also add a "(default)" suffix to each selection menu for the system default value (which is always "To disk"), what you think?

Sopor, 2019-07-27:
Yes, that would be nice!

MichaIng, 2019-07-27:
@Sopor 
Done: https://github.com/MichaIng/DietPi/commit/6d8d6bbce6aab800c2ea9fafc8fdaf5f6add199b
I mark this issue as closed. Feel free to reopen if required or if there are any other clarifications/enhancements that we can do to the APT cache options.

Sopor, 2019-07-27:
The APT cache was set to Disabled and i choose some of the other two options but now this is gone and it will not show that option again. If i choose Reset nothing happens

[image]

Sopor, 2019-07-27:
These settings are very confusing...

MichaIng, 2019-07-27:
@Sopor 
Okay I will add a Wiki page an/or a Help entry to the menu for further explanation. Basically you can leave the defaults if unsure.

#### APT lists
- Plain text files that contain all information about APT/.deb packages that are available in your APT repositories.
- Default disk location: `/var/lib/apt/lists`
- RAM location: `/tmp/apt/lists`
- E.g. here is the file for Raspbian Buster "main" component: http://raspbian.raspberrypi.org/raspbian/dists/buster/main/binary-armhf/Packages
- As you can see there is information about every available package, including dependencies and such.
- These list files are required whenever you want to install any APT package via `apt-get install <package_name>` or `apt-get upgrade` etc as APT then checks if the package is actually available, the download URL and which dependencies need to be installed alongside.
- `apt-get update` downloads these files or updates them, if required. For this reason before doing any package install it is recommended to run `apt-get update`, so it is assured that APT has the correct current information available.
- As you can see the size is ~60 MiB or more, depending on which APT sources you have added to your `/etc/apt/sources.list` and `/etc/apt/sources.list.d/*.list` files. Each entry leads to one file each component (e.g. main, contrib, nonfree). E.g. these are the available components for Raspbian Buster: http://raspbian.raspberrypi.org/raspbian/dists/buster/
Each directory contains an own `Packages` (list) file (in the `binary-<architecture>` sub directory).

#### APT cache
- Binary format files of all the information that is contained in the APT list files.
- Default disk location: `/var/cache/apt/pkgcache.bin` and `/var/cache/apt/srcpkgcache.bin`
- RAM location: `/tmp/apt/pkgcache.bin` and `/tmp/apt/srcpkgcache.bin`
- These files are never required, but APT can read available package information much faster from those then from the plain text list files above. Basically APT first tries to read from cache, if available, otherwise tries to read the plain text list files, and if those are not available as well, it aborts and tells you that you need to run `apt-get update`, which then downloads fresh list files and, if enabled, compiles the two cache files from them.
- The file size is, as you can easily check, nearly the same as for the plain text list files, which makes sense as the contained information is the same, just in a different format.
- `apt-get update` will always completely rewrite those two files, if cache is enabled.

#### APT archives
- Location where packages (the `<package_name>.deb` files) are downloaded to, before being installed.
- Default disk location: `/var/cache/apt/archives`
- RAM location: `/tmp/apt/archives`
- It's these files that APT downloads and installs when you install or upgrade an APT package, here e.g. for the APT tool packages themselves: http://raspbian.raspberrypi.org/raspbian/pool/main/a/apt/

#### To choose the options
- APT cache is as said never required, just speeds up certain APT command calls.
  - On disk this leads to additional 60+ MiB disk writes on every `apt-get update` call, thus nearly on every single APT install that you do, which is quite heavy for SDcards and reduces their lifetime.
  - In RAM this means 60+ MiB RAM are blocked, which is quite much on low RAM devices, e.g. on 256 MiB RAM RPi then at least a quarter is simply not available.
  - Both above is not optimal in certain situations (system on SDcard and/or low RAM device), so we chose to disable the cache by default on DietPi images. If you either have your system on a HDD/SSD or you have enough free RAM, you enable it or move it to RAM.
- APT lists are always required.
  - On disk those are not fully rewritten on every `apt-get update`, but just the individual files for which newer version are available online. So it is not as critical to have on disk as the cache.
  - On RAM it again leads to 60+ MiB RAM space being blocked. Additionally after reboot, where RAM content of course is lost, one needs to run `apt-get update` once to re-download them. This is however done automatically by DietPi scripts.
  - Since the light comfort loss when being in RAM and the disk writes are not that large, we leave them by default on disk. If you have enough free RAM space, you can however move them to RAM, which speeds up APT calls when cache is disabled.
- APT archives are by default on disk **and option is hidden**, if the cache is on disk as well. Since if you choose to keep the cache enabled and on disk, you most likely don't want that much RAM being used. And the downloaded packages/archives can consume very much RAM if you do larger installs. So this should be usually kept on disk, as long as you are not very sure that you have enough free RAM space and/or do not plan to do larger software installs via APT.

Having anything in RAM of course speeds up any APT actions since both, reads and writes, to RAM are much faster. However I would not move anything to RAM as long as there is not ~50% free on normal operation. Peak times or certain cron jobs can increase RAM usage temporarily an I personally would better be sure to avoid regular swap file usage (system writing RAM content to disk, if RAM is getting full) during those peak times.

`apt-get clean` btw can be used to remove the cache files (until next `apt-get update` recreates it) and all downloaded packages/archives that are not required anymore. The list files are not touched by this.

I hope this clarifies things?

Sopor, 2019-07-27:
If i disable APT cache i suddenly have three to choose from

[image]

So APT archives will go away if i choose APT cache on disk?
I would like it to be there but grayed out or something because as i mentioned earlier it will confuse (at least me) the user when one option disappear when i change another option.

MichaIng, 2019-07-28:
> So APT archives will go away if i choose APT cache on disk?

Exactly

> I would like it to be there but grayed out or something because as i mentioned earlier it will confuse (at least me) the user when one option disappear when i change another option.

It is mentioned at the top "Resets to disk if cache is moved to disk", however I agree that this can be confusing.
Menu entries in whiptail cannot be greyed out or something, but we can add an info either behind the "[On disk]" text or to the selection menu itself, hiding the "To RAM" option then.
I will play around with this.
Many thanks for your feedback btw 👍.

AtkLordOverAll, 2019-08-13:
@MichaIng 

Want me to look into this? Can't be too bad to add some extra text to some menu options

Do we want to show both Raspbian and DietPi defaults? I think it'd make sense to.

Are there are other dialogues other than APT caching that we should consider?

MichaIng, 2019-09-16:
@AtkLordOverAll 
Ah sorry I have overseen your post here.

> Can't be too bad to add some extra text to some menu options

Jep, however the defaults we applied with v6.25 are fine in very most cases. The understanding about those settings needs to be good when touching them.

> Do we want to show both Raspbian and DietPi defaults? I think it'd make sense to.

I would just show system defaults, as how they are applied when selecting the "Reset" option. I expect users understand what those settings do when they touch them, i.e. enable APT cache which is disabled now by default on DietPi.

> Are there are other dialogues other than APT caching that we should consider?

Not sure currently. I would not spend much effort to add multiple explanations as long as there is no explicit user request. Some are obvious enough, some might require much more text then we can reasonably add, where it would be advised to read man pages to get a real understanding.

In case of APT cache/lists/archives I at least have an idea about how to explain things and it also would be nice to show the current size and location of each.
_____________
However since I want to release v6.26 soon I will postpone this to v6.27. If someone is willing to work on this now, I will help to implement for v6.26, but IMO it is nothing urgent:
- Never touch settings that you don't understand.
- Read the above to hopefully get some understanding: https://github.com/MichaIng/DietPi/issues/3011#issuecomment-515711474

MichaIng, 2019-11-13:
Reopen, "default" has been added, but detailed info about the three values/options is outstanding.

commit 6d8d6bbc (2019-07-27): v6.26

+ DietPi-Config | APT cache: Add "(default)" suffix to the default value of each setting for user info. Fixes #3011