MichaIng/DietPi #3014: Check log dependencies

PeterLacknase, 2019-07-27:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

OpenVPN pulls rsyslog as (an unnecessary) dep, when ramlog is active.

#### Describe the solution you'd like:
 Dependency checks for superficial log daemons.

#### Describe alternatives you've considered:

`sudo apt-get purge rsyslog
sudo apt-get autoremove`

(The former unnecessary remaining deps from rsyslog like libestr0, libfastjson4, liblognorm5.)

MichaIng, 2019-07-27:
@PeterLacknase 
Thanks for your report.

I could not find unbound pulling rsyslog: https://packages.debian.org/stretch/unbound
Is this on RPi or non-RPi and which `echo $G_DISTRO`?

However yeah generally DietPi-RAMlog and rsyslog do not conflict and purging rsyslog could purge dependants as well. Actually we should be more careful, the other way round, when enabling RAMlog not purging rsyslog, but instead do `apt-mark auto rsyslog; G_AGA` to only autoremove it, if nothing depends on it.

There is another open issue tangent to this topic btw: https://github.com/MichaIng/DietPi/issues/2454

My middle-term aim for this is the following:
- Remove rsyslog from our logging options completely, as it is an obsolete overhead on systemd systems, where journald is anyway running as in-RAM syslog daemon.
- rsyslog will then only be installed as dependency, if any 3rd party package depends on it. Is so, actually we should open a bug report to get all packages removing rsyslog from their deps list. Every modern Debian package should allow to use journald instead, if any system logs are required.
- As an alternative, when RAMlog is disabled, one will be able to select logrotate and select journald to be boot persistent. With this set, rsyslog is superseded in every use case 😉.

PeterLacknase, 2019-07-28:
Raspberry Pi 3B+
Debian/Raspbian Buster 
DietPi 6.25.3

`echo $G_DISTRO` yields „5“.

MichaIng, 2019-07-28:
@PeterLacknase 
Indeed, DietPi-Software itself installs rsyslog together with OpenVPN and some other software titles as well.

Before blaming package dependencies and 3rd party software, we should instead re-check those requirements first and in case remove or workaround this in DietPi-Software 😄.

MichaIng, 2020-03-20:
Removed from all software titles but CloudPrint. However the latter pulls it as APT dependency. I committed the removal of this dependency to upstream (see above) but it will likely never reach Debian repo anymore. Also not that CloudPrint will shutdown til the end of the year, so no reason to start larger efforts on this.

closed 2020-03-20 by MichaIng

MichaIng, 2020-05-17:
Dependency has been removed from newest version on Debian as well 👍: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=954309
The new version `0.14-14` has not yet reached any repo suite: https://packages.debian.org/cloudprint