MichaIng/DietPi #3017: Cant install nextcloud

aspinks, 2019-07-28:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. ...Trying to install nextcloud
2. ...

#### Expected behaviour

- ...nextcloud gets installed

#### Actual behaviour

- ...I get the following:

#### Details:
- Date           | Sun 28 Jul 15:18:27 CDT 2019
- Bug report     | N/A
- DietPi version | v6.25.3 (MichaIng/master)
- Img creator    | n/a
- Pre-image      | n/a
- SBC device     | Native PC (x86_64) (index=21)
- Kernel version | #1 SMP Debian 4.9.168-1+deb9u4 (2019-07-19)
- Distro         | stretch (index=4)
- Command        | systemctl restart mariadb
- Exit code      | 1
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Job for mariadb.service failed because the control process exited with error code.
See "systemctl status mariadb.service" and "journalctl -xe" for details.

#### Extra details
<!-- Please post any extra details that might help solve the issue -->
- ...

MichaIng, 2019-07-29:
@aspinks 
Many thanks for your report.

Please paste the output of: `journalctl -u mariadb`

aspinks, 2019-07-29:
root@DietPi:~# journalctl -u mariadb
-- Logs begin at Sun 2019-07-28 15:01:33 CDT, end at Mon 2019-07-29 12:56:09 CDT
. --
Jul 28 15:12:35 DietPi systemd[1]: Starting MariaDB 10.1.38 database server...
Jul 28 15:12:35 DietPi mysqld[11330]: 2019-07-28 15:12:35 140456022207872 [Note]
 /usr/sbin/mysqld (mysqld 10.1.38-MariaDB-0+deb9u1) starting as process 11330 ..
.
Jul 28 15:12:35 DietPi mysqld[11330]: 2019-07-28 15:12:35 140456022207872 [Note]
 InnoDB: innodb_empty_free_list_algorithm has been changed to legacy because of 
small buffer pool size. In order to use backoff, increase buffer pool at least u
p to 20MB.
Jul 28 15:12:36 DietPi mysqld[11330]: 2019-07-28 15:12:36 140456022207872 [Note]
 InnoDB: Using mutexes to ref count buffer pool pages
Jul 28 15:12:36 DietPi mysqld[11330]: 2019-07-28 15:12:36 140456022207872 [Note]
 InnoDB: The InnoDB memory heap is disabled
Jul 28 15:12:36 DietPi mysqld[11330]: 2019-07-28 15:12:36 140456022207872 [Note]
 InnoDB: Mutexes and rw_locks use GCC atomic builtins
Jul 28 15:12:36 DietPi mysqld[11330]: 2019-07-28 15:12:36 140456022207872 [Note]
 InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
Jul 28 15:12:36 DietPi mysqld[11330]: 2019-07-28 15:12:36 140456022207872 [Note]
 InnoDB: Compressed tables use zlib 1.2.8
Jul 28 15:12:36 DietPi mysqld[11330]: 2019-07-28 15:12:36 140456022207872 [Note]
 InnoDB: Using Linux native AIO
Jul 28 15:12:36 DietPi mysqld[11330]: 2019-07-28 15:12:36 140456022207872 [Note]
 InnoDB: Using SSE crc32 instructions
Jul 28 15:12:36 DietPi mysqld[11330]: 2019-07-28 15:12:36 140456022207872 [Note] InnoDB: Initializing buffer pool
, size = 128.0M
Jul 28 15:12:36 DietPi mysqld[11330]: 2019-07-28 15:12:36 140456022207872 [Note] InnoDB: Completed initialization
 of buffer pool
Jul 28 15:12:36 DietPi mysqld[11330]: 2019-07-28 15:12:36 140456022207872 [Note] InnoDB: Highest supported file f
ormat is Barracuda.
Jul 28 15:12:36 DietPi mysqld[11330]: 2019-07-28 15:12:36 140456022207872 [Note] InnoDB: 128 rollback segment(s) 
are active.
Jul 28 15:12:36 DietPi mysqld[11330]: 2019-07-28 15:12:36 140456022207872 [Note] InnoDB: Waiting for purge to sta
rt
Jul 28 15:12:36 DietPi mysqld[11330]: 2019-07-28 15:12:36 140456022207872 [Note] InnoDB:  Percona XtraDB (http://
www.percona.com) 5.6.42-84.2 started; log sequence number 1616727
Jul 28 15:12:36 DietPi mysqld[11330]: 2019-07-28 15:12:36 140456022207872 [Note] Plugin 'FEEDBACK' is disabled.
Jul 28 15:12:36 DietPi mysqld[11330]: 2019-07-28 15:12:36 140455270414080 [Note] InnoDB: Dumping buffer pool(s) n
ot yet started
Jul 28 15:12:36 DietPi mysqld[11330]: 2019-07-28 15:12:36 140456022207872 [Note] Server socket created on IP: '::
'.
Jul 28 15:12:36 DietPi mysqld[11330]: 2019-07-28 15:12:36 140456022207872 [Note] /usr/sbin/mysqld: ready for conn
ections.
Jul 28 15:12:36 DietPi mysqld[11330]: Version: '10.1.38-MariaDB-0+deb9u1'  socket: '/var/run/mysqld/mysqld.sock' 
 port: 3306  Debian 9.8
Jul 28 15:12:36 DietPi systemd[1]: mariadb.service: Control process exited, code=exited status=203
Jul 28 15:12:36 DietPi mysqld[11330]: 2019-07-28 15:12:36 140456021367552 [Note] /usr/sbin/mysqld: Normal shutdow
n
Jul 28 15:12:36 DietPi mysqld[11330]: 2019-07-28 15:12:36 140456021367552 [Note] Event Scheduler: Purging the que
ue. 0 events
Jul 28 15:12:36 DietPi mysqld[11330]: 2019-07-28 15:12:36 140455324915456 [Note] InnoDB: FTS optimize thread exit
ing.
Jul 28 15:12:36 DietPi mysqld[11330]: 2019-07-28 15:12:36 140456021367552 [Note] InnoDB: Starting shutdown...
Jul 28 15:12:37 DietPi mysqld[11330]: 2019-07-28 15:12:37 140456021367552 [Note] InnoDB: Waiting for page_cleaner
 to finish flushing of buffer pool
Jul 28 15:12:38 DietPi mysqld[11330]: 2019-07-28 15:12:38 140456021367552 [Note] InnoDB: Shutdown completed; log 
sequence number 1616737
Jul 28 15:12:38 DietPi mysqld[11330]: 2019-07-28 15:12:38 140456021367552 [Note] /usr/sbin/mysqld: Shutdown compl
ete
Jul 28 15:12:38 DietPi systemd[1]: Failed to start MariaDB 10.1.38 database server.
Jul 28 15:12:38 DietPi systemd[1]: mariadb.service: Unit entered failed state.
Jul 28 15:12:38 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.
Jul 28 15:18:20 DietPi systemd[1]: Starting MariaDB 10.1.38 database server...
Jul 28 15:18:21 DietPi mysqld[18844]: 2019-07-28 15:18:21 139854858591616 [Note] /usr/sbin/mysqld (mysqld 10.1.38
-MariaDB-0+deb9u1) starting as process 18844 ...
Jul 28 15:18:21 DietPi mysqld[18844]: 2019-07-28 15:18:21 139854858591616 [Note] InnoDB: innodb_empty_free_list_a
lgorithm has been changed to legacy because of small buffer pool size. In order to use backoff, increase buffer p
ool at least up to 20MB.
Jul 28 15:18:21 DietPi mysqld[18844]: 2019-07-28 15:18:21 139854858591616 [Note] InnoDB: Using mutexes to ref cou
[... 25 lines cut ...]
Jul 28 15:18:21 DietPi mysqld[18844]: 2019-07-28 15:18:21 139854109210368 [Note] InnoDB: Dumping buffer pool(s) n
ot yet started
Jul 28 15:18:21 DietPi mysqld[18844]: 2019-07-28 15:18:21 139854858591616 [Note] Server socket created on IP: '::
'.
Jul 28 15:18:21 DietPi mysqld[18844]: 2019-07-28 15:18:21 139854858591616 [Note] /usr/sbin/mysqld: ready for conn
ections.
Jul 28 15:18:21 DietPi mysqld[18844]: Version: '10.1.38-MariaDB-0+deb9u1'  socket: '/var/run/mysqld/mysqld.sock' 
 port: 3306  Debian 9.8
Jul 28 15:18:21 DietPi systemd[1]: mariadb.service: Control process exited, code=exited status=203
[... cut ...]
Jul 28 15:18:23 DietPi mysqld[18844]: 2019-07-28 15:18:23 139854857751296 [Note] InnoDB: Shutdown completed; log 
sequence number 1616747
Jul 28 15:18:23 DietPi mysqld[18844]: 2019-07-28 15:18:23 139854857751296 [Note] /usr/sbin/mysqld: Shutdown compl
ete
Jul 28 15:18:24 DietPi systemd[1]: Failed to start MariaDB 10.1.38 database server.
Jul 28 15:18:24 DietPi systemd[1]: mariadb.service: Unit entered failed state.
Jul 28 15:18:24 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.

MichaIng, 2019-07-29:
@aspinks 
Okay no hint from the systemd unit. Please paste: `cat /var/log/mysql/error.log`
If it's empty, please do another restart attempt beforehead:
```
systemctl restart mariadb
cat /var/log/mysql/error.log
```

And if there is still nothing reported, perhaps calling the binary manually produces some meaningful output: `mysqld`

aspinks, 2019-07-29:
root@DietPi:~# cat /var/log/mysql/error.log
root@DietPi:~# systemctl restart mariadb
Job for mariadb.service failed because the control process exited with error code.
See "systemctl status mariadb.service" and "journalctl -xe" for details.
root@DietPi:~# sleep 4
root@DietPi:~# cat /var/log/mysql/error.log
root@DietPi:~# mysqld
bash: mysqld: command not found

Now if it helps anything, i had nextcloud installed and running, but then it would not let me login with the correct password.  I deleted the mysql and nextdoor back up folders because when i tried to reinstall it told me i had no space for it.

MichaIng, 2019-07-29:
@aspinks 
Indeed strange, since the `mysqld` is somehow missing, but that does not really fit to the service output above.

Okay how much space do you actually have free:
```
df -h
readlink -f /mnt/dietpi_userdata
```

aspinks, 2019-07-29:
root@DietPi:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            7.9G     0  7.9G   0% /dev
tmpfs           1.6G   21M  1.6G   2% /run
/dev/sda2        28G  8.4G   20G  31% /
tmpfs           7.9G  8.0K  7.9G   1% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           7.9G     0  7.9G   0% /sys/fs/cgroup
/dev/sda1       511M  256K  511M   1% /boot/efi
/dev/sdc1       1.9T   86G  1.8T   5% /mnt/PHOTOS
/dev/sdd2       4.6T  246M  4.6T   1% /mnt/MOVIES5TB
/dev/sdb2       3.7T  3.4T  284G  93% /mnt/MOVIES4TB
tmpfs            50M  2.1M   48M   5% /var/log
tmpfs            10M  1.4M  8.7M  14% /DietPi
tmpfs           7.9G   28K  7.9G   1% /tmp
tmpfs           1.6G     0  1.6G   0% /run/user/0
root@DietPi:~# readlink -f /mnt/dietpi_userdata
/mnt/dietpi_userdata

aspinks, 2019-07-29:
root@DietPi:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            7.9G     0  7.9G   0% /dev
tmpfs           1.6G   21M  1.6G   2% /run
/dev/sda2        28G  8.4G   20G  31% /
tmpfs           7.9G  8.0K  7.9G   1% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           7.9G     0  7.9G   0% /sys/fs/cgroup
/dev/sda1       511M  256K  511M   1% /boot/efi
/dev/sdc1       1.9T   86G  1.8T   5% /mnt/PHOTOS
/dev/sdd2       4.6T  246M  4.6T   1% /mnt/MOVIES5TB
/dev/sdb2       3.7T  3.4T  284G  93% /mnt/MOVIES4TB
tmpfs            50M  2.1M   48M   5% /var/log
tmpfs            10M  1.4M  8.7M  14% /DietPi
tmpfs           7.9G   28K  7.9G   1% /tmp
tmpfs           1.6G     0  1.6G   0% /run/user/0
root@DietPi:~# readlink -f /mnt/dietpi_userdata
/mnt/dietpi_userdata

Not sure why first paste had strikethrough on it.

MichaIng, 2019-07-29:
@aspinks 
> Not sure why this all has a strike through on it.

You should wrap pasts from console as code (as I quick edited inside 😉). It's better readable and GitHub does not expend markdown code, e.g. `~` leads to everything that follows is struck through. Select the pasted lines and click the "Insert code" `<>` button at top of the editor.

Okay, more then enough free space. Are there any I/O errors from the drive? `dmesg | tail`

Lets start from fresh with MariaDB, whatever went wrong with that. It was installed right with Nextcloud, so there is not other software installed that uses MariaDB, right?

dietpi-services stop
```
G_AGP mariadb-server mariadb-server-10.1
G_AGA
rm -R /var/lib/mysql /mnt/dietpi_userdata/mysql
mkdir /mnt/dietpi_userdata/mysql
ln -s /mnt/dietpi_userdata/mysql /var/lib/mysql
apt -y install mariadb-server
systemctl status mariadb
```

aspinks, 2019-07-29:
root@DietPi:~# dmesg | tail
[    6.057814] e1000e: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
[    6.057848] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   12.345965] FAT-fs (sda1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[   12.371478] fuse init (API version 7.26)
[ 6058.106809] usb 4-4: USB disconnect, device number 5
[ 6058.137244] sd 9:0:0:0: [sde] Synchronizing SCSI cache
[ 6058.137278] sd 9:0:0:0: [sde] Synchronize Cache(10) failed: Result: hostbyte=DID_NO_CONNECT driverbyte=DRIVER_OK
[ 6058.471630] usb 4-3: USB disconnect, device number 4
[ 6058.473167] sd 8:0:0:0: [sdc] Synchronizing SCSI cache
[ 6058.711777] sd 8:0:0:0: [sdc] Synchronize Cache(10) failed: Result: hostbyte=DID_ERROR driverbyte=DRIVER_OK
root@DietPi:~# dietpi-services stop
[  OK  ] DietPi-Services | Root access verified.

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: stop 

[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : webmin
[  OK  ] DietPi-Services | stop : netdata
[  OK  ] DietPi-Services | stop : deluge-web
[  OK  ] DietPi-Services | stop : plexmediaserver
[  OK  ] DietPi-Services | stop : deluged
[  OK  ] DietPi-Services | stop : apache2
[  OK  ] DietPi-Services | stop : php7.3-fpm
[  OK  ] DietPi-Services | stop : php7.0-fpm
[  OK  ] DietPi-Services | stop : mariadb
[  OK  ] DietPi-Services | stop : redis-server
root@DietPi:~# G_AGP mariadb-server mariadb-server-10.1
[  OK  ] Root access verified.
[ INFO ] APT removal for: mariadb-server
mariadb-server-10.1, please wait...
(Reading database ... 93618 files and directories currently installed.)
Removing mariadb-server (10.1.38-0+deb9u1) ...
Removing mariadb-server-10.1 (10.1.38-0+deb9u1) ...
(Reading database ... 93489 files and directories currently installed.)
Purging configuration files for mariadb-server-10.1 (10.1.38-0+deb9u1) ...
/var/lib/dpkg/info/mariadb-server-10.1.postrm: line 48: update-rc.d: command not found
dpkg: error processing package mariadb-server-10.1 (--purge):
 subprocess installed post-removal script returned error exit status 127
Errors were encountered while processing:
 mariadb-server-10.1
E: Sub-process /usr/bin/dpkg returned an error code (1)
[FAILED] G_AGP mariadb-server mariadb-server-10.1

--------------------------------------------------------------------
- DietPi has encounted an error, and, is unable to continue        -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket             -
--------------------------------------------------------------------

#### Details:
- Date           | Mon 29 Jul 15:33:52 CDT 2019
- Bug report     | N/A
- DietPi version | v6.25.3 (MichaIng/master)
- Img creator    | n/a
- Pre-image      | n/a
- SBC device     | Native PC (x86_64) (index=21)
- Kernel version | #1 SMP Debian 4.9.168-1+deb9u4 (2019-07-19)
- Distro         | stretch (index=4)
- Command        | G_AGP mariadb-server mariadb-server-10.1
- Exit code      | 100
- Software title | 

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
(Reading database ... 93618 files and directories currently installed.)
Removing mariadb-server (10.1.38-0+deb9u1) ...
Removing mariadb-server-10.1 (10.1.38-0+deb9u1) ...
(Reading database ... 93489 files and directories currently installed.)
Purging configuration files for mariadb-server-10.1 (10.1.38-0+deb9u1) ...
/var/lib/dpkg/info/mariadb-server-10.1.postrm: line 48: update-rc.d: command not found
dpkg: error processing package mariadb-server-10.1 (--purge):
 subprocess installed post-removal script returned error exit status 127
Errors were encountered while processing:
 mariadb-server-10.1
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

--------------------------------------------------------------------

[FAILED] Unable to continue,  will now terminate.

MichaIng, 2019-07-29:
@aspinks 
> Synchronize Cache(10) failed

Although it only occurred on the external drives. Did you unplug them without running `umount <mount_point>` before? This should be always done on journaling file systems since smaller written data is cached first and not written directly to disk. You simply unplugging it can lead to data loss or corruption.

> FAT-fs (sda1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.

This does not necessarily mean something bad, sometimes for some reasons on shutdown systemd does not succeed in unmounting the drives, however all data should be synced to disk before that automatically. But in case of system crash of course a fsck should be done. At least it is worth to run once on EFI and root partition, especially because of stranger error below:
```
fsck /dev/sda1
fsck -f /dev/sda2
```

> update-rc.d: command not found

There is something more deeply broken on the system. `update-rc.d` is a system command that should always be present, otherwise service failures are totally expected.

Does the file really not exist or is perhaps the `$PATH` corrupted where the shell searches for binaries?
```
ls -l /usr/sbin/update-rc.d
echo $PATH
hash -r # This clears all remembered binary locations and forces the shell to recheck
command -v update-rc.d
```

If the last command above does not produce any output, we can try to reinstall the related package. It should not depend on the binaries itself contains:
```
apt -y install --reinstall init-system-helpers
```

When did you install this system? Did you follow method 1 or 2 from [here](https://github.com/MichaIng/DietPi/issues/1171#issuecomment-336522021)?

aspinks, 2019-07-30:
root@DietPi:~# fsck /dev/sda1
fsck from util-linux 2.29.2
fsck.fat 4.1 (2017-01-24)
0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
1) Remove dirty bit
2) No action
? 1
Perform changes ? (y/n) y
/dev/sda1: 5 files, 64/130812 clusters
root@DietPi:~# fsck /dev/sda1
fsck from util-linux 2.29.2
fsck.fat 4.1 (2017-01-24)
/dev/sda1: 5 files, 64/130812 clusters
root@DietPi:~# fsck -f /dev/sda2
fsck from util-linux 2.29.2
e2fsck 1.43.4 (31-Jan-2017)
/dev/sda2 is mounted.
e2fsck: Cannot continue, aborting.

root@DietPi:~# ls -l /usr/sbin/update-rc.d
-rwxr-xr-x 1 root root 16062 May  2  2017 /usr/sbin/update-rc.d
root@DietPi:~# echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
root@DietPi:~# hash -r # This clears all remembered binary locations and forces the shell to recheck
root@DietPi:~#
root@DietPi:~#
root@DietPi:~# command -v update-rc.d
/usr/sbin/update-rc.d

I installed about 3-4 months ago using method 2.

MichaIng, 2019-07-30:
@aspinks 
So the binary is there, does this still fail? `apt purge mariadb-server-10.1`

aspinks, 2019-07-30:
Looks like that purge worked this time:

root@DietPi:~# apt purge mariadb-server-10.1
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  galera-3 libaio1 libconfig-inifiles-perl libdbi-perl libreadline5 lsof
  mariadb-client-10.1 mariadb-client-core-10.1 mariadb-common
  mariadb-server-core-10.1 mysql-common rsync socat
Use 'apt autoremove' to remove them.
The following packages will be REMOVED:
  mariadb-server-10.1*
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
(Reading database ... 93478 files and directories currently installed.)
Purging configuration files for mariadb-server-10.1 (10.1.38-0+deb9u1) ...

MichaIng, 2019-07-30:
@aspinks 
Okay nice, then you can go with the next steps I mentioned above, to have a fresh MariaDB install:
```
G_AGA
rm -R /var/lib/mysql /mnt/dietpi_userdata/mysql
mkdir /mnt/dietpi_userdata/mysql
ln -s /mnt/dietpi_userdata/mysql /var/lib/mysql
apt -y install mariadb-server
systemctl status mariadb
```

aspinks, 2019-07-30:
so i did all that and it installed mariadb sucesfully.  I then went to install nextcloud and get this error now.

#### Details:
- Date           | Mon 29 Jul 20:11:05 CDT 2019
- Bug report     | N/A
- DietPi version | v6.25.3 (MichaIng/master)
- Img creator    | n/a
- Pre-image      | n/a
- SBC device     | Native PC (x86_64) (index=21)
- Kernel version | #1 SMP Debian 4.9.168-1+deb9u4 (2019-07-19)
- Distro         | stretch (index=4)
- Command        | ncc maintenance:install
- Exit code      | 1
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
PHP Warning:  PHP Startup: Unable to load dynamic library 'pdo_mysql.so' (tried: /usr/lib/php/20180731/pdo_mysql.so (/usr/lib/php/20180731/pdo_mysql.so: undefined symbol: mysqlnd_get_client_info), /usr/lib/php/20180731/pdo_mysql.so.so (/usr/lib/php/20180731/pdo_mysql.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0
No database drivers (sqlite, mysql, or postgresql) installed.

An unhandled exception has been thrown:
Exception: Environment not properly prepared. in /var/www/nextcloud/lib/private/Console/Application.php:166
Stack trace:
#0 /var/www/nextcloud/console.php(96): OC\Console\Application->loadCommands(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#1 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#2 {main}PHP Warning:  fileperms(): stat failed for /var/www/nextcloud/data/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 140
{"reqId":"2VitRvr0AVRDGrGo0wMq","level":3,"time":"2019-07-30T01:10:37+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fileperms(): stat failed for \/var\/www\/nextcloud\/data\/nextcloud.log at \/var\/www\/nextcloud\/lib\/private\/Log\/File.php#140","userAgent":"--","version":""}
{"reqId":"2VitRvr0AVRDGrGo0wMq","level":3,"time":"2019-07-30T01:10:37+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"PHP Startup: Unable to load dynamic library 'pdo_mysql.so' (tried: \/usr\/lib\/php\/20180731\/pdo_mysql.so (\/usr\/lib\/php\/20180731\/pdo_mysql.so: undefined symbol: mysqlnd_get_client_info), \/usr\/lib\/php\/20180731\/pdo_mysql.so.so (\/usr\/lib\/php\/20180731\/pdo_mysql.so.so: cannot open shared object file: No such file or directory)) at Unknown#0","userAgent":"--","version":""}
```

--------------------------------------------------------------------

[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

MichaIng, 2019-07-30:
@aspinks 
Does this file again actually exist or not?
```
ls -l /usr/lib/php/20180731/pdo_mysql.so
```

However lets reinstall the module and enable it manually:
```
apt -y install --reinstall php7.3-mysql
phpenmod pdo_mysql
php -m | grep 'pdo_mysql'
```
- The last command should then produce `pdo_mysql` as output.

aspinks, 2019-07-30:
root@DietPi:~# ls -l /usr/lib/php/20180731/pdo_mysql.so
-rw-r--r-- 1 root root 30824 Jul 25 06:53 /usr/lib/php/20180731/pdo_mysql.so
root@DietPi:~# apt -y install --reinstall php7.3-mysql
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 115 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 https://packages.sury.org/php stretch/main amd64 php7.3-mysql amd64 7.3.7-2+0~20190725.42+debian9~1.gbp848ca5 [115 kB]
Fetched 115 kB in 1s (114 kB/s)
(Reading database ... 93618 files and directories currently installed.)
Preparing to unpack .../php7.3-mysql_7.3.7-2+0~20190725.42+debian9~1.gbp848ca5_amd64.deb ...
Unpacking php7.3-mysql (7.3.7-2+0~20190725.42+debian9~1.gbp848ca5) over (7.3.7-2+0~20190725.42+debian9~1.gbp848ca5) ...
Processing triggers for libapache2-mod-php7.3 (7.3.7-2+0~20190725.42+debian9~1.gbp848ca5) ...
Setting up php7.3-mysql (7.3.7-2+0~20190725.42+debian9~1.gbp848ca5) ...
Not replacing deleted config file /etc/php/7.3/mods-available/mysqlnd.ini
WARNING: Module mysqlnd ini file doesn't exist under /etc/php/7.3/mods-available
WARNING: Module mysqlnd ini file doesn't exist under /etc/php/7.3/mods-available
WARNING: Module mysqlnd ini file doesn't exist under /etc/php/7.3/mods-available
Not replacing deleted config file /etc/php/7.3/mods-available/mysqli.ini
WARNING: Module mysqli ini file doesn't exist under /etc/php/7.3/mods-available
WARNING: Module mysqli ini file doesn't exist under /etc/php/7.3/mods-available
WARNING: Module mysqli ini file doesn't exist under /etc/php/7.3/mods-available
Processing triggers for php7.3-fpm (7.3.7-2+0~20190725.42+debian9~1.gbp848ca5) ...
NOTICE: Not enabling PHP 7.3 FPM by default.
NOTICE: To enable PHP 7.3 FPM in Apache2 do:
NOTICE: a2enmod proxy_fcgi setenvif
NOTICE: a2enconf php7.3-fpm
NOTICE: You are seeing this message because you have apache2 package installed.
Processing triggers for libapache2-mod-php7.3 (7.3.7-2+0~20190725.42+debian9~1.gbp848ca5) ...
root@DietPi:~# phpenmod pdo_mysql
WARNING: Module pdo_mysql ini file doesn't exist under /etc/php/7.0/mods-available
WARNING: Module pdo_mysql ini file doesn't exist under /etc/php/7.0/mods-available
root@DietPi:~# php -m | grep 'pdo_mysql'
PHP Warning:  PHP Startup: Unable to load dynamic library 'pdo_mysql.so' (tried: /usr/lib/php/20180731/pdo_mysql.so (/usr/lib/php/20180731/pdo_mysql.so: undefined symbol: mysqlnd_get_client_info), /usr/lib/php/20180731/pdo_mysql.so.so (/usr/lib/php/20180731/pdo_mysql.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0

MichaIng, 2019-07-30:
@aspinks 
> Not replacing deleted config file /etc/php/7.3/mods-available/mysqlnd.ini
WARNING: Module mysqlnd ini file doesn't exist under /etc/php/7.3/mods-available

It has been manually removed??
Okay then purge and install to have DPKG forgetting about "config" files:
```
apt -y purge php7.3-mysql
G_AGA
apt -y install php7.3-mysql
```

aspinks, 2019-07-30:
`root@DietPi:~# apt -y purge php7.3-mysql
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages will be REMOVED:
  php7.3-mysql*
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
After this operation, 450 kB disk space will be freed.
(Reading database ... 93618 files and directories currently installed.)
Removing php7.3-mysql (7.3.7-2+0~20190725.42+debian9~1.gbp848ca5) ...
WARNING: Module mysqlnd ini file doesn't exist under /etc/php/7.3/mods-available
WARNING: Module mysqlnd ini file doesn't exist under /etc/php/7.3/mods-available
WARNING: Module mysqlnd ini file doesn't exist under /etc/php/7.3/mods-available
WARNING: Module mysqli ini file doesn't exist under /etc/php/7.3/mods-available
WARNING: Module mysqli ini file doesn't exist under /etc/php/7.3/mods-available
WARNING: Module mysqli ini file doesn't exist under /etc/php/7.3/mods-available
Processing triggers for libapache2-mod-php7.3 (7.3.7-2+0~20190725.42+debian9~1.gbp848ca5) ...
Processing triggers for php7.3-fpm (7.3.7-2+0~20190725.42+debian9~1.gbp848ca5) ...
NOTICE: Not enabling PHP 7.3 FPM by default.
NOTICE: To enable PHP 7.3 FPM in Apache2 do:
NOTICE: a2enmod proxy_fcgi setenvif
NOTICE: a2enconf php7.3-fpm
NOTICE: You are seeing this message because you have apache2 package installed.
(Reading database ... 93605 files and directories currently installed.)
Purging configuration files for php7.3-mysql (7.3.7-2+0~20190725.42+debian9~1.gbp848ca5) ...
root@DietPi:~# G_AGA
[  OK  ] Root access verified.
[ INFO ] APT autoremove + purge, please wait...
[  OK  ] G_AGA
root@DietPi:~# apt -y install php7.3-mysql
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  php7.3-mysql
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 115 kB of archives.
After this operation, 450 kB of additional disk space will be used.
Get:1 https://packages.sury.org/php stretch/main amd64 php7.3-mysql amd64 7.3.7-2+0~20190725.42+debian9~1.gbp848ca5 [115 kB]
Fetched 115 kB in 1s (115 kB/s)
Selecting previously unselected package php7.3-mysql.
(Reading database ... 93605 files and directories currently installed.)
Preparing to unpack .../php7.3-mysql_7.3.7-2+0~20190725.42+debian9~1.gbp848ca5_amd64.deb ...
Unpacking php7.3-mysql (7.3.7-2+0~20190725.42+debian9~1.gbp848ca5) ...
Processing triggers for libapache2-mod-php7.3 (7.3.7-2+0~20190725.42+debian9~1.gbp848ca5) ...
Setting up php7.3-mysql (7.3.7-2+0~20190725.42+debian9~1.gbp848ca5) ...

Creating config file /etc/php/7.3/mods-available/mysqlnd.ini with new version

Creating config file /etc/php/7.3/mods-available/mysqli.ini with new version

Creating config file /etc/php/7.3/mods-available/pdo_mysql.ini with new version
Processing triggers for php7.3-fpm (7.3.7-2+0~20190725.42+debian9~1.gbp848ca5) ...
NOTICE: Not enabling PHP 7.3 FPM by default.
NOTICE: To enable PHP 7.3 FPM in Apache2 do:
NOTICE: a2enmod proxy_fcgi setenvif
NOTICE: a2enconf php7.3-fpm
NOTICE: You are seeing this message because you have apache2 package installed.
Processing triggers for libapache2-mod-php7.3 (7.3.7-2+0~20190725.42+debian9~1.gbp848ca5) ...
root@DietPi:~# phpenmod pdo_mysql
WARNING: Module pdo_mysql ini file doesn't exist under /etc/php/7.0/mods-available
WARNING: Module pdo_mysql ini file doesn't exist under /etc/php/7.0/mods-available
root@DietPi:~# php -m | grep 'pdo_mysql'
pdo_mysql
`
Yes it was manually removed in the graphical desktop.  I checked and at this moment have no webservers installed.   So i went ahead and installed Lighttpd.

MichaIng, 2019-07-30:
@aspinks 
> Yes it was manually removed in the graphical desktop.

Be very careful with removing files like this manually. If you want to get rid of a PHP module or something else, remove the APT package instead of removing its files manually. There is a bunch of issues that can raise, as you can see now. And these `mysqlnd.ini` kind of files are only tiny config files, not the large libraries of the modules, even the documentation/info files of the packages are way larger, if your aim was to free space.
- Check `apt-mark showmanual` for all installed packages, that are no dependencies of others.
- If you want to remove one, and only do that if you exactly know what it is, then the safest way is to mark it as "auto" installed and leave it to autoremove: `apt-mark auto <package_name> && apt -y autoremove --purge`
  - That way it is assured that it is not removed if any other installed package requires it.
- `G_AGA` is btw the same as `apt -y autoremove --purge`, just with an error handler + prompt if an issue occurs. That is required for scripts that must not continue after an error, but when running in console it could be a bid annoying and spamming the screen, especially if you already know that currently error messages appear 😉.

> I checked and at this moment have no webservers installed. So i went ahead and installed Lighttpd.

Then please remove the Apache PHP module:
```
apt -y purge libapache2-mod-php7.3
apt -y autoremove --purge
```

And note that with DietPi-Software you never need to install any webserver manually. If you choose a software title that requires one, e.g. Nextcloud, the webserver that was chosen in the main menu of dietpi-software will be installed automatically (Lighttpd by default).

I still see:
```
WARNING: Module pdo_mysql ini file doesn't exist under /etc/php/7.0/mods-available
WARNING: Module pdo_mysql ini file doesn't exist under /etc/php/7.0/mods-available
```
but `php -m` lists it as loaded.

Did you remove the whole /etc/php/... directory?

aspinks, 2019-07-30:
/etc/php/ was not removed, it still remains.  the only directories i removed were the nextcloud and mysql manually.

I ran the commands above to remove apache and it did not remove it.

`root@DietPi:~# apt-mark showmanual
alsa-utils
apt-transport-https
apt-utils
bash-completion
bc
binutils
bzip2
ca-certificates
console-setup
crda
cron
curl
dbus
dbus-x11
debconf
debian-archive-keyring
deluge-console
deluge-web
deluged
dirmngr
dosfstools
ethtool
fake-hwclock
filebot
firefox-esr
firmware-atheros
firmware-brcm80211
firmware-intel-sound
firmware-linux-nonfree
firmware-misc-nonfree
firmware-realtek
fslint
gnome-icon-theme
gnupg
grub-efi-amd64
haveged
hdparm
htop
initramfs-tools
intel-microcode
iputils-ping
isc-dhcp-client
iw
libapache2-mod-php7.3
libgl1-mesa-dri
lighttpd
linux-image-amd64
locales
mariadb-server
mesa-utils
mesa-utils-extra
mono-complete
mono-runtime
nano
netdata
ntfs-3g
openssh-server
p7zip
p7zip-full
parted
perl
php-apcu
php-redis
php7.0-cli
php7.0-common
php7.0-curl
php7.0-fpm
php7.0-gd
php7.0-json
php7.0-mbstring
php7.0-mcrypt
php7.0-opcache
php7.0-readline
php7.0-xml
php7.0-zip
php7.3-cli
php7.3-curl
php7.3-fpm
php7.3-gd
php7.3-intl
php7.3-mbstring
php7.3-mysql
php7.3-xml
php7.3-zip
plexmediaserver
policykit-1
procps
psmisc
python
python-setuptools
redis-server
resolvconf
rfkill
rsyslog
sudo
systemd-sysv
tango-icon-theme
tigervnc-common
tzdata
unrar
unzip
upower
usbutils
vnc4server
webmin
wget
whiptail
wireless-tools
wpasupplicant
wput
x11-utils
x11-xserver-utils
x11vnc
xcompmgr
xfce4
xfce4-terminal
xfonts-base
xinit
xrdp
xserver-xorg
xterm
zip
root@DietPi:~# apt -y purge libapache2-mod-php7.3
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  apache2-bin libapr1 libaprutil1 libaprutil1-dbd-sqlite3 libaprutil1-ldap
  liblua5.2-0
Use 'apt autoremove' to remove them.
The following packages will be REMOVED:
  libapache2-mod-php7.3*
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
After this operation, 4,647 kB disk space will be freed.
(Reading database ... 93199 files and directories currently installed.)
Removing libapache2-mod-php7.3 (7.3.7-2+0~20190725.42+debian9~1.gbp848ca5) ...
/var/lib/dpkg/info/libapache2-mod-php7.3.prerm: 12: /var/lib/dpkg/info/libapache2-mod-php7.3.prerm: phpquery: not found
dpkg: error processing package libapache2-mod-php7.3 (--remove):
 subprocess installed pre-removal script returned error exit status 127
Warning: Could not load Apache 2.4 maintainer script helper.
Errors were encountered while processing:
 libapache2-mod-php7.3
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@DietPi:~# apt -y autoremove --purge
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
`

aspinks, 2019-07-30:
that came out weird...

root@DietPi:~# apt-mark showmanual
alsa-utils
apt-transport-https
apt-utils
bash-completion
bc
binutils
bzip2
ca-certificates
console-setup
crda
cron
curl
dbus
dbus-x11
debconf
debian-archive-keyring
deluge-console
deluge-web
deluged
dirmngr
dosfstools
ethtool
fake-hwclock
filebot
firefox-esr
firmware-atheros
firmware-brcm80211
firmware-intel-sound
firmware-linux-nonfree
firmware-misc-nonfree
firmware-realtek
fslint
gnome-icon-theme
gnupg
grub-efi-amd64
haveged
hdparm
htop
initramfs-tools
intel-microcode
iputils-ping
isc-dhcp-client
iw
libapache2-mod-php7.3
libgl1-mesa-dri
lighttpd
linux-image-amd64
locales
mariadb-server
mesa-utils
mesa-utils-extra
mono-complete
mono-runtime
nano
netdata
ntfs-3g
openssh-server
p7zip
p7zip-full
parted
perl
php-apcu
php-redis
php7.0-cli
php7.0-common
php7.0-curl
php7.0-fpm
php7.0-gd
php7.0-json
php7.0-mbstring
php7.0-mcrypt
php7.0-opcache
php7.0-readline
php7.0-xml
php7.0-zip
php7.3-cli
php7.3-curl
php7.3-fpm
php7.3-gd
php7.3-intl
php7.3-mbstring
php7.3-mysql
php7.3-xml
php7.3-zip
plexmediaserver
policykit-1
procps
psmisc
python
python-setuptools
redis-server
resolvconf
rfkill
rsyslog
sudo
systemd-sysv
tango-icon-theme
tigervnc-common
tzdata
unrar
unzip
upower
usbutils
vnc4server
webmin
wget
whiptail
wireless-tools
wpasupplicant
wput
x11-utils
x11-xserver-utils
x11vnc
xcompmgr
xfce4
xfce4-terminal
xfonts-base
xinit
xrdp
xserver-xorg
xterm
zip
root@DietPi:~# apt -y purge libapache2-mod-php7.3
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  apache2-bin libapr1 libaprutil1 libaprutil1-dbd-sqlite3 libaprutil1-ldap
  liblua5.2-0
Use 'apt autoremove' to remove them.
The following packages will be REMOVED:
  libapache2-mod-php7.3*
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
After this operation, 4,647 kB disk space will be freed.
(Reading database ... 93199 files and directories currently installed.)
Removing libapache2-mod-php7.3 (7.3.7-2+0~20190725.42+debian9~1.gbp848ca5) ...
/var/lib/dpkg/info/libapache2-mod-php7.3.prerm: 12: /var/lib/dpkg/info/libapache2-mod-php7.3.prerm: phpquery: not found
dpkg: error processing package libapache2-mod-php7.3 (--remove):
 subprocess installed pre-removal script returned error exit status 127
Warning: Could not load Apache 2.4 maintainer script helper.
Errors were encountered while processing:
 libapache2-mod-php7.3
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@DietPi:~# apt -y autoremove --purge
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

MichaIng, 2019-07-30:
@aspinks 
Ayy now I see at least one issue:
```
php7.0-cli
php7.0-common
php7.0-curl
php7.0-fpm
php7.0-gd
php7.0-json
php7.0-mbstring
php7.0-mcrypt
php7.0-opcache
php7.0-readline
php7.0-xml
php7.0-zip
```
- Two PHP versions are installed concurrently.

Try:
```
apt-mark auto php7.0-cli php7.0-common php7.0-curl php7.0-fpm php7.0-gd php7.0-json php7.0-mbstring php7.0-mcrypt php7.0-opcache php7.0-readline php7.0-xml php7.0-zip
apt -y autoremove --purge
apt -y purge $(dpkg --get-selections | mawk '/^php7.0/ {print $1}')
```

> /var/lib/dpkg/info/libapache2-mod-php7.3.prerm: phpquery: not found

That is again wired, this binary is part of `php-common` which should definitely be installed (as dependency). Is it again a binary that exist and was just not found somehow by dpkg?
```
ls -Al /usr/sbin/phpquery
command -v phpquery
hash -r
```

aspinks, 2019-07-30:
root@DietPi:~# apt-mark auto php7.0-cli php7.0-common php7.0-curl php7.0-fpm php7.0-gd php7.0-json php7.0-mbstring php7.0-mcrypt php7.0-opcache php7.0-readline php7.0-xml php7.0-zip
php7.0-cli set to automatically installed.
php7.0-common set to automatically installed.
php7.0-curl set to automatically installed.
php7.0-fpm set to automatically installed.
php7.0-gd set to automatically installed.
php7.0-json set to automatically installed.
php7.0-mbstring set to automatically installed.
php7.0-mcrypt set to automatically installed.
php7.0-opcache set to automatically installed.
php7.0-readline set to automatically installed.
php7.0-xml set to automatically installed.
php7.0-zip set to automatically installed.
root@DietPi:~# apt -y autoremove --purge
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages will be REMOVED:
  libltdl7* libmcrypt4* php7.0-curl* php7.0-gd* php7.0-mbstring*
  php7.0-mcrypt* php7.0-xml* php7.0-zip*
0 upgraded, 0 newly installed, 8 to remove and 0 not upgraded.
After this operation, 3,067 kB disk space will be freed.
(Reading database ... 93199 files and directories currently installed.)
Removing php7.0-mcrypt (7.0.33-9+0~20190712.17+debian9~1.gbp3af52c) ...
/var/lib/dpkg/info/php7.0-mcrypt.postrm: 215: /var/lib/dpkg/info/php7.0-mcrypt.postrm: phpquery: not found
dpkg: error processing package php7.0-mcrypt (--remove):
 subprocess installed post-removal script returned error exit status 127
Removing libltdl7:amd64 (2.4.6-2) ...
Removing libmcrypt4 (2.5.8-3.3) ...
Removing php7.0-curl (7.0.33-9+0~20190712.17+debian9~1.gbp3af52c) ...
/var/lib/dpkg/info/php7.0-curl.postrm: 215: /var/lib/dpkg/info/php7.0-curl.postrm: phpquery: not found
dpkg: error processing package php7.0-curl (--remove):
 subprocess installed post-removal script returned error exit status 127
Removing php7.0-gd (7.0.33-9+0~20190712.17+debian9~1.gbp3af52c) ...
/var/lib/dpkg/info/php7.0-gd.postrm: 215: /var/lib/dpkg/info/php7.0-gd.postrm: phpquery: not found
dpkg: error processing package php7.0-gd (--remove):
 subprocess installed post-removal script returned error exit status 127
Removing php7.0-mbstring (7.0.33-9+0~20190712.17+debian9~1.gbp3af52c) ...
/var/lib/dpkg/info/php7.0-mbstring.postrm: 215: /var/lib/dpkg/info/php7.0-mbstring.postrm: phpquery: not found
dpkg: error processing package php7.0-mbstring (--remove):
 subprocess installed post-removal script returned error exit status 127
Removing php7.0-xml (7.0.33-9+0~20190712.17+debian9~1.gbp3af52c) ...
/var/lib/dpkg/info/php7.0-xml.postrm: 215: /var/lib/dpkg/info/php7.0-xml.postrm: phpquery: not found
dpkg: error processing package php7.0-xml (--remove):
 subprocess installed post-removal script returned error exit status 127
Removing php7.0-zip (7.0.33-9+0~20190712.17+debian9~1.gbp3af52c) ...
/var/lib/dpkg/info/php7.0-zip.postrm: 215: /var/lib/dpkg/info/php7.0-zip.postrm: phpquery: not found
dpkg: error processing package php7.0-zip (--remove):
 subprocess installed post-removal script returned error exit status 127
Errors were encountered while processing:
 php7.0-mcrypt
 php7.0-curl
 php7.0-gd
 php7.0-mbstring
 php7.0-xml
 php7.0-zip
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@DietPi:~# apt -y purge $(dpkg --get-selections | mawk '/^php7.0/ {print $1}')
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages will be REMOVED:
  php7.0-cli* php7.0-common* php7.0-curl php7.0-fpm* php7.0-gd php7.0-json*
  php7.0-mbstring php7.0-mcrypt php7.0-opcache* php7.0-readline* php7.0-xml
  php7.0-zip
0 upgraded, 0 newly installed, 12 to remove and 0 not upgraded.
6 not fully installed or removed.
After this operation, 16.1 MB disk space will be freed.
(Reading database ... 93119 files and directories currently installed.)
Removing php7.0-curl (7.0.33-9+0~20190712.17+debian9~1.gbp3af52c) ...
[... 66 lines cut ...]
 php7.0-cli depends on php7.0-readline; however:
  Package php7.0-readline is to be removed.

Removing php7.0-readline (7.0.33-9+0~20190712.17+debian9~1.gbp3af52c) ...
/var/lib/dpkg/info/php7.0-readline.postrm: 215: /var/lib/dpkg/info/php7.0-readline.postrm: phpquery: not found
dpkg: error processing package php7.0-readline (--remove):
 subprocess installed post-removal script returned error exit status 127
Errors were encountered while processing:
 php7.0-curl
 php7.0-gd
 php7.0-mbstring
 php7.0-mcrypt
 php7.0-xml
 php7.0-zip
 php7.0-fpm
 php7.0-opcache
 php7.0-common
 php7.0-json
 php7.0-cli
 php7.0-readline
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@DietPi:~# ls -Al /usr/sbin/phpquery
-rwxr-xr-x 1 root root 6389 Jul 14  2018 /usr/sbin/phpquery
root@DietPi:~# command -v phpquery
root@DietPi:~# hash -r

MichaIng, 2019-07-30:
@aspinks 
All fail because of the missing binary, which is actually there...
```
root@DietPi:~# ls -Al /usr/sbin/phpquery
 -rwxr-xr-x 1 root root 6389 Jul 14 2018 /usr/sbin/phpquery
root@DietPi:~# command -v phpquery
root@DietPi:~# hash -r
```
- The first output shows the file is there.
- The second command did not output anything, which means that the command if for some strange reason not found by the system.
- Just the same as it was `update-rc.d` earlier.

Did you do a reboot some time ago? Perhaps this helps, as there seem to be something broken with the shell session.

Perhaps also `hash -r` already helped, if `command -v phpquery` now shows `/usr/sbin/phpquery`, then rerun the above apt purge command.

aspinks, 2019-07-30:
So i did do a reboot this morning earlier but for good measure I just did another reboot.  I reran the apt purge command and still issues, although they seem different.

root@DietPi:~# apt-mark auto php7.0-cli php7.0-common php7.0-curl php7.0-fpm php7.0-gd php7.0-json php7.0-mbstring php7.0-mcrypt php7.0-opcache php7.0-readline php7.0-xml php7.0-zip
php7.0-cli set to automatically installed.
php7.0-common set to automatically installed.
php7.0-curl set to automatically installed.
php7.0-fpm set to automatically installed.
php7.0-gd set to automatically installed.
php7.0-json set to automatically installed.
php7.0-mbstring set to automatically installed.
php7.0-mcrypt set to automatically installed.
php7.0-opcache set to automatically installed.
php7.0-readline set to automatically installed.
php7.0-xml set to automatically installed.
php7.0-zip set to automatically installed.
root@DietPi:~# apt -y autoremove --purge
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 php7.0-mcrypt : Depends: libltdl7 (>= 2.4.6) but it is not installed
                 Depends: libmcrypt4 but it is not installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
root@DietPi:~# apt -y purge $(dpkg --get-selections | mawk '/^php7.0/ {print $1}')
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages will be REMOVED:
  php7.0-cli* php7.0-common php7.0-curl php7.0-fpm* php7.0-gd php7.0-json
  php7.0-mbstring php7.0-mcrypt php7.0-opcache php7.0-readline php7.0-xml
  php7.0-zip
0 upgraded, 0 newly installed, 12 to remove and 0 not upgraded.
12 not fully installed or removed.
After this operation, 16.1 MB disk space will be freed.
(Reading database ... 93037 files and directories currently installed.)
Removing php7.0-fpm (7.0.33-9+0~20190712.17+debian9~1.gbp3af52c) ...
/var/lib/dpkg/info/php7.0-fpm.prerm: 12: /var/lib/dpkg/info/php7.0-fpm.prerm: phpquery: not found
dpkg: error processing package php7.0-fpm (--remove):
 subprocess installed pre-removal script returned error exit status 127
/var/lib/dpkg/info/php7.0-fpm.postinst: 93: /var/lib/dpkg/info/php7.0-fpm.postinst: update-rc.d: not found
dpkg: error while cleaning up:
 subprocess installed post-installation script returned error exit status 127
Removing php7.0-zip (7.0.33-9+0~20190712.17+debian9~1.gbp3af52c) ...
/var/lib/dpkg/info/php7.0-zip.postrm: 215: /var/lib/dpkg/info/php7.0-zip.postrm: phpquery: not found
dpkg: error processing package php7.0-zip (--remove):
 subprocess installed post-removal script returned error exit status 127
Removing php7.0-xml (7.0.33-9+0~20190712.17+debian9~1.gbp3af52c) ...
/var/lib/dpkg/info/php7.0-xml.postrm: 215: /var/lib/dpkg/info/php7.0-xml.postrm: phpquery: not found
dpkg: error processing package php7.0-xml (--remove):
 subprocess installed post-removal script returned error exit status 127
Removing php7.0-curl (7.0.33-9+0~20190712.17+debian9~1.gbp3af52c) ...
/var/lib/dpkg/info/php7.0-curl.postrm: 215: /var/lib/dpkg/info/php7.0-curl.postrm: phpquery: not found
dpkg: error processing package php7.0-curl (--remove):
 subprocess installed post-removal script returned error exit status 127
Removing php7.0-gd (7.0.33-9+0~20190712.17+debian9~1.gbp3af52c) ...
/var/lib/dpkg/info/php7.0-gd.postrm: 215: /var/lib/dpkg/info/php7.0-gd.postrm: phpquery: not found
dpkg: error processing package php7.0-gd (--remove):
 subprocess installed post-removal script returned error exit status 127
Removing php7.0-mbstring (7.0.33-9+0~20190712.17+debian9~1.gbp3af52c) ...
/var/lib/dpkg/info/php7.0-mbstring.postrm: 215: /var/lib/dpkg/info/php7.0-mbstring.postrm: phpquery: not found
dpkg: error processing package php7.0-mbstring (--remove):
 subprocess installed post-removal script returned error exit status 127
Removing php7.0-mcrypt (7.0.33-9+0~20190712.17+debian9~1.gbp3af52c) ...
/var/lib/dpkg/info/php7.0-mcrypt.postrm: 215: /var/lib/dpkg/info/php7.0-mcrypt.postrm: phpquery: not found
dpkg: error processing package php7.0-mcrypt (--remove):
 subprocess installed post-removal script returned error exit status 127
dpkg: php7.0-cli: dependency problems, but removing anyway as you requested:
 php7.0-fpm depends on php7.0-cli.

[... 30 lines cut ...]
/var/lib/dpkg/info/php7.0-opcache.postrm: 215: /var/lib/dpkg/info/php7.0-opcache.postrm: phpquery: not found
dpkg: error processing package php7.0-opcache (--remove):
 subprocess installed post-removal script returned error exit status 127
dpkg: php7.0-readline: dependency problems, but removing anyway as you requested:
 php7.0-cli depends on php7.0-readline; however:
  Package php7.0-readline is to be removed.

Removing php7.0-readline (7.0.33-9+0~20190712.17+debian9~1.gbp3af52c) ...
/var/lib/dpkg/info/php7.0-readline.postrm: 215: /var/lib/dpkg/info/php7.0-readline.postrm: phpquery: not found
dpkg: error processing package php7.0-readline (--remove):
 subprocess installed post-removal script returned error exit status 127
Errors were encountered while processing:
 php7.0-fpm
 php7.0-zip
 php7.0-xml
 php7.0-curl
 php7.0-gd
 php7.0-mbstring
 php7.0-mcrypt
 php7.0-cli
 php7.0-common
 php7.0-json
 php7.0-opcache
 php7.0-readline
E: Sub-process /usr/bin/dpkg returned an error code (1)

MichaIng, 2019-07-30:
@aspinks 
Okay, I have no idea where is problem comes from, but of course when various major binaries/commands are somehow not available/found by the system, a bunch of different errors can happen 🤔.

However lets focus on this particular one again, I am somehow interested where this leads to.
Btw, I see you have a desktop and vnc4server installed. Would you mind to grant me VNC credentials which would make debug and hopefully fixing the issue easier.
If so, send me info to micha@dietpi.com.

Otherwise the next steps:
- `command -v phpquery` must show and output, otherwise the PHP related apt commands are expected to fail.
- If the above does not show any output, check if `hash -r` fixes it.
- Otherwise perhaps linking the binary to a more prominent location helps: `ln -s /usr/sbin/phpquery /bin/phpquery && hash -r`

aspinks, 2019-07-30:
I emailed you my VPN acces.  Sent from thespinks@gmail.com

MichaIng, 2019-07-30:
@aspinks 
I meant VNC 😉, so remote deskop access, if it is accessible from outside of your local network (port forwarded from router?).
SSH would be an easy alternative, but then you must definitely change the root password, before opening it to www and bad about SSH is that you cannot watch what I am doing.

aspinks, 2019-07-30:
I don't have remote ssh setup.  

It might help if I gave you the ip and port.  They are 71.135.10.211:3389

MichaIng, 2019-07-30:
Okay found one main issue when opening a terminal from desktop:
```
root@DietPi:~# echo $PATH
/sbin:/bin:/usr/bin:/usr/local/bin
```
The path variable does not include `/usr/bin` `/usr/sbin` and `/usr/local/sbin` which it must to work properly.

aspinks, 2019-07-30:
Is there anything that can be done in the meantime to re-install nextcloud?

MichaIng, 2019-07-30:
@aspinks 
Okay things are fixed and Nextcloud has been successfully installed.

The issue indeed is XRDP as mentioned above. I fixed it by placing the following file:
```
root@DietPi:~# cat /etc/bashrc.d/01-xrdp_fix_env.sh
export PATH='/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin'
```
But this is no optimal solution as this overwrites any custom additions to the $PATH variable. However those are rare and I gave the file a very low priority (01-) so it is overridden by any other script, either in `/etc/bashrc.d/` `/etc/profile.d/` `~/.bashrc` or `~/.profile`.

aspinks, 2019-07-30:
So i attempted to access nextcloud both locally and remotely by going to (ip address/nextcloud) and get this:

<?php
/**
 *
 * Your webserver seems to be not configured to use PHP or PHP is not installed. 
 * Please contact your administrator or follow our documentation: 
 * https://docs.nextcloud.com/server/15/admin_manual/installation/source_installation.html 
 *
 * @copyright Copyright (c) 2016, ownCloud, Inc.
 *
 * @author Georg Ehrke <oc.list@georgehrke.com>
 * @author Joas Schilling <coding@schilljs.com>
 * @author JÃ¶rn Friedrich Dreyer <jfd@butonic.de>
 * @author Lukas Reschke <lukas@statuscode.ch>
 * @author Morris Jobke <hey@morrisjobke.de>
 * @author Robin Appelman <robin@icewind.nl>
 * @author Sergio BertolÃ­n <sbertolin@solidgear.es>
 * @author Thomas MÃ¼ller <thomas.mueller@tmit.eu>
 * @author Vincent Petry <pvince81@owncloud.com>
 *
 * @license AGPL-3.0
 *
 * This code is free software: you can redistribute it and/or modify
 * it under the terms of the GNU Affero General Public License, version 3,
 * as published by the Free Software Foundation.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
 * GNU Affero General Public License for more details.
 *
 * You should have received a copy of the GNU Affero General Public License, version 3,
 * along with this program.  If not, see <http://www.gnu.org/licenses/>
 *
 */

require_once __DIR__ . '/lib/versioncheck.php';

try {

	require_once __DIR__ . '/lib/base.php';

	OC::handleRequest();

} catch(\OC\ServiceUnavailableException $ex) {
	\OC::$server->getLogger()->logException($ex, array('app' => 'index'));

	//show the user a detailed error page
	OC_Template::printExceptionErrorPage($ex, 503);
} catch (\OC\HintException $ex) {
	try {
		OC_Template::printErrorPage($ex->getMessage(), $ex->getHint(), 503);
	} catch (Exception $ex2) {
		try {
			\OC::$server->getLogger()->logException($ex, array('app' => 'index'));
			\OC::$server->getLogger()->logException($ex2, array('app' => 'index'));
		} catch (Throwable $e) {
			// no way to log it properly - but to avoid a white page of death we try harder and ignore this one here
		}

		//show the user a detailed error page
		OC_Template::printExceptionErrorPage($ex, 500);
	}
} catch (\OC\User\LoginException $ex) {
	OC_Template::printErrorPage($ex->getMessage(), $ex->getMessage(), 403);
} catch (Exception $ex) {
	\OC::$server->getLogger()->logException($ex, array('app' => 'index'));

	//show the user a detailed error page
	OC_Template::printExceptionErrorPage($ex, 500);
} catch (Error $ex) {
	try {
		\OC::$server->getLogger()->logException($ex, array('app' => 'index'));
	} catch (Error $e) {
		http_response_code(500);
		header('Content-Type: text/plain; charset=utf-8');
		print("Internal Server Error\n\n");
		print("The server encountered an internal error and was unable to complete your request.\n");
		print("Please contact the server administrator if this error reappears multiple times, please include the technical details below in your report.\n");
		print("More details can be found in the webserver log.\n");

		throw $ex;
	}
	OC_Template::printExceptionErrorPage($ex, 500);

MichaIng, 2019-07-30:
@aspinks 
Hmm when I access from within the XRDP session via browser and open `http://localhost/nextcloud` it shows the correct login panel.

aspinks, 2019-07-30:
Ya my apologies, disregard.  that was total operator error...  Thanks so much for all of the help!

MichaIng, 2019-07-30:
@aspinks 
Okay great.

I am testing this behaviour of XRDP now on VM.

MichaIng, 2019-07-30:
@aspinks 
One other thing:
I mailed you the suggestion to remove the VNC server, but do NOT do that, as it seems to be required to allow XRDP connection: https://github.com/MichaIng/DietPi/issues/1727

MichaIng, 2019-07-30:
Issue confirmed on VM with XRDP.
Testing VNC...

MichaIng, 2019-07-30:
Lol, to check if IPv6 is solved with newer version, I installed XRDP package from stretch-backports and now I see the /etc/profile | $PATH issue is solved by this:
```
root@VM-Stretch:/tmp# cat /etc/xrdp/startwm.sh
#!/bin/sh
# xrdp X session start script (c) 2015, 2017 mirabilos
# published under The MirOS Licence

if test -r /etc/profile; then
        . /etc/profile
fi

if test -r /etc/default/locale; then
        . /etc/default/locale
        test -z "${LANG+x}" || export LANG
        test -z "${LANGUAGE+x}" || export LANGUAGE
        test -z "${LC_ADDRESS+x}" || export LC_ADDRESS
        test -z "${LC_ALL+x}" || export LC_ALL
        test -z "${LC_COLLATE+x}" || export LC_COLLATE
        test -z "${LC_CTYPE+x}" || export LC_CTYPE
        test -z "${LC_IDENTIFICATION+x}" || export LC_IDENTIFICATION
        test -z "${LC_MEASUREMENT+x}" || export LC_MEASUREMENT
        test -z "${LC_MESSAGES+x}" || export LC_MESSAGES
        test -z "${LC_MONETARY+x}" || export LC_MONETARY
        test -z "${LC_NAME+x}" || export LC_NAME
        test -z "${LC_NUMERIC+x}" || export LC_NUMERIC
        test -z "${LC_PAPER+x}" || export LC_PAPER
        test -z "${LC_TELEPHONE+x}" || export LC_TELEPHONE
        test -z "${LC_TIME+x}" || export LC_TIME
        test -z "${LOCPATH+x}" || export LOCPATH
fi

if test -r /etc/profile; then
        . /etc/profile
fi

test -x /etc/X11/Xsession && exec /etc/X11/Xsession
exec /bin/sh /etc/X11/Xsession
```

@aspinks 
Please do the following:
```
cat << _EOF_ > /etc/apt/preferences.d/dietpi-xrdp
Package: xrdp
Pin: release n=stretch-backports
Pin-Priority: 501
_EOF_
apt update
apt install xrdp
rm /etc/bashrc.d/01*
```
IPv6 issue solved by this as well 👍 Will add pulling this from backports on DietPi by default.

MichaIng, 2019-07-30:
PR up to fix this on fresh installs as well as next DietPi-Update: https://github.com/MichaIng/DietPi/pull/3020

closed 2019-10-15 by MichaIng