MichaIng/DietPi #3019: Buster Image - SCP not enabled by default

trueaspects, 2019-07-30:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.25.3
- Distro version | buster
- Kernel version | 4.19.58-v7l+
- SBC device | RPi4

Started from a fresh image of Buster.  Attempted to use winSCP but repeatedly got error until i changed software to openSSH. From other posts, it appeared that SCP should be enabled by default for Stretch and beyond. 

(side note re dietpi-config/Performance Options)
Should the default ARM Temp Limit be 65'c for a RPI4?
This is the default currently when browsing options.

MichaIng, 2019-07-30:
@trueaspects 
Many thanks for your report.

Jep SCP is and was never enabled by default, same as SFTP, as DietPi comes with Dropbear which simply does not include this features.

There are ways to install the OpenSSH sftp-server (=scp as well) as standalone binary to be used with Dropbear, which er probably will implement as an option at later time.

However for now indeed switching to OpenSSH is the way to enable SCP + SFTP.

> From other posts, it appeared that SCP should be enabled by default for Stretch and beyond.

On which other posts you found this? Raspbian an Debian do no come with SSH at all enabled by default. When installing OpenSSH, SFTP (thus SCP) is disabled by default and needs to be enabled via config file, AFAIK.

> Should the default ARM Temp Limit be 65'c for a RPI4?

This has actually not yet been decided. On RPi3 it has been raised to 75°C. I read much reviews that state that RPi4 heats up much slower then RPi3, but our benchmarks show the opposite result: https://dietpi.com/survey/#benchmark
So yeah I think we should raise it to 75°C as well.

trueaspects, 2019-07-30:
Thank you. I may have been mistaken but I thought starting from stretch that SFTP/SCP may have been enabled. 

As for 65'c, does that mean that RPi4 will throttle beginning at 65'c? I thought in Raspbian, RPi4 does not throttle until 80'c.

MichaIng, 2019-07-30:
@techdabbler 
See here: https://github.com/MichaIng/DietPi/issues/356#issuecomment-222128177
It starts to throttle down some degrees earlier. The closer it comes to the temp limit, the harder it clocks down. I guess if the limit would still be reached, it would clock down to the defined minimum temp (as well config.txt). So the limit under normal circumstances should be really the max temperature ever reached (asymtote).

Raspbian default temp limit is indeed 85°C so it most likely does not start throttling until 80°C, however I don't know if you ever had an 80°C core in a plastic case that starts to smell. I would do anything to keep the temperature below 65°C, large passiv cooler or fan and or course case that does not dam the heat.

MichaIng, 2019-08-11:
Okay I mark this issue as closed. Feel free to reopen if required.

Btw some other issue about RPi4 idle temperatures where we could collect some steps to lower power consumption => temps: https://github.com/MichaIng/DietPi/issues/3046

closed 2019-08-11 by MichaIng

xekon, 2019-12-17:
If its just a text/config file you can use cat to copy the file to the remote server using only cat, ssh, and pipes:

https://superuser.com/questions/291423/how-to-copy-a-file-without-using-scp-inside-an-ssh-session

Also DietPi does appear to have rsync support, but if scp is necessary then yes, openssh.

Joulinar, 2022-08-19:
> Why make our lives hard by not having SCP/SFTP enabled by default?

Our images are shipped with the smallest number of packages. Not everybody needs to use SCP/SFTP

> Or it could be a step in the already lengthy config process.

If needed this can be done during initial setup process or using `dietpi.txt` configuration file beforehand

> Manually replacing Dropbear with OpenSSH is not a small task, especially if you do it over SSH.

Not true. Quite a straightforward process if you use `dietpi-software` to switch SSH server. As well SSH server could be changed inside `dietpi.txt` before first boot process.

MichaIng, 2022-08-19:
> Manually replacing Dropbear with OpenSSH is not a small task, especially if you do it over SSH.

The menu option is right there at first login, and it can be switched regardless whether you are currently connected though SSH or not: The current session isn't killed even when purging an SSH server and installing a new one 🙂.