MichaIng/DietPi #3021: Can't see ownCloud in a list of software

vldslvmgy, 2019-08-01:
Can't see ownCloud in a list of software. Is this a known issue?

MichaIng, 2019-08-01:
@vldslvmgy 
Which SBC and distro version you run? `echo $G_HW_MODEL_DESCRIPTION $G_DISTRO_NAME`

ownCloud does currently not support PHP7.3. This is planned for next release v10.3, which should be soon: https://owncloud.org/download/
Because of this on Buster, where PHP7.3 is available only, it is not supported and on all other systems, it will disappear, when you install any other PHP-depending software title, as we install PHP7.3 there as well via sury.org repo.

So in fact currently you can only install ownCloud on Stretch systems if PHP was not installed before.
The same applies for phpBB, as well there the next release v7.3 fixes this: https://www.phpbb.com/

Use Nextcloud instead, which offers very similar features.

MichaIng, 2019-08-29:
I mark this as closed, question should have been answered. Feel free to reopen if required.

closed 2019-08-29 by MichaIng

MichaIng, 2019-10-21:
@vldslvmgy 
You can do now the following which allows you to install ownCloud on ARMv6 RPi models:
```
wget https://raw.githubusercontent.com/MichaIng/DietPi/owncloud/dietpi/dietpi-software -O /DietPi/dietpi/dietpi-software
```

I will keep this branch active until v6.27 release, since it does not contain any other changes yet, hence is safe to use on v6.26.

vldslvmgy, 2019-10-22:
got it, thanks