MichaIng/DietPi #3022: DietPi-Software | XRDP: Second attempt Xorg login without VNC server

MichaIng, 2019-08-01:
Initial issue: https://github.com/MichaIng/DietPi/issues/1727
- Since Stretch, [xorgxrdp](https://packages.debian.org/stretch/xorgxrdp) drivers are no fix part of the [xrdp](https://packages.debian.org/stretch/xrdp).
- Thus it needs to be installed separately so allow login via "Xorg" without VNC server.
- During my tests it then shows the desktop, but cursor and keyboard is not functional.
- Tested on multiple servers (Stretch + Buster, VM + PC), multiple clients (also freerdp client from DietPi system), install of whole [xorg](https://packages.debian.org/stretch/xorg) package and some other packages to be sure nothing is missing.
- Due to the vast amount of tests it is nearly assured that the issue is the xorgxrdp drivers and/or how they are configured by default on Debian.
- Needs forwarding to xrdp devs.

Workaround via VNC server and Xvnc login works well on all setups.

MichaIng, 2020-10-16:
A solution to test: https://github.com/neutrinolabs/xorgxrdp/issues/164

MichaIng, 2020-11-03:
Strange, now that a solution for the broken keyboard and mouse input has been found, on all my VMs (all Debian versions) it works  OOTB. Will run some more tests on RPi and another SBC environment to be sure. If anyone else is in mood:
```
apt install xorgxrdp
```
Then connect via Windows Remote Desktop (or FreeRDP) not via `Xvnc` but via `Xorg` method.

Joulinar, 2020-11-03:
@MichaIng 
it's working fine on my Pi3B+

MichaIng, 2020-11-05:
Tested on another notebook and here it does not work OOTB. Followed the above linked solution and added input device options to `/etc/X11/xrdp/xorg.conf`:
```
Section "InputDevice"
    Identifier "xrdpKeyboard"
    Driver "xrdpkeyb"
    Option "CoreKeyboard"
EndSection

Section "InputDevice"
    Identifier "xrdpMouse"
    Driver "xrdpmouse"
    Option "CorePointer"
EndSection
```
That solved the issue. I think to be sure, we should add this during install (and patch) as well:
```
grep -q 'Option "CoreKeyboard"' /etc/X11/xrdp/xorg.conf || sed -i '/^[[:blank:]]*Driver "xrdpkeyb"/a\    Option "CoreKeyboard"' /etc/X11/xrdp/xorg.conf
grep -q 'Option "CorePointer"' /etc/X11/xrdp/xorg.conf || sed -i '/^[[:blank:]]*Driver "xrdpmouse"/a\    Option "CorePointer"' /etc/X11/xrdp/xorg.conf
```

MichaIng, 2020-11-16:
Applied the above workaround now as well, to be failsafe:
https://github.com/MichaIng/DietPi/commit/6cad568dd91eead2616c9760e27b70e706643b04
https://github.com/MichaIng/DietPi/commit/fad46dae11ffeffa3dfc62f376c1b083a069df90

closed 2020-12-20 by MichaIng