MichaIng/DietPi #3023: dietpie-update cannot download files while connected to internet

enwi, 2019-08-01:
#### Details:
- Date           | Thu  1 Aug 19:42:44 CEST 2019
- Bug report     | N/A
- DietPi version | v6.24.1 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid XU3/XU4/HC1/HC2 (armv7l) (index=11)
- Kernel version | #1 SMP PREEMPT Thu Apr 5 12:46:33 UTC 2018
- Distro         | stretch (index=4)
- Command        | Connection test: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
- Exit code      | 4
- Software title | DietPi-Set_software

#### Steps to reproduce:
run dietpie-update

#### Expected behaviour:
Update should not have an error

#### Actual behaviour:
Updater cannot connect to some websites, even though the internet connection is working and has been tested during the update from another terminal, downloading the exact same files as in the update.

#### Extra details:
Even though this warning appeared the system is now at version v6.25.3 which is the current version and was before at v6.24.1

#### Additional logs:
```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2019-08-01 19:41:00--  https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... failed: Connection timed out.
wget: unable to resolve host address ‘raw.githubusercontent.com’
```

```
[FAILED] DietPi-Survey | Connection test: ssh.dietpi.com
```

MichaIng, 2019-08-01:
@enwi 
Many thanks for your report.

> Connection timed out.

Please go to `dietpi-config` > `Network Options: Misc` > `G_CHECK_URL timeout` and raise the value.

The update should then work.
Afterwards I would play around the DNS resolver since > 5 seconds is about 10 times more than usual.

To test the time is requires for resolving:
```
time wget --spider https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
```
For quick testing you could adjust the nameserver entry in `/etc/resolv.conf`.
Try your local router in the first place. Try 8.8.8.8 (Google DNS) if that too long.

But there are known cases where for unknown some reason DNS resolving takes that long on Debian, hence we added the `dietpi-config` option to adjust the timeout.

enwi, 2019-08-01:
@MichaIng So I changed the first dns server to be my router and now resolving is a lot faster.

Before adding router ip:
```
time wget --spider https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
Spider mode enabled. Check if remote file exists.
--2019-08-01 20:11:22--  https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.0.133, 151.101.64.133, 151.101.128.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.0.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 11949 (12K) [text/plain]
Remote file exists.

real    0m5.385s
user    0m0.075s
sys     0m0.050s
```

After adding router ip:
```
time wget --spider https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
Spider mode enabled. Check if remote file exists.
--2019-08-01 20:13:19--  https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.112.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.112.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 11949 (12K) [text/plain]
Remote file exists.

real    0m0.168s
user    0m0.045s
sys     0m0.030s
```

closed 2019-08-01 by enwi

MichaIng, 2019-08-01:
@enwi 
Perfect, yeah usually the router servers as local DNS resolver with DNS cache, so that should be fastest in most cases.

Since updating dietpi.txt failed during update, please run this task manually:
```
/DietPi/dietpi/func/dietpi-set_software verify_dietpi.txt
```

enwi, 2019-08-01:
@MichaIng I reran the command, the result was 
```
[  OK  ] DietPi-Set_software | Root access verified.
[  OK  ] DietPi-Set_software | RootFS R/W access verified.

 DietPi-Set_software
─────────────────────────────────────────────────────
 Mode: verify_dietpi.txt ()

[  OK  ] DietPi-Set_software | Connection test: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
[  OK  ] DietPi-Set_software | Downloading current dietpi.txt
[ INFO ] DietPi-Set_software | Updating dietpi.txt with new entry: AUTO_SETUP_AUTOSTART_LOGIN_USER=root
[  OK  ] DietPi-Set_software | dietpi.txt verification completed
[  OK  ] verify_dietpi.txt  | Completed
```
So I guess everything is fine now, and as always thanks for the fast and awesome help 👍