MichaIng/DietPi #3024: DietPi-Software | Pi-hole: Block access to blocking page from web by default

MichaIng, 2019-08-01:
Ref: https://dietpi.com/phpbb/viewtopic.php?p=18964#p18964

AtkLordOverAll, 2019-08-09:
Hey @MichaIng 

Happy to take a look at this one if you'd like - do I just need to slap your suggested config into `/etc/lighttpd.conf`? If yes, a few more questions so I can make this happen:

1. Is lighttpd.conf's file using regex or globs? I'll see if I can compact the expressions you're using further to save redundant code

2. Does the config need to be different if a user has Nextcloud installed or not?

3. Is it safe/okay to assume that all internal devices will be on the same third part (for lack of a better term) IP? i.e. if the pi-hole is 192.168.**0**.2 can we assume all other local devices are 192.168.**0**.x, as opposed to something like 192.168.**1**.100? (note the bolded digits)

And just to note, I don't actually have a setup with which to test this, I'm just happy to make the changes for someone to check/approve.

MichaIng, 2019-08-10:
@AtkLordOverAll 
Many thanks, would be great of course. However we should add this not to the base webserver config, instead via drop-in configs/modules, e.g.
```
/etc/lighttpd/conf-available/dietpi-pihole.conf
lighttpd-enable-mod dietpi-pihole
```
something like this and of course to all three webservers.

> Is lighttpd.conf's file using regex or globs?

Jep they all support regex, in case with some additional syntax, e.g. `<string> ~ <pattern>` in Nginx.
Note that the config part in forum thread was to block access to EVERYTHING but Nextcloud. We only want to block access to the Pi-hole blocking page. So the location match should be the outer/first match `^(/html)?/pihole` and local IP inner match.

> Does the config need to be different if a user has Nextcloud installed or not?

No, with above outer location match this affects `/pihole` (symlink) and `/html/pihole` locations only, so there can be no conflict with other websites.

> Is it safe/okay to assume that all internal devices will be on the same third part (for lack of a better term) IP? 

Usually local networks are 8 bit only, so yeah should be safe to assume. A 16 bit local network should be only present in cases where one will be able to configure the webserver oneself correctly 😉.
But the issue is that one might want to see the blocking page as well when connecting through a VPN.
What I though about is checking against all official local IP addresses: https://en.wikipedia.org/wiki/Private_network
This is not too heavy actually: `(192.168|10|172.(1[6-9]|2[0-9]|3[0-1])).*`
For IPv6 addresses `fd??:*` should be correct.
Perhaps some webserver has an internal syntax/variable for this as well, not sure.

AtkLordOverAll, 2019-08-10:
Okay most of this makes sense, seems I opened a bigger can of worms than I anticipated.

> all three webservers

I'm not familiar with lighttpd, I'm just happy to drop in config files with certain contents and I know my way around a regex. What would be the 3 webservers I need to add these changes to and how do I do that? For applying the "mod" I'd be making, presumably this'd be something to do on install in `dietpi-software`?

> block access to EVERYTHING but Nextcloud

Gotcha.

> yeah should be safe to assume

Awesome.

>`(192.168|10|172.(1[6-9]|2[0-9]|3[0-1])).*`
> `fd??:*`

You're too kind, this was the bit I was trying to do 😆

I'll start work and open the PR probably before I consider things ready to merge so you can double check I'm going in the right direction and fill in the bits I don't get, as I have no intimate experience with lighttpd except having Pi-hole installed myself. Hope that's alright 😄

MichaIng, 2019-08-10:
@AtkLordOverAll 
There is already a config we place when installing Pi-hole. The blocking page handling can just be added there: https://github.com/MichaIng/DietPi/blob/dev/.conf/dps_93/lighttpd.pihole.conf

Aj I recognise that this currently applies only when accessing via symlink, but it should apply as well when accessing via /html/admin... however I can take care of this.

> What would be the 3 webservers

Apache, Nginx and Lighttpd are the three we support currently, so generally it makes sense to add software specific configs for all of them. But it is okay as well to do a start with one webserver and we can add those for the others at a later time.
For Lighttpd this is quite easy now since it already exists and is implemented into DietPi-Software.

AtkLordOverAll, 2019-08-10:
> There is already a config we place when installing Pi-hole. The blocking page handling can just be added there

Huh, this makes my life simple

> Apache, Nginx and Lighttpd

Gotcha, makes the sense.

Shall get to work soon, doesn't actually look like there's too much to do, all in all.

MichaIng, 2019-08-29:
General blocking page implementation fix + apply access permissions for all webservers: https://github.com/MichaIng/DietPi/pull/3072

closed 2019-10-15 by MichaIng

anubis-genix, 2019-11-06:
Did a fresh install with lighttpd as webserver and everything worked as intended. Nextcloud was running great, the Pi-Hole admin panel was only accessible from the internal network and subdomain.domain.tld gave a 403 access forbidden. Unfortunately my password manager wouldn't establish a webdav connection with lighttpd, so I was forced to switch to nginx.
Back on nginx I first installed Nextcloud, Certbot and Pi-Hole. Even though I denied the access from external to the Pi-Hole admin panel during the initial installation, it's still accessible from outside and subdomain.domain.tld gives a welcome to nginx page.

MichaIng, 2019-11-06:
@anubis-genix 
Many thanks for your report. Did you fully uninstall Lighttpd first?
Please paste:
```
cat /etc/nginx/sites-dietpi/dietpi-pihole-block_public_admin.on
```

MichaIng, 2019-11-07:
@anubis-genix 
> It looks like -block_public_admin.on didn't get set on initial configuration of Pi-Hole.

So `/etc/nginx/sites-dietpi/dietpi-pihole-block_public_admin.off` still exist instead? I just found a bug that it is enabled actually in every case due to `;` in the command which breaks the yes/no choice. But then its even stranger that you needed to enable manually.
But the choice prompt did appear where you could choose to enable/disable admin panel block, right?
I'm spinning up a VM to test this now.

> Btw. any idea how to set the nginx welcoming page to 403 access forbidden?

The webroot index page is removed when you choose to enable the blocking page, otherwise it remains: `/var/www/index.nginx-debian.html` (or any other index.* file there)

anubis-genix, 2019-11-07:
@MichaIng 
> So `/etc/nginx/sites-dietpi/dietpi-pihole-block_public_admin.off` still exist instead? I just found a bug that it is enabled actually in every case due to `;` in the command which breaks the yes/no [...]
The dietpi-pihole-block_public_admin.off still existed even though I've chosen to enable the admin panel block from outside my LAN via the prompt. Due to the bug you mentioned it should have been set? That's really weird!
> But the choice prompt did appear where you could choose to enable/disable admin panel block, right?
That's right. It appeared but my preferred setting didn't get set.

MichaIng, 2019-11-07:
@anubis-genix 
Found the issue, resolved with: https://github.com/MichaIng/DietPi/commit/14cd9726bfb8407966c3f230ff69601c02c21596
- When using a variable to call commands, `;` `&&` `||` `|` and all those command "connectors" are not interpreted as such but as additional command arguments, hence for Nginx where the initial command was to navigate to configs dir (to keep the line shorter) it failed.
- I used now another method to fix the issue but even reducing line size 🙂.

The following can be used to enable admin panel block on Nginx:
```
mv /etc/nginx/sites-dietpi/dietpi-pihole-block_public_admin.{off,on}
```
To disable:
```
mv /etc/nginx/sites-dietpi/dietpi-pihole-block_public_admin.{on,off}
```
And of course to reload config:
```
systemctl reload nginx
```