MichaIng/DietPi #3028: Kodi not working on Odroid c2

alexeylutskov, 2019-08-03:
Fresh install, Kodi and Mpd only,  after all things done just screen blink and message that Kodi can't create gui (Kodi autorun enabled). Same time Kodi runs perfect on Coreelec.

MichaIng, 2019-08-03:
@alexeylutskov 
Many thanks for your report.

I'll check our driver installs, I remember Meveric asked for Kodi 18 testers, was this probably released now? Which Kodi version is it? `dpkg -l kodi-odroid`
In case this might probably need some change in drivers/configs.

alexeylutskov, 2019-08-03:
dpkg -l kodi-odroid
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                 Version                 Architecture            Description
+++-====================================-=======================-=====[=================-=============================================================================
ii  kodi-odroid                          17.6-5+deb9             arm64                   Kodi - award-winning free and open source (GPL) software media player and ent

MichaIng, 2019-08-04:
@alexeylutskov 
Okay I made some comparison of Kodi, desktop and X.org install steps between dietpi-software and the setup-odroid tool by Meveric, that should definitely work with the packages provided by his repo.

Differences are tiny, however worth to give it a try:
```
wget https://raw.githubusercontent.com/MichaIng/DietPi/dev/.conf/dps_6/xorg_c2.conf -O /etc/X11/xorg.conf
apt purge xf86-video-mali-odroid # Check that there are not other packages depending on it
apt purge pulseaudio # We added PA to install some time ago to fix corrupt sound, however it should be not required (anymore) and might even conflict with our default ALSA setup
apt autoremove --purge
apt install --reinstall aml-libs-odroid mali450-odroid libump-odroid xf86-video-fbturbo-odroid
```

Failsafe reboot into console:
```
/DietPi/dietpi/dietpi-autostart 0
reboot
```

Then please try to start Kodi from console:
```
xinit kodi-standalone
```

alexeylutskov, 2019-08-25:
Solution works! Kodi works fine for me for now.

MichaIng, 2019-08-26:
@alexeylutskov 
Okay perfect. Not sure which of these more-minor changes did the trick, however all of them are implemented in our install code with v6.26.

Does `dietpi-autostart` into Kodi also work well now?

MichaIng, 2019-08-26:
@alexeylutskov 
Many thanks for testing. We'll apply most of the steps during next DietPi-Update. All but purging PulseAudio can be done safely, in case of PulseAudio we should ask user.

alexeylutskov, 2019-08-26:
Yes, it works, thank you.

MichaIng, 2019-09-02:
Done: https://github.com/MichaIng/DietPi/commit/3a53ef75d9502fcee3f4d8f2a939263bd3609c98
Changelog: https://github.com/MichaIng/DietPi/commit/57916d1face7a9dba9fc46d7ef54f9301a9716e2
Covers as well: https://github.com/MichaIng/DietPi/pull/3035

alexeylutskov, 2019-09-03:
Thank you, great news!

closed 2019-10-15 by MichaIng