MichaIng/DietPi #3029: Automated Sonarr Installation Fails

GregoryDosh, 2019-08-03:
#### Details:
- Date           | Sat  3 Aug 12:15:04 PDT 2019
- Bug report     | N/A
- DietPi version | v6.25.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | #1244 SMP Thu Jul 4 18:45:25 BST 2019
- Distro         | buster (index=5)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:
1. Added Sonarr to automated install process in dietpi.txt
2. Add SD card to Raspberry Pi & wait for boot

#### Expected behaviour:
- Should automatically install Sonarr during the installation process

#### Actual behaviour:
- Installation loop due to an inability to find a signed GPG key

#### Extra details:
- ...

#### Additional logs:
```
Log file contents:
Hit:1 https://download.mono-project.com/repo/debian raspbianstretch InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Get:3 https://apt.sonarr.tv develop InRelease [13.3 kB]
Hit:4 https://archive.raspberrypi.org/debian buster InRelease
Err:3 https://apt.sonarr.tv develop InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY EBFF6B99D9B78493
Reading package lists...
W: GPG error: https://apt.sonarr.tv develop InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY EBFF6B99D9B78493
E: The repository 'https://apt.sonarr.tv develop InRelease' is not signed.
```

GulyFMG, 2019-08-03:
Im also trying to update my build because my SD card just died and im having the same problem...
Just checked the instructions to sonarr and they have:

`sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys FDA5DFFC
echo "deb https://apt.sonarr.tv/ master main" | sudo tee -a /etc/apt/sources.list.d/sonarr.list`

and the error is showing:
`The repository **'https://apt.sonarr.tv develop InRelease'**`

MichaIng, 2019-08-04:
@GregoryDosh @GulyFMG 
Many thanks for your report.

Lol I just moved to the long key version some days ago, since I found this short key not fitting modern security standards: https://github.com/MichaIng/DietPi/commit/a3b0f7ae447fd01ee5d2d0e5c92945c877bd2f2c#diff-d92a6ee04e02fd2a2dc23d5bec3e6a98R5456
Close to good timing now that they obviously dropped support for the short version key, at least this is very likely the issue here. I just tested long key on Stretch and Buster VM at least.

Please run:
```
apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 0xA236C58F409091A18ACA53CBEBFF6B99D9B78493
G_AGUP
G_AGI nzbdrone
```

If this works, then re-run the install:
```
dietpi-software install 144
```
________________
> apt-key adv --keyserver keyserver.ubuntu.com --recv-keys FDA5DFFC

This is what we have in master code as well, but it is obviously either outdated repo-side or possibly not allowed by Buster (although worked in the past) anymore.
Here you find up-to-date Sonarr install instructions, that match what we have in dev code now: https://github.com/Sonarr/Sonarr/wiki/Installation#linux

> https://apt.sonarr.tv develop InRelease

And yes we use the develop version, which due to our testing and user experience provides the better share between stability and feature/security/bugfix updates.

GulyFMG, 2019-08-04:
@MichaIng 

Yep i also tested and its working as intended now.
thank very much for the prompt reply.

MichaIng, 2019-08-04:
@GulyFMG 
Perfect, many thanks for testing. I mark this issue as closed then. Both of you, feel free to reopen if required,

closed 2019-08-04 by MichaIng

snoodler, 2020-01-02:
just got this trying to install jackett:

#### Details:
- Date           | Thu  2 Jan 21:30:01 GMT 2020
- Bug report     | 6fdb9c44-6939-4839-9232-1aa88405a615
- DietPi version | v6.27.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | tar xf 147.tar --one-top-level=/opt
- Exit code      | 2
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```

gzip: stdin: unexpected end of file
tar: Unexpected EOF in archive
tar: Unexpected EOF in archive
tar: Error is not recoverable: exiting now
```

MichaIng, 2020-01-02:
@snoodler 
Strange, I just tested the archive that is downloaded on Jackett:
```
https://github.com/Jackett/Jackett/releases/download/v0.12.1398/Jackett.Binaries.LinuxARM32.tar.gz
```

Could you verify this works:
```
cd /tmp
wget https://github.com/Jackett/Jackett/releases/download/v0.12.1398/Jackett.Binaries.LinuxARM32.tar.gz
tar xf Jackett.Binaries.LinuxARM32.tar.gz --one-top-level=/opt
```

snoodler, 2020-01-03:
download stopped at 84% with this error:

2020-01-03 12:21:43 (6.52 KB/s) - Read error at byte 38593082/45736441 (Error in the pull function.). Retrying.

--2020-01-03 12:21:44--  (try: 2)  https://github-production-release-asset-2e65be.s3.amazonaws.com/45195614/456e7f80-2d29-11ea-9430-f0240eb4472b?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200103%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200103T104507Z&X-Amz-Expires=300&X-Amz-Signature=5c487379b690f1774dd403b35e7aba55bf7950aff1400906bfddabe77f4bca23&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DJackett.Binaries.LinuxARM32.tar.gz&response-content-type=application%2Foctet-stream
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)|52.216.206.3|:443... connected.
HTTP request sent, awaiting response... 403 Forbidden
2020-01-03 12:21:46 ERROR 403: Forbidden.

MichaIng, 2020-01-03:
@snoodler 
> Read error at byte 38593082/45736441 (Error in the pull function.)

Not sure what this means. Could you open a separate issue?
That the second attempt fails makes sense, as the URL looks like regenerated with access token, hence its only allowed to be used once as a redirect result and not a second time.

snoodler, 2020-01-04:
I will, thanks