MichaIng/DietPi #3031: Kodi doesn't start - ERROR: Unable to create GUI. Exiting

GulyFMG, 2019-08-03:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=25
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
10.0
- Kernel version | `uname -a`
Linux blacksun 4.19.58-v7+ #1245 SMP Fri Jul 12 17:25:51 BST 2019 armv7l GNU/Linux
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
RPi 3 Model B+ (armv7l)
- Power supply used 5V 
- SDcard used | (EG: SanDisk ultra 32Gb

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

ac116c8e-1087-479b-ab4e-524d0d415e7d
- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce
NEw image installed and just installed KODI, since sonarr is not working either.

#### Expected behaviour
should start

#### Actual behaviour

does not start
error message - ERROR: Unable to create GUI. Exiting
munmap_chunk(): invalid pointer

#### Extra details

[kodi_crashlog-20190803_224216.log](https://github.com/MichaIng/DietPi/files/3464365/kodi_crashlog-20190803_224216.log)

MichaIng, 2019-08-04:
@GulyFMG 
Yeah sadly this is known since the RPi-compatible Kodi 18 from archive.raspberrypi.org repo has not yet been ported to the Buster branch. So DietPi-Software currently installs the Kodi package from Raspbian Buster, which at least most likely does not support full hardware acceleration.

However you should be able to start this Kodi version either from inside a running desktop via `kodi` or from console (outside of desktop) via `xinit kodi-standalone`.

Kodi 18 is expected to be ported soon which will then resolve the issue.

GulyFMG, 2019-08-04:
Thanks for the info... I went back to a legacy image because this is like a "production" system.

It was a lot of stuff together the SD card was corrupted the backup was a couple of versions outdated and so and so...

derdershat, 2019-08-04:
if you want a version for the meantime it is in official raspberry buster repo you could use this
[https://www.raspberrypi.org/forums/memberlist.php?mode=viewprofile&u=103676](url)

If I understood it right, Rascas Kodi version is the version that will be in official repo than.
He has vacation time...

MichaIng, 2019-08-04:
@GulyFMG 
 > I went back to a legacy image because this is like a "production" system.

Note that Raspbian Stretch does not receive much updates anymore and other things will start to break, as I will not invest much time any more to fix things on Raspbian Stretch (arising as external issues) after official Buster release.

> if you want a version for the meantime it is in official raspberry buster repo you could use this

Many thanks for this info! Perhaps I will add this repo to our installer, if until v6.26 release none has arrived on Raspbian Buster official list yet.
No RPi4 support 🤔. But there is another custom package compatible with RPi4: https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=192499&start=200#p1507286
I am currently downloading it and check contained binaries. However is getting complicated to add this to the installer if official repo package is expected soon after.

Will go with an informational issue with the workarounds and a related MOTD.

GulyFMG, 2019-08-04:
That's not a problem right now I have a spare SD card I'm gonna prepare a buster image and check regularly here for issues and stuff.
As soon as I feel that it's stable enough I will migrate.

Thanks for the heads-up.

derdershat, 2019-08-04:
> No RPi4 support 🤔. But there is another custom package compatible with RPi4: https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=192499&start=200#p1507286
> I am currently downloading it and check [...]

for PI4 I have installed the version from your link and added infos on Friday in [DietPi forum](https://dietpi.com/phpbb/viewtopic.php?f=9&t=6065&p=19014#p19014)

MichaIng, 2019-08-04:
@derdershat 
Ah very nice, I did not find the time lately to go through all forum threads, have to redo that.
I am wondering if this version is RPi4 only or works on lower versions as well 🤔. Would be easier to provide one meantime solution for all Buster users.

derdershat, 2019-08-04:
> 
> 
> @derdershat
> Ah very nice, I did not find the time lately to go through all forum threads, have to redo that.
> I am wondering if this version is RPi4 only or works on lower versions as well 🤔. Would be easier to provide one meantime solution for all Buster users.

as I read [here](https://github.com/popcornmix/omxplayer/issues/714#issuecomment-513841111) it is compiled from https://github.com/popcornmix/xbmc/tree/leia_pi4 branch

would wait for Rascas / official version and give an info for PI4 users that there is an other version for the meantime.

MichaIng, 2019-08-04:
Okay so workaround on DietPi Buster for RPi:
Install via `dietpi-software` first to have X.org installed, have it correctly marked and config steps applied. Then:
#### On RPi4
```
cd /tmp
wget https://bit.ly/2LIaq7F -O kodi.deb
dpkg -i kodi.deb
G_AGF
sed -i 's/kodi-standalone/kodi-rpi4/' /DietPi/dietpi/misc/start_kodi
```
#### On all other RPi models
```
wget -q -O- http://pipplware.pplware.pt/pipplware/key.asc | apt-key add -
echo 'deb http://pipplware.pplware.pt/pipplware/dists/buster/main/binary /' > /etc/apt/sources.list.d/pipplware.list
G_AGUP
G_AGI kodi
```

We'll inform via MOTD once Kodi 18 has been added to official Raspberry Pi Buster repo.

jason-a69, 2019-08-08:
> 
> 
> Okay so workaround on DietPi Buster for RPi:
> Install via `dietpi-software` first to have X.org installed, have it correctly marked and config steps applied. Then:
> #### On RPi4
> 
> ```
> cd /tmp
> wget https://bit.ly/2LIaq7F -O kodi.deb
> dpkg -i kodi.deb
> G_AGF
> sed -i 's/kodi-standalone/kodi-rpi4/' /DietPi/dietpi/misc/start_kodi
> ```
> 
> #### On all other RPi models
> 
> ```
> wget -q -O- http://pipplware.pplware.pt/pipplware/key.asc | apt-key add -
> echo 'deb http://pipplware.pplware.pt/pipplware/dists/buster/main/binary /' > /etc/apt/sources.list.d/pipplware.list
> G_AGUP
> G_AGI kodi
> ```
> 
> I added this info to MOTD for RPi Buster systems, which will be shown after next daily cron job.
> We'll inform via MOTD (and here) once Kodi 18 has been added to official Raspberry Pi Buster repo.

After you have done the install you may need to run dietpi-config -> Display Options -> GPU/RAM Split -> 320 : Kodi

I was getting problems starting the Xserver until I made that change.

This was on a Raspberry Pi 3

MichaIng, 2019-08-08:
@jason-a69 
This should be actually included in the `dietpi-software` install step: https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-software#L12079-L12084

jonyskids, 2019-09-02:
I get:

─────────────────────────┤ DietPi Error Handler: ├──────────────────────────────
g-deb: error: paste subprocess was killed by signal (Broken pipe)^M
paring to unpack .../kodi-bin_2%3a18.3-2~buster_armhf.deb ...^M
acking kodi-bin (2:18.3-2~buster) over (2:17.6+dfsg1-4+b1) ...^M
g: error processing archive
r/cache/apt/archives/kodi-bin_2%3a18.3-2~buster_armhf.deb (--unpack):^M
ying to overwrite '/usr/bin/kodi-standalone', which is also in package kodi
7.6+dfsg1-4+b1^M
ors were encountered while processing:^M
ar/cache/apt/archives/kodi_2%3a18.3-2~buster_all.deb^M
ar/cache/apt/archives/kodi-bin_2%3a18.3-2~buster_armhf.deb^M
Sub-process /usr/bin/dpkg returned an error code (1)

ble to continue,  will now terminate.

When I follow these steps? Any help would be appreciated.

Jon

MichaIng, 2019-09-02:
@jonyskids
You got it solved? Just got the mail, but you removed the commend again?

In case, we would need the whole APT output to investiage:
```
apt install kodi
```

jonyskids, 2019-09-02:
─────────────────────────┤ DietPi Error Handler: ├──────────────────────────────
GI kodi
Exit code: 100
DietPi version: v6.25.3 (MichaIng/master) | HW_MODEL:3 | HW_ARCH:2 | DISTRO:5
Image creator: DietPi Core Team
Pre-image: Raspbian Lite

 file contents:
ading database ... ^M(Reading database ... 5%^M(Reading database ... 10%^M(Readi
abase ... 15%^M(Reading database ... 20%^M(Reading database ... 25%^M(Reading da
 30%^M(Reading database ... 35%^M(Reading database ... 40%^M(Reading database ..
ading database ... 50%^M(Reading database ... 55%^M(Reading database ... 60%^M(R
abase ... 65%^M(Reading database ... 70%^M(Reading database ... 75%^M(Reading da
 80%^M(Reading database ... 85%^M(Reading database ... 90%^M(Reading database ..
ading database ... 100%^M(Reading database ... 27511 files and directories curre
talled.)^M

This is what I am getting. (I noticed your comment: "After you have done the install you may need to run dietpi-config -> Display Options -> GPU/RAM Split -> 320 : Kodi" double-checked this setting and tried again. Above is the output post-second run.

MichaIng, 2019-09-02:
@jonyskids 
Please see above and paste the whole output.

jonyskids, 2019-09-02:
`root@DietPi:~# apt install kodi
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  fontconfig kodi-data libaom0 libavcodec58 libavfilter7 libavformat58
  libavutil56 libbs2b0 libcairo2 libchromaprint1 libcodec2-0.8.1 libcroco3
  libcrossguid0 libdatrie1 libfftw3-double3 libflite1 libgdk-pixbuf2.0-0
  libgdk-pixbuf2.0-common libgme0 libgsm1 libjs-iscroll libjs-jquery
  liblilv-0-0 libmp3lame0 libmpg123-0 libmysofa0 libnorm1 libopenjp2-7
  libopenmpt0 libopus0 libpango-1.0-0 libpangocairo-1.0-0 libpangoft2-1.0-0
  libpgm-5.2-0 libpostproc55 librsvg2-2 librubberband2 libserd-0-0 libshine3
  libsnappy1v5 libsodium23 libsord-0-0 libsoxr0 libspeex1 libsratom-0-0
  libssh-gcrypt-4 libswresample3 libswscale5 libthai-data libthai0 libtheora0
  libtwolame0 libva-drm2 libva-x11-2 libva2 libvdpau1 libvidstab1.1 libvpx5
  libwavpack1 libx264-155 libx265-165 libxcb-render0 libxcb-shm0 libxvidcore4
  libyajl2 libzmq5 libzvbi-common libzvbi0 shared-mime-info
Use 'apt autoremove' to remove them.
The following additional packages will be installed:
  kodi-bin
Suggested packages:
  kodi-pvr-mythtv kodi-pvr-vuplus kodi-pvr-vdr-vnsi kodi-pvr-njoy
  kodi-pvr-nextpvr kodi-pvr-mediaportal-tvserver kodi-pvr-tvheadend-hts
  kodi-pvr-dvbviewer kodi-pvr-argustv kodi-pvr-iptvsimple
  kodi-audioencoder-vorbis kodi-audioencoder-flac kodi-audioencoder-lame
Recommended packages:
  libva-intel-vaapi-driver libva1
The following packages will be upgraded:
  kodi kodi-bin
2 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
13 not fully installed or removed.
Need to get 0 B/33.5 MB of archives.
After this operation, 49.5 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
(Reading database ... 27511 files and directories currently installed.)
Preparing to unpack .../kodi_2%3a18.3-2~buster_all.deb ...
Unpacking kodi (2:18.3-2~buster) over (2:17.6+dfsg1-4+b1) ...
dpkg: error processing archive /var/cache/apt/archives/kodi_2%3a18.3-2~buster_all.deb (--unpack):
 trying to overwrite '/usr/share/man/man1/kodi.bin.1.gz', which is also in package kodi-bin:armhf 2:17.6+dfsg1-4+b1
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Preparing to unpack .../kodi-bin_2%3a18.3-2~buster_armhf.deb ...
Unpacking kodi-bin (2:18.3-2~buster) over (2:17.6+dfsg1-4+b1) ...
dpkg: error processing archive /var/cache/apt/archives/kodi-bin_2%3a18.3-2~buster_armhf.deb (--unpack):
 trying to overwrite '/usr/bin/kodi-standalone', which is also in package kodi 2:17.6+dfsg1-4+b1
Errors were encountered while processing:
 /var/cache/apt/archives/kodi_2%3a18.3-2~buster_all.deb
 /var/cache/apt/archives/kodi-bin_2%3a18.3-2~buster_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)`

Is that what you need?  (Appreciate all help. I am a novice.)

MichaIng, 2019-09-02:
@jonyskids 
Strange, these are the failures:
> trying to overwrite '/usr/share/man/man1/kodi.bin.1.gz', which is also in package kodi-bin:armhf 2:17.6+dfsg1-4+b1
trying to overwrite '/usr/bin/kodi-standalone', which is also in package kodi 2:17.6+dfsg1-4+b1

Since `The following packages will be upgraded:  kodi kodi-bin`, of course these two files are part of these other packages, but I don't get it why it fails overwriting files that are to be upgraded of course.

Could you please check if they exist, or probably not which is the issue then?
```
ls -Al /usr/share/man/man1/kodi.bin.1.gz /usr/bin/kodi-standalone
```

If they do not exist, try to create them:
```
> /usr/share/man/man1/kodi.bin.1.gz
> /usr/bin/kodi-standalone
```

If it still fails, the old Kodi packages need to be removed first:
```
apt remove kodi kodi-bin
apt install kodi
apt autoremove --purge # To remove non-required packages, leftover from the old version
```

jonyskids, 2019-09-02:
The last set of instructions did it. For reference, this was a fresh install on a RPi3. Installed Kodi,
Ran:
`apt remove kodi kodi-bin
apt install kodi
apt autoremove --purge # To remove non-required packages, leftover from the old version`

Then:

`wget -q -O- http://pipplware.pplware.pt/pipplware/key.asc | apt-key add -
echo 'deb http://pipplware.pplware.pt/pipplware/dists/buster/main/binary /' > /etc/apt/sources.list.d/pipplware.list
G_AGUP
G_AGI kodi`

Thanks for the help!

MichaIng, 2019-09-02:
@jonyskids 
Great, thanks for testing! Hope this comes now with GPU acceleration, allowing fluent full HD videos on RPi.

Out of interest were the two files I mentioned actually existing? I mean output of the `ls -Al` command?

I am thinking now how we should handle existing Kodi installs. Shall we uninstall and reinstall Kodi on all RPi Buster on next DietPi-Update? Probably not since users will recognise fast the it does not work, so either uninstall it, manage to fix it themselves, of recognise MOTD or this issue to follow instructions.

MichaIng, 2019-09-02:
@jderdershat @jonyskids 
Another question:
- On Raspbian Stretch, the Kodi 18 version required `fbset` to be installed, to allow changing resolution.
- Could you verify that either: The Kodi package pulls `fbset` as dependency on install, or otherwise, if you can either change resolution without it?

Check if fbset is installed: `dpkg -l fbset`
Remove it: `apt purge fbset`
- If APT asks you to confirm that Kodi will be removed as well, of course do not do it, then it is pulled as dependency.

MichaIng, 2019-09-02:
PR up: https://github.com/MichaIng/DietPi/pull/3092

derdershat, 2019-09-02:
> 
> 
> @jderdershat @jonyskids
> Another question:
> 
>     * On Raspbian Stretch, the Kodi 18 version required `fbset` to be installed, to allow changing resolution.
> 
>     * Could you verify that either: The Kodi package pulls `fbset` as dependency on install, or otherwise, if you can either change resolution without it?
> 
> 
> Check if fbset is installed: `dpkg -l fbset`
> Remove it: `apt purge fbset`
> 
>     * If APT asks you to confirm that Kodi will be removed as well, of course do not do it, then it is pulled as dependency.
> 
> 
> And another question about the `kodi-rpi4` binary. As far as I can see, the script, that it is, does not add the `--standalone` option. Does it nevertheless start well when using via `dietpi-autostart`, as I stated above? [#3031 (comment)](https://github.com/MichaIng/DietPi/issues/3031#issuecomment-518038469)

on my PI4 with Kodi 18.3 no fbset package found
dpkg -l fbset
dpkg-query: Kein Paket gefunden, das auf fbset passt

autostart works well with your change to kodi-rpi4 - had no problem without --standalone 
but before this change I could start it too with old start script kodi-standalone

MichaIng, 2019-09-02:
@derdershat 
Okay, and switching to another screen resolution from within Kodi GUI works as well?

Interesting. The `kodi-rpi4` script is another wapper:
```
#!/bin/sh
sudo /bin/systemctl start kodi-shell@${USER}.service
vt=$(sudo /bin/fgconsole)
sudo /bin/chvt 8
kodi "${@}"
sudo /bin/chvt ${vt}
sudo /bin/systemctl stop kodi-shell@${USER}.service
```
It starts another autologin shell session on TTY8:
```
[Unit]
Description=Kodi Shell for %I

[Service]
ExecStart=/sbin/agetty --autologin %I tty8 $TERM
```
After Kodi exits, it reverts back to the original shell.

Indeed this is not required. When running within current shell session, you will have some Kodi logs there, but that is not a bad thing, in case something crashes. Also the session is clean when starting via `dietpi-autostart`.

Actually the `kodi-standalone` "binary" is just another wrapper script as well. It does nothing else then initiating PulseAudio sessions (if PA is installed) and starts `kodi --standalone` in an error handler loop. If it crashes, it is restarted automatically multiple times. Actually not something I like very much, cause it usually just spams your log. If it crashes, usually something needs to be done instead of just restarting it directly.

So we will run `kodi --standalone` without all these additional shell wrappers, which reduces process count and RAM usage slightly and as well saves probably 1 second start time 😃.

Another question:
Pulseaudio is not installed, is it?
```
which start-pulseaudio-x11
which pulse-session
```

derdershat, 2019-09-02:
thanks for the clarification ... 

no problems with switching resolutions (have no 4k) 

and no pulseaudio installed here 

root@DietPi:~# which start-pulseaudio-x11
root@DietPi:~# which pulse-session
root@DietPi:~#

btw
version 18.4 will not be long in coming.
official source is out and another unofficial version is on its way (according to some posts in the raspberry forum)
they found out that the last version was build as debug version and hopfully the next will come without the 30-50% load on one cpu on idle
we will see ...

MichaIng, 2019-09-02:
@derdershat 
Thanks for testing and info. That is all very great. Yeah I was hoping for official Raspberry Pi repo release before DietPi v6.26, however want to release it soon, so cannot wait for it any longer.

derdershat, 2019-09-06:
unofficial Kodi 18.4 is out!

[https://www.raspberrypi.org/forums/viewtopic.php?f=38&t=246837&start=100#p1530531](url)

- MPEG-2 crash is fixed 
- it seams to be better with cpu load on idle
(depends on skin as I use Xonfluence skin and have 20% with spikes to 30% but on Estuary skin only max 13%)

- Estuary skin has some faulty dialog boxes after update. 
Could fix this by copying skin folder from my Windows Kodi after install.
do a backup before update would be recommended
path is /root/usr/share/kodi/addons/skin.estuary
or pull again from xbmc repo? 
https://github.com/xbmc/xbmc/tree/master/addons/skin.estuary

MichaIng, 2019-09-06:
@derdershat 
Perfect, I will update the link.

PanderMusubi, 2019-09-09:
Perhaps this can become also an official fix? See https://bugs.launchpad.net/raspbian/+bug/1834970 Not sure who to contact to get this fixed in the Raspbian repo for all RPi models. Anybody?

MichaIng, 2019-09-09:
@PanderMusubi 
This is exactly what this issue is about and fixed/applied for v6.26 😉.

PanderMusubi, 2019-09-09:
Super :+1: Now who can merge this in the upstream Raspbian repo, so everyone can benefit from it directly? Are those people participating in this issue here?

MichaIng, 2019-09-10:
@PanderMusubi 
The RPi devs are working an a package to be provided through the archive.raspberrypi.org APT repo: https://github.com/popcornmix/xbmc/commits/leia_pi4
These provided workarounds are just custom compilations based on the current dev stage.

derdershat, 2019-09-17:
@MichaIng

Rascas is back and yet Kodi has one Buster repo for all PIs at http://pipplware.pplware.pt
[https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=251645&sid=ea78a1bb1966eb7d279e6f69a70403b8](url) 

will not take too long to see this in official RPI repo as it will be the same version

MichaIng, 2019-09-17:
@derdershat 
Jep we implemented this repo already, but it does not work (well) with RPi4, does it?
**EDIT: Ah it seems to work after some manual tweaks. However we added another custom package especially for RPi4 that seems to include those. Will do for now until official packages are available.**

derdershat, 2019-10-08:
some information:

[Kodi is back on official rpi repo](https://www.raspberrypi.org/forums/viewtopic.php?p=1548414#p1548414)

MichaIng, 2019-10-08:
@derdershat 
Wonderful, just the right time before v6.26 release, so we can remove all those workarounds.

MichaIng, 2019-10-08:
Done: https://github.com/MichaIng/DietPi/commit/25e68611a95756150a010c829549d2ecf0417350

PanderMusubi, 2019-10-08:
Thanks, it works. ♥ Time to celebrate. Hope next Debian release will not break Kodi on Raspbian for three months.

PanderMusubi, 2019-10-08:
Do more people get this message after `apt-get install kodi`?

    installed kodi package post-installation script subprocess returned error exit status 1

However, Kodi still works even with this error. The error also occurs after all Kodi files have been removed and install is done again.

MichaIng, 2019-10-09:
@PanderMusubi 
Hmm it worked well on my RPi2 test install. If you installed the kodi-rpi4 package before, this indeed needs to be purged first: `G_AGP kodi-rpi4`
I guess the pipplware repo package can simply be upgraded since it is the same package name.

To investigate, please paste the full output of the apt-get. The actual error mostly hides somewhere in the middle.

PanderMusubi, 2019-10-09:
I have `Raspberry Pi 2 Model B Rev 1.1` with `Raspbian GNU/Linux 10 (buster)` and this is the output

    apt-get install kodi
    Reading package lists... Done
    Building dependency tree       
    Reading state information... Done
    The following additional packages will be installed:
      kodi-bin
    Suggested packages:
      kodi-pvr-mythtv kodi-pvr-vuplus kodi-pvr-vdr-vnsi kodi-pvr-njoy kodi-pvr-nextpvr kodi-pvr-mediaportal-tvserver kodi-pvr-tvheadend-hts kodi-pvr-dvbviewer kodi-pvr-argustv kodi-pvr-iptvsimple kodi-audioencoder-vorbis
      kodi-audioencoder-flac kodi-audioencoder-lame
    Recommended packages:
      libva-intel-vaapi-driver libva1
    The following NEW packages will be installed:
      kodi kodi-bin
    0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
    Need to get 0 B/40.2 MB of archives.
    After this operation, 67.2 MB of additional disk space will be used.
    Do you want to continue? [Y/n] 
    Selecting previously unselected package kodi-bin.
    (Reading database ... 87025 files and directories currently installed.)
    Preparing to unpack .../kodi-bin_2%3a18.4-5~buster_armhf.deb ...
    Unpacking kodi-bin (2:18.4-5~buster) ...
    Selecting previously unselected package kodi.
    Preparing to unpack .../kodi_2%3a18.4-5~buster_all.deb ...
    Unpacking kodi (2:18.4-5~buster) ...
    Setting up kodi-bin (2:18.4-5~buster) ...
    Setting up kodi (2:18.4-5~buster) ...
    dpkg: error processing package kodi (--configure):
     installed kodi package post-installation script subprocess returned error exit status 1
    Processing triggers for hicolor-icon-theme (0.17-2) ...
    Processing triggers for man-db (2.8.5-2) ...
    Processing triggers for mime-support (3.62) ...
    Errors were encountered while processing:
     kodi
    E: Sub-process /usr/bin/dpkg returned an error code (1)

    dpkg -l|grep kodi
    iF  kodi                                   2:18.4-5~buster                     all          Kodi Media Center (arch-independent data package)
    ii  kodi-bin                               2:18.4-5~buster                     armhf        Kodi Media Center (binary data package)

Where can I find the output of the failing post inst script?

PanderMusubi, 2019-10-09:
Does this help?

    cd /var/lib/dpkg/info
    ./kodi.postinst configure
    echo $?
    1
    ./kodi.postinst
    echo $?
    1

Let me know what I need to do to add debugging to that script.

MichaIng, 2019-10-09:
@PanderMusubi 
Does this script not produce any output?

Can you paste it's content: `cat /var/lib/dpkg/info/kodi.postinst`
From the source it should be this: https://github.com/popcornmix/xbmc/blob/master/cmake/cpack/deb/postinst
Very simple, the only thing that can it make fail is `update-menus`. If this matches yours, can you run this from console, in case with `-v` argument for verbose output, if it fails quite.

Ref: https://manpages.debian.org/buster/menu/update-menus.1.en.html

farid77cl, 2019-10-10:
try with

`sudo mv /var/lib/dpkg/info/kodi* /tmp
apt upgrade`

it work for me

MichaIng, 2019-10-10:
Hammer of course works, bit to prevent follow up issues or unexpectations it would be good to check what actually causes this error. I am not keen to ship such a hack with release.

PanderMusubi, 2019-10-10:
This is the content of the post inst file https://pastebin.com/Tcvz5feq

The files on the system are:

    ls -tlra kodi*
    -rwxr-xr-x 1 root root   1096 Sep 23 20:11 kodi.postinst
    -rw-r--r-- 1 root root 143812 Sep 23 20:11 kodi.md5sums
    -rwxr-xr-x 1 root root    374 Sep 23 20:11 kodi.postrm
    -rw-r--r-- 1 root root    509 Sep 23 20:13 kodi-bin.md5sums
    -rw-r--r-- 1 root root    662 Oct  9 16:42 kodi-bin.list
    -rw-r--r-- 1 root root 148070 Oct  9 16:42 kodi.list

and they re-appear after removing them, doing `dpkg -P kodi kodi-bin; apt-get install kodi` and give the same error.

I have changed the second line from `set -e` to `set -ex` and then this is the output

    ./kodi.postinst configure
    + [ configure = configure ]
    + which update-menus
    + [ -x  ]
    + [ ! -f /etc/sudoers.d/020_kodi ]
    + [ ! -f /etc/udev/rules.d/99-kodi.rules ]
    + [ ! -f /etc/polkit-1/localauthority/50-local.d/kodi.pkla ]
    + tee -a /etc/polkit-1/localauthority/50-local.d/kodi.pkla
    echo $?
    1

Does this help find the problem?

MichaIng, 2019-10-11:
@PanderMusubi 
Jep, before installing Kodi, please do:
```
mkdir /etc/polkit-1
```

Something we should contribute to the RPi Kodi package devs, since one cannot assume that this is always installed/present especially since Kodi runs outside of desktop and Xserver in case.

I will add this step to our install code.

PanderMusubi, 2019-10-11:
Should it not be `mkdir -p /etc/polkit-1/localauthority/50-local.d` ? That fixed it for me.

MichaIng, 2019-10-14:
@PanderMusubi 
Ah jep you're right!

closed 2019-10-15 by MichaIng

MichaIng, 2019-10-16:
@PanderMusubi 
I just reported this to the RPi repo: https://github.com/RPi-Distro/repo/issues/153
Couldn't find the source fork or platform to add a PR directly 😄.

PR #3092 (2019-09-16): DietPi-Software | Kodi: Fix install on RPi Buster via 3rd party repo/builds
**Status**: Testing
- [x] Fix /DietPi/dietpi/misc/start_kodi for RPi4, since the custom package ships a different binary
- [x] Changelog

**Testing**:
- [ ] Install on RPi4
- [ ] Install on other RPis on Buster
- [x] RPi4, running `start_kodi` from console, respectively via dietpi-autostart

Fixes: #3031

**Commit list/description**:
+ DietPi-Software | Kodi: Fix Kodi install on RPi Buster, since archive.raspbian.org does not yet contain a Buster-compatible package. Use a 3rd-party repo for RPi1-3 for now and a custom  [... cut]