MichaIng/DietPi #3034: network share intermittent mount failures

psi5asp, 2019-08-04:
### Creating a bug report/issue
using the inbuild network share manager sometimes does not auto-mount the share on reboot.
#//<ip>/share /folder/name cifs username=<username>,password=<password>,iocharset=utf8,uid=dietpi,gid=dietpi,file_mode=0770,dir_mode=0770,vers=3.1.1,_netdev,nofail 0 0

However, this works fine:
//<ip>/share /folder/name cifs credentials=/etc/.creds,iocharset=utf8,uid=dietpi,gid=dietpi,file_mode=0770,dir_mode=0770,vers=3.0,_netdev,nofail 0 0

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=24
G_DIETPI_VERSION_RC=1
G_GITBRANCH=master
G_GITOWNER=MichaIng

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
9.9

- Kernel version | `uname -a`
Linux Pi-4 4.19.57-v7+ #1244 SMP Thu Jul 4 18:45:25 BST 2019 armv7l GNU/Linux

- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
RPi 3 Model B+ (armv7l)

- Power supply used | (EG: 5V 1A RAVpower)
5V 2.4A

- SDcard used | (EG: SanDisk ultra)
SanDisk Ultra

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
DietPi-Drive_manager

- Was the software title installed freshly or updated/migrated?
Fresh

- Can this issue be replicated on a fresh installation of DietPi?
Yes, intermittently 

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. ...setup network shares with DietPi-Drive_manager
2. ...reboot, check network share, repeat

#### Expected behaviour

- ...network share should mount after network established

#### Actual behaviour

- ...network share doesn't always mount

#### Extra details

- ...manual sudo mount -a works fine after reboot.

MichaIng, 2019-08-04:
@psi5asp 
Many thanks for your report.

Did you check for related boot messages (`journalctl` or `dmesg`)?
And how often did you run the test with each fstab entry version?

I don't believe that is has something to do with the way the login credentials are given. Either it is supported that way or not, and both are supported officially by man pages. CIFS version might be related, but this is a server-side issue then.
Probably it is simply about network net yet fully being up. I am not 100% sure currently how well remote mounts respect network targets and how reliable those (network-online.target) assure that a default gateway has been assigned. If one CIFS version allows or requires longer connection time, it might be related.
However as you say it sometimes work, sometimes not, probably you simply had luck with CIFS v3.0?
But logs as above should give a hint 🙂.

MichaIng, 2019-10-18:
I close this issue. Feel free to reopen if required.

closed 2019-10-18 by MichaIng