MichaIng/DietPi #3036: The repository 'https://apt.sonarr.tv develop InRelease' is not signed.

Vansinnet, 2019-08-05:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=25
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
`
- Distro version | 10.0
- Kernel version | Linux DietPi 4.19.58-v7+ #1245 SMP Fri Jul 12 17:25:51 BST 2019 armv7l GNU/Linux
- SBC device | RPi 3 Model B+ (armv7l)
- Power supply used | original)
- SDcard used | sandisk ultra

#### Additional Information (if applicable)
- Software title | 
Sonarr/radarr/any
- Was the software title installed freshly or updated/migrated?
Freshly

#### Steps to reproduce

1. Fresh install of latest Buster image. Tried installing Sonarr/Radarr along with other software
2. All installations halts because of this error
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 https://download.mono-project.com/repo/debian raspbianstretch InRelease
Get:3 https://apt.sonarr.tv develop InRelease [13.3 kB]
Hit:4 https://archive.raspberrypi.org/debian buster InRelease
Hit:5 https://cdn-aws.deb.debian.org/debian sid InRelease
Err:3 https://apt.sonarr.tv develop InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY EBFF6B99D9B78493
Reading package lists...
W: GPG error: https://apt.sonarr.tv develop InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY EBFF6B99D9B78493
E: The repository 'https://apt.sonarr.tv develop InRelease' is not signed.

#### Expected behaviour
Should install as usual.

#### Actual behaviour
Getting error message that the repository is not signed

MichaIng, 2019-08-05:
@Vansinnet 
Many thanks for your report.

Please see here for solution: https://github.com/MichaIng/DietPi/issues/3029#issuecomment-518025659

Vansinnet, 2019-08-05:
> @Vansinnet
> Many thanks for your report.
> 
> Please see here for solution: [#3029 (comment)](https://github.com/MichaIng/DietPi/issues/3029#issuecomment-518025659)

I can confirm this fixes it!

Thanks.

MichaIng, 2019-08-05:
@Vansinnet 
Perfect, many thanks for feedback. Perhaps something we should share via MOTD.

MichaIng, 2019-08-11:
I placed a MOTD regarding this issue, linking to the solution. v6.26 has it included, hence I mark this issue as closed.

closed 2019-08-11 by MichaIng