MichaIng/DietPi #3037: diet-software error | net data

smittyj, 2019-08-05:
#### Details:
- Date           | Mon  5 Aug 18:05:46 ADT 2019
- Bug report     | N/A
- DietPi version | v6.25.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid N2 (aarch64) (index=15)
- Kernel version | #1 SMP PREEMPT Mon Jul 22 10:33:16 CEST 2019
- Distro         | stretch (index=4)
- Command        | G_AGI netdata
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. I had installed and uninstalled some software via the dietpi-software utility.  There was an error about unable to start mariadb.  I expect that the uninstaller removed something out of order.
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
dpkg: unrecoverable fatal error, aborting:
 unknown group 'netdata' in statoverride file
E: Sub-process /usr/bin/dpkg returned an error code (2)
```

--------------------------------------------------------------------

[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

closed 2019-08-06 by smittyj

MichaIng, 2019-08-06:
@smittyj
Were you able to solve it? Strange error since this group should be created by the package automatically.
My attempt would have been to manually pre-create it.

smittyj, 2019-08-06:
I removed the entry from “/var/lib/dpkg/statoverride".  The article I found indicated it was a Debian bug.  I don’t have the link handy but I can find it if you like.

J

MichaIng, 2019-08-06:
@smittyj 
Okay great, perhaps a temporary faulty package inside the repo or it was due to transition from from our own compiled package we used some version ago to the newer version from backports repository that is available since a while.