MichaIng/DietPi #3038: Dev2day repository not found when upgrading dietpi from 6.22.3 to 6.25.3

p-roman, 2019-08-05:
#### Details:
- Date           | Mon  5 Aug 19:00:52 EDT 2019
- Bug report     | N/A
- DietPi version | v6.22.3 (Fourdee/master)
- Img creator    | n/a
- Pre-image      | n/a
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1200 SMP Tue Feb 12 20:27:48 GMT 2019
- Distro         | stretch (index=4)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:
1. Try to upgrade from 6.22.3 to latest version 6.25.3

#### Expected behaviour:
Upgrade succeeds

#### Actual behaviour:
Get below error about dev2day repository not found.

#### Extra details:
I believe this is related to the Plex migration to new repository somewhat recently, though not sure where/how to remove the dev2day repository from being checked during update process. Sorry if this was covered somewhere else, but didn't notice anyone else with a similar issue.

#### Additional logs:
```
Log file contents:
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Ign:2 http://dev2day.de/pms stretch InRelease
Err:3 http://dev2day.de/pms stretch Release
  404  Not Found
Get:4 https://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Reading package lists...
E: The repository 'http://dev2day.de/pms stretch Release' does no longer have a Release file.
```

MichaIng, 2019-08-06:
@p-roman
Many thanks for your report.

Indeed the old 3rd party dev2day repository for Plex Media Server has been shut down in favour of the new official repo.
We apply the related patch and reinstall during update to v6.23 but need to move that to pre-patches, to have it applied before G_AGP is called.

Please run:
```
rm /etc/apt/sourced.list.d/plex.list
G_AGUP
dietpi-update 1
```

p-roman, 2019-08-06:
@MichaIng Thanks for the reply! Had to modify to modify the above command to below, but simple enough and was able to finish upgrade without any issues.

`rm /etc/apt/sources.list.d/plex.list`

MichaIng, 2019-08-06:
@p-roman 
Whoopsie, was a typo (fixed it). Great it worked now, thanks for feedback.

I will keep this issue opened until I moved this fix from update patch_file to pre-patches.

MichaIng, 2019-08-11:
Done: https://github.com/MichaIng/DietPi/pull/3050
With v6.26 release, the obsolete and offline dev2day repo is removed via pre-patches now, prior to G_AGUP being called. So DietPi updates from prior to v6.23 version, which this repo in place, will not run into this error anymore.

I mark this issue as closed. Feel free to reopen if required.

closed 2019-08-11 by MichaIng