MichaIng/DietPi #3039: copying via cp -rv unmounts all harddrives

benjaminpreiss, 2019-08-06:
#### Required Information
- DietPi version | 6.25.3, master branch
- Distro version | 9.9
- Kernel version | Linux DietPi 4.9.185+ #1 SMP PREEMPT Mon Jul 22 10:33:16 CEST 2019 aarch64 GNU/Linux
- SBC device | Odroid N2 (aarch64)
- Power supply used | 
- SDcard used |

#### Steps to reproduce

1. mount a usb stick with dietpi drive manager to location `/mnt/usbstick/`
2. mount a harddrive with dietpi drive manager to location `/mnt/backup_hdd/`
3. copy any existing directory with subdirectories from the stick to the harddrive: `cp -rv /mnt/usbstick /mnt/backup_hdd`

#### Expected behaviour

1. copy should simply copy all files.

#### Actual behaviour

1. copy will not finish successfully. It should give an error somewhat like: `cp: cannot stat ...`
2. launching ranger (the terminal file explorer) one should find that all drives are `not accessible`
3. launching dietpi-drive_manager one should find that the drives are no longer mounted

MichaIng, 2019-08-06:
@benjaminpreiss 
Many thanks for your report.

If the copy breaks in the middle, I guess the USB connection is lost, probably due to insufficient or unstable power supply.

Please paste `dmesg | tail -20` to check for kernel/voltage/IO errors.

benjaminpreiss, 2019-08-06:
`dmesg | tail -20` gives:

```
[    8.122086]  sdb: sdb1
[    8.125613] sd 2:0:0:0: [sdb] Attached SCSI disk
[    8.195650]  sdc: sdc1
[    8.196226] sd 1:0:0:0: [sdc] Very big device. Trying to use READ CAPACITY(16).
[    8.201094] sd 1:0:0:0: [sdc] Attached SCSI disk
[    8.289950] EXT4-fs (sdb1): mounted filesystem with ordered data mode. Opts: (null)
[    9.731497] meson6-dwmac ff3f0000.ethernet eth0: fail to init PTP.
[    9.732544] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   11.275409] vout: aml_tvout_mode_work: monitor_timeout
[   13.836070] meson6-dwmac ff3f0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   13.839239] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   20.266719] EXT4-fs (sdc1): mounted filesystem with ordered data mode. Opts: (null)
[   29.591205] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[   29.594447] Bridge firewalling registered
[   29.615676] nf_conntrack version 0.5.0 (16384 buckets, 65536 max)
[   30.130557] ctnetlink v0.93: registering with nfnetlink.
[   30.605331] IPv6: ADDRCONF(NETDEV_UP): docker0: link is not ready
[   64.523633] fb: mem_free_work, free memory: addr:800000
[  611.339969] fb: osd[0] enable: 0 (kworker/0:4)
[  979.567003] hrtimer: interrupt took 369250 ns
```

MichaIng, 2019-08-06:
@benjaminpreiss 
This was in the same session where the cp error occurred? Nothing suspicious. A hrtimer interrupt but that should not be related.

What does the following show:
```
journalctl -u mnt-usbstick.mount
journalctl -u mnt-backup_hdd.mount
```

If you did a reboot meanwhile, please redo the cp attempt to reproduce the error before calling the above journal checks.

benjaminpreiss, 2019-08-07:
Okey, here comes a more detailed error log:

when i type `cp -uvr /mnt/usbstick/* /mnt/backup_hdd/stephan_backup/upload/` the output is:

1. a lot of lines formatted as: `... -> ...`
2. Then `cp: error reading '...': Input/output error`
3. a line saying:`cp: cannot stat '...': No such file or directory`
4. Then a lot of lines like: `cp: cannot stat '...': Input/output error`
5. Then one line saying: `... -> ...`
6. Then `cp: error reading '...': Input/output error`
7. a line saying:`cp: cannot stat '...': No such file or directory`

after that i can type commands again as if the command would have finished.

`dmesg | tail -20` gives:

```
[116053.244150] [<ffffff8009234220>] __vfs_write+0x108/0x158
[116053.244152] [<ffffff800923518c>] vfs_write+0xac/0x1b8
[116053.244154] [<ffffff8009236a8c>] SyS_pwrite64+0xb4/0xd8
[116053.244158] [<ffffff8009083a1c>] __sys_trace_return+0x0/0x4
[116053.244731] EXT4-fs warning (device sdb1): ext4_end_bio:313: I/O error -5 writing to inode 6553700 (offset 0 size 0 starting block 61868)
[116053.244735] buffer_io_error: 1270 callbacks suppressed
[116053.244736] Buffer I/O error on device sdb1, logical block 61868
[116053.297318] EXT4-fs error (device sdb1): ext4_find_entry:1441: inode #6553606: comm addr2line: reading directory lblock 0
[116053.297371] EXT4-fs error (device sdb1): ext4_find_entry:1441: inode #6553606: comm addr2line: reading directory lblock 0
[116058.890988] Aborting journal on device sdb1-8.
[116058.891001] buffer_io_error: 128 callbacks suppressed
[116058.891005] Buffer I/O error on dev sdb1, logical block 62423040, lost sync page write
[116058.891008] JBD2: Error -5 detected when updating journal superblock for sdb1-8.
[116059.636865] EXT4-fs error (device sdb1): ext4_find_entry:1441: inode #6553606: comm mysqld: reading directory lblock 0
[116059.636903] EXT4-fs error (device sdb1): ext4_find_entry:1441: inode #6553606: comm mysqld: reading directory lblock 0
[116059.636951] EXT4-fs error (device sdb1): ext4_read_inode_bitmap:201: comm mysqld: Cannot read inode bitmap - block_group = 800, inode_bitmap = 26214416
[116059.636969] EXT4-fs error (device sdb1): ext4_journal_check_start:56: Detected aborted journal
[116059.636973] EXT4-fs (sdb1): Remounting filesystem read-only
[116059.641148] EXT4-fs error (device sdb1): ext4_find_entry:1441: inode #6553608: comm mysqld: reading directory lblock 0
[116059.641513] EXT4-fs error (device sdb1): ext4_find_entry:1441: inode #6553606: comm mysqld: reading directory lblock 0
```

`journalctl -u mnt-usbstick.mount` gives `-- No entries --` (maybe because I didint mount it via dietpi drive manager)

`journalctl -u mnt-backup_hdd.mount` gives

```
-- Logs begin at Tue 2019-08-06 11:45:35 BST, end at Wed 2019-08-07 20:09:25 BST. --
Aug 06 11:45:48 DietPi systemd[1]: Mounting /mnt/backup_hdd...
Aug 06 11:45:48 DietPi systemd[1]: Mounted /mnt/backup_hdd.
Aug 07 19:57:20 DietPi systemd[1]: Unmounting /mnt/backup_hdd...
Aug 07 19:57:20 DietPi systemd[1]: Unmounted /mnt/backup_hdd.
```

MichaIng, 2019-08-08:
@benjaminpreiss 
`/dev/sdb1` == `/mnt/backup_hdd`? Since it is mounted at boot, it should be the one producing journalctl entries.

`journalctl` btw is the command to see regular system logs. The mount/umount entries are, I guess, only present of the drive has been added on boot via `fstab` (which is done automatically by `dietpi-drive_manager`), so systemd creates a `.mount` file for it. If you mounted the USB stick via `mount` command, that is skipped.

However we have the faulty drive `/dev/sdb1`. I hope, as above, it's the backup drive, so no important data on it.
Try to solve file system corruption:
```
umount /dev/sdb1
fsck /dev/sdb1
```
If this does not really help, try to re-format it (e.g. via `dietpi-drive_manager)`.

benjaminpreiss, 2019-08-08:
Nope, sadly `/dev/sdb1` is the drive mounted as ˋ/mnt/data_ssdˋ. Therefore already the mount fails due to a ˋdevice busyˋ. I did format all drives (except the usb) with dietpi_drive-manager...

Could it help if I shutdown the server, disconnect the drive to another computer and run ˋfsckˋ then? Will ˋfsckˋ delete the data on the drive?

MichaIng, 2019-08-08:
@benjaminpreiss 
> Could it help if I shutdown the server, disconnect the drive to another computer and run fsck then?

Jep this is the safest way if you have another Linux system.

> Will fsck delete the data on the drive?

It will only check for errors and ask you, if found, if it should attempt to fix them. Only if it is unable to fix a file data error, so the file is unrecoverable destroyed, it will remove it, more precisely unlink it from file system.
In some cases the link from an inode (actual data on drive) to the file system will be lost for other corruption reasons. So the file does not appear on the file system, but the data is still there. In such cases a new link will be created to `/lost+found/#<inode>`. So after running `fsck` you should check this directory and the contained files. Perhaps from the content you are able to derive which file it initially was, or of course if you are missing a particular file. So you can recover it from there.

benjaminpreiss, 2019-08-09:
@MichaIng Thank you so much for the very detailed answers. Never too late to learn something :).

The thing is I cannot physically reach the server at the moment, so I will have to somehow unmoving the drive and repair it.

Does `dietpi-drive_manager` stop all services running on the user data drive? Then maybe I could I mount the drive there...

Otherwise, how can I manually stop all the services running on the drive to unmount and repair it?

MichaIng, 2019-08-10:
@benjaminpreiss 
> Does dietpi-drive_manager stop all services running on the user data drive? Then maybe I could I mount the drive there...

All relevant services are stopped automatically when doing "Check & Repair" via `dietpi-drive_manager` on any drive: https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-drive_manager#L1445
So no need to change anything about the mounts 🙂.

benjaminpreiss, 2019-08-10:
Okey after repairing all drives using dietpi-drive_manager and retrying to copy the error still exists...

`dmesg | tail -100` gives [http://paste.debian.net/1095275](http://paste.debian.net/1095275)

`journlctl -u mnt-usbstick.mount` gives [http://paste.debian.net/1095276](http://paste.debian.net/1095276)

`journlctl -u mnt-backup_hdd.mount` gives [http://paste.debian.net/1095277](http://paste.debian.net/1095277)

`journlctl -u mnt-data_ssd.mount` gives [http://paste.debian.net/1095278](http://paste.debian.net/1095278)

I am a bit stuck, sorry für all the fuss

MichaIng, 2019-08-10:
@benjaminpreiss 
Strange thing, not sure what this "interactive unit" thing is about.

Could you try to comment the entries for these drives in `/etc/fstab`, run `systemctl daemon-reload`, `dietpi-services stop` and `umount /mnt/usbstick` (and the others) manually? Check `df` that they were properly unmounted.

Then run `fsck` manually on these drives. Interactive runs do not fix all issues sometimes or fsck needs to run multiple times.

What I can think about as well, although there are no related dmesg entries, is that the SBC is simply not able to power all those USB drives at once. USB overall power is limited. Do SSD and HDD have an external/dedicated power supply?

Last idea of course would be dying drive, but since all of them show I/O errors, it is very unlikely that all die the same time of course.

benjaminpreiss, 2019-08-10:
Okey, interesting... I will investigate and report back

MichaIng, 2019-08-10:
Another idea is to disable the the USB power management. E.g. this is default on RPi, not sure how well the Odroid N2 handles this. I would assume one will find some related recommends/hints then, which I couldn't, but at least worth to give it a try:
- In dietpi-drive_manager adjust spin-down time to `0 (disabled)`, or even purge the hdparm package: `apt purge hdparm`
- And in `/DietPi/boot.ini` on `setenv bootargs` line add `dwc_otg.lpm_enable=0` to the arguments, so it starts like:
```
setenv bootargs "dwc_otg.lpm_enable=0 root=UUID=...
```

benjaminpreiss, 2019-09-02:
Sorry for letting you folks wait so long, I dont have access to the server at the moment... I will reply in 2 weeks then?

benjaminpreiss, 2019-09-22:
I just tested copying with a new, more powerful power supply and it worked like a charm. What a stupid mistake... Thanks a lot for the help!

closed 2019-09-22 by benjaminpreiss