MichaIng/DietPi #3042: use mosquitto official repo

marcobrianza, 2019-08-07:
I have noticed that the mosquitto binaries are not the latest version but depend on the distribution (stretch / buster)

In this way we are not up not up to date with bugfix and features.
Now MQTT 5 is getting introduced and I think that having the latest version helps to be on track with the new features.

I could easily install the latest version with this instructions
https://mosquitto.org/blog/2013/01/mosquitto-debian-repository/

It looks like there are also Arm builds available, maybe they are not optimised for very arm model but at least are the latest version.

What do you think?

MichaIng, 2019-08-07:
@marcobrianza 
Looks good actually. No ARMv8 binaries and those armhf packages should at least be tested one time on a non-RPi ARMv7 SBC. Aside from that looks like a great enhancement.

ToDo:
- [ ] Test this repo on any non-RPi SBC with ARMv7 chip.

ilgrank, 2020-05-23:
Tested on 6.30 (Buster) on Virtualbox OVA and Mosquitto 1.6.9 works perfectly

MichaIng, 2020-09-28:
PR up for implementation: #3794

closed 2020-10-17 by MichaIng