MichaIng/DietPi #3043: attempt to switch from ram-log to full-log failed

te-anhydros, 2019-08-07:
#### Details:
- Date           | Wed  7 Aug 13:24:37 EDT 2019
- Bug report     | N/A
- DietPi version | v6.25.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | #1244 SMP Thu Jul 4 18:45:25 BST 2019
- Distro         | stretch (index=4)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. run **dietpi-software**
2. select **Log System**
3. select **Full**
4. select **Ok** a few times
5. Select **Install** and **Ok**

#### Expected behaviour:

- take out ram-log
- install full-log

#### Actual behaviour:

- error message:
DietPi Error Handler:
DietPi-Software: G_AGUG                                                                  
- Exit code: 100                                                                        
- DietPi version: v6.25.3 (MichaIng/master) | HW_MODEL:3 | HW_ARCH:2 | DISTRO:4         
- Image creator: DietPi Core Team                                                       
 - Pre-image: Raspbian Lite                                                              
                                                                                        
Log file contents:                                                                       
E: Packages were downgraded and -y was used without --allow-downgrades.                  
Unable to continue, DietPi-Software will now terminate.                                  
                                                 │

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
E: Packages were downgraded and -y was used without --allow-downgrades.
```

MichaIng, 2019-08-07:
@te-anhydros 
Many thanks for your report.

> E: Packages were downgraded and -y was used without --allow-downgrades.

Can you please run and paste output of: `apt-get upgrade`
If it asks you to confirm and packages would be downgraded, please press "n" to decline for now. Lets see first what would be downgraded and why.

te-anhydros, 2019-08-08:
~# apt-get upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  libraspberrypi-bin libraspberrypi0 php7.3 raspberrypi-bootloader raspberrypi-kernel
The following packages will be DOWNGRADED:
  nodejs-dev
5 upgraded, 0 newly installed, 1 downgraded, 0 to remove and 0 not upgraded.
Need to get 40.5 MB of archives.
After this operation, 8,192 B of additional disk space will be used.
Do you want to continue? [Y/n] n
Abort.

te-anhydros, 2019-08-09:
It turns out that we no longer need the package ''nodejs-dev''.
I have 
- purged `nodejs-dev`
- ran `apt autoremove`
- ran `apt update && apt upgrade -y`.

Looks like the host has crashed during `Setting up raspberrypi-bootloader (1.20190718~stretch-1) ...`
I now need to wait for a colleague (on another continent) to power-cycle the host :-(

to be continued...

MichaIng, 2019-08-10:
@te-anhydros 
Good to know Stretch has now also received the `1.20190718` firmware version.

But a crash during bootloader update sounds veeeery bad. Broken bootloader means unbootable system. However the related files can be created/restored manually, even from a Windows system, thanks to the FAT boot partition. But lets hope it boots without this being required.

MichaIng, 2019-10-18:
The RPi firmware upgrades have been greatly enhanced: On Stretch, RPi4 parts are skipped automatically and some space check is done to skip 64bit kernel, RPi4 and possibly other foreign parts if required. So that should be no reason for failure anymore.

I close this issue, feel free to reopen if required.

closed 2019-10-18 by MichaIng