MichaIng/DietPi #3045: DietPi-NordVPN selecting other UDP server doesn't work

svh1985, 2019-08-09:
### Creating a bug report/issue

#### Required Information
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=25
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
stretch
Linux RockPi 4.4.182-rockchip64 #1 SMP Fri Jun 28 17:34:00 CEST 2019 aarch64 GNU/Linux
Rock64 (aarch64)
Rock64 official power supply
Rock64 EMMC

#### Additional Information (if applicable)
- Software title | dietpi-nordvpn
- Fresh
- Don't know

#### Steps to reproduce

When I run the dietpi-nordvpn command and want to select another UDP server, the list is not shown and the tool immediately goes back to the Status screen.

This happens only when I select the "UDP : Recommended" option.

#### Expected behaviour

- Should list the available UDP servers

#### Actual behaviour

- It goes immediatly back to the dietpi-nordvpn status screen

#### Extra details

I did add this to one of the files, but I'm not sure this causes it. (I wanted to add a killswitch)

```
script-security 2
up /mnt/dietpi_userdata/scripts/openvpn/up.sh
down /mnt/dietpi_userdata/scripts/openvpn/down.sh
```
I've double checked the ownership and rights on the files and they are all on 644 and root:root

MichaIng, 2019-08-09:
@svh1985 
Many thanks for your report.

Hmm, the TCP server list works fine? Strange since both lists are created exactly the same way: https://github.com/MichaIng/DietPi/blob/dev/dietpi/misc/dietpi-nordvpn#L241-L267
I also checked the download we do within DietPi-Software install and it contains both directories, for UDP and TCP: https://downloads.nordcdn.com/configs/archives/servers/ovpn.zip

However since if does not work in your case, the dir must be wrong or empty.
Please paste: `ls -l /etc/openvpn/ovpn_udp | head`

svh1985, 2019-08-09:
@MichaIng 
I found the error, by accident there was a file named "--header" created in the directory.
Happened probably created during my up / down script setup.

But because there are many files there it got hidden from "sight".

Maybe it would help if the list script just ignores all files other than *.ovpn

I've tried changing the loop to this but it did not work:
```
                                for i in *
                                do
                                        if ((${i:(-4)} == 'ovpn')); then

                                                G_WHIP_MENU_ARRAY+=("$i" '')
                                        fi

                                done
```

svh1985, 2019-08-09:
It seems that if I rename the "--header" file to "header" that it works. So kind of an edge case here.

Just curious, will you add something to support up / down scripts out of the box in the future?
Or maybe I do a pull request to help for this feature.

MichaIng, 2019-08-09:
Ah okay that makes sense. This is then interpreted as whiptail command line option with is invalid, thus the menu fails.

Jep I will add a check to the loop to only add `.ovpn` files to the array.

svh1985, 2019-08-09:
I've tested it and it works!

I created my first pull request for DietPi that includes this fix:
https://github.com/MichaIng/DietPi/pull/3047

closed 2019-08-10 by svh1985