MichaIng/DietPi #3046: rPi 4: Running very hot

SimplyGardner, 2019-08-09:
I have installed the latest raspberry pi 4 buster image yesterday from the dietpi website. 

I know there has been issues in the past and the raspberry pi foundation released a firmware update to lower the temp - I am not sure if this has been integrated?

However, on idle - I am running 60 degrees (c) - that’s hot! The pi is not doing anything (as checked in `htop`) It’s in a cool well ventilated environment. 

Is this normal?

Not at home at the moment but will complete this later...

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

Trunkzeh, 2019-08-09:
That power supply is not suitable for the RPI4 for starters.

SimplyGardner, 2019-08-09:
@Trunkzeh - What power supply?

Trunkzeh, 2019-08-09:
Ahh just noticed its a default template; i read 5v 1A RAVPower as the power supply you are using which is far too low for the RPI4 :)

MichaIng, 2019-08-09:
@SimplyGardner 
Thanks for your report.

Jep I read through the long discussions about the RPi4 temperature, that actually made me not yet ordering one for my home production server. The new firmware was a slight enhancement but there are still many similar reports https://www.raspberrypi.org/forums/. Last result is usually that without any cooling, you must not put it in any housing and at best place it vertically to allow a better passive head flow. But the very vast majority argue that it simply _requires_ a little fan or something like this: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=243500&start=325#p1509456

But aside from that we can assure that CPU usage is really around zero:
- When checking `htop`, due to the longer 2 seconds update interval, short CPU usage peaks can be missed. `htop -d 5` lowers the update interval to 0.5 seconds. Or of course the 1min/5min/15min average numbers give a clearer overview.
- You can run `G_OBTAIN_CPU_USAGE` to check the overall current (2 seconds, I think) CPU usage (all cores).
- Do you have any RPi4 specific HDMI functionality enabled, i.e, 4k? If it's a headless system you might try to enable the headless mode from dietpi-config > Display Options > Resolution menu. This disables the whole framebuffer pipe. And v6.26 will bring some more slight enhancements when disabling BT/WiFi/Camera module, which you can try if you don't use either of them?
- Do you have USB devices attached? I read that those USB3.0 ports can get quite hot when having a non-self powered device on it, which then raises the housing temp significantly.
- And as you say 60 °C, currently when not changing CPU temperature limit (dietpi-config > Performance Options), it defaults to 65 °C which works very well for RPi Zero/1/2 but means that clocks are started to throttle down at 60°C. You can run `cpu` to get current clocks, which might be already slightly lower then maximum. With v6.26 we raise this to 75 °C for RPi4 (same as RPi3), so throttling starts at 70°C.

SimplyGardner, 2019-08-09:
@michalng

Thanks for the reply - i’ll definitely read more into the links you provided this evening when I am home. 

I am running the rPi 4 without any case and a heat sync. I’ve not installed any software yet on it so it is just a basic dietpi install. To rule out any corruption - went ahead and did a fresh install again.

It’s running headless without any HDMI plugged in (I plan to run headless).

The only thing I have edited is I plugged in a Western Digital Elements HD (has its own power) into the USB3.0 although this has been idle.

MichaIng, 2019-08-09:
@SimplyGardener
As said, try to select the headless mode from dietpi-config. This will be made more accessible/clearer with v6.26 as well, basically a rework of the display menu and I will post some other minor enhancements later to test.

SimplyGardner, 2019-08-09:
`G_OBTAIN_CPU_USAGE` reports between **1.2 to 1.9**
Headless was **enabled**
Installed a mini fan and has brought down the temp to **43 °C** on idle. 

I have also updated the required information in post 1.

MichaIng, 2019-08-09:
@SimplyGardner 
Okay 1.2% to 1.9% close enough to zero, as it should.

Also without case, you could touch the USB port to check if it's hot even in idle. Not sure how well this SBC handles currents for USB devices with external power supply.

Some other things to try:
- Disable RPi camera modules, at least this is the only use case I for them I know about:
  - Before kernel v4.19 these were not enabled, unless you set `start_x=1`, which loads the extended device tree, especially required for the RPi camera module. Due to internal changes these are now enabled as well without extended device tree, and they pull a bunch of dependency modules as well. You can compare `lsmod` before this change and afterwards (+reboot), which should list ~10 modules less then. This will be done with v6.26 update as well, when camera module/extended device tree is disabled.
```
echo -e 'blacklist bcm2835_codec\nblacklist bcm2835_v4l2' > /etc/modprobe.d/dietpi-disable_rpi_camera.conf
```
- Disable Bluetooth via dtoverlay, if you don't use it:
```
G_CONFIG_INJECT 'dtoverlay=disable-bt' 'dtoverlay=disable-bt' /DietPi/config.txt
dtoverlay disable-bt
```
- Disable WiFi via dtoverlay if you use Ethernet only:
```
G_CONFIG_INJECT 'dtoverlay=disable-wifi' 'dtoverlay=disable-wifi' $FP_RPI_CONFIG
dtoverlay disable-wifi
```

We disable Bluetooth and WiFi currently only by blacklisting the kernel modules, but the RPi allows to not load them in the first place via dtoverlays, which might further reduce load on hardware level. Would be actually interesting to compare power usage with blacklisted modules only and dtoverlays on the other hand, as I am not sure if there is really a difference and how large it is.

And as we are about to implement some RPi enhancements from v6.26, there is another one, although should not affect CPU usage/temps at all, only reduce RAM usage a bid:
```
systemctl disable --now haveged
G_AGP haveged
G_AGI rng-tools
G_AGA
```
- This changes the entropy daemon. `haveged` uses slightly more RAM, but works on all devices. `rng-tools` (`rngd`) requires some hardware support, which is the case on RPi. So we can switch this. Only downside is that it produces some system log entries every hour, which cannot be disabled currently, however it is default on Raspbian as well so it's best to follow them.

CPU governor is ondemand, right?
You can verify that it is indeed clocked down on idle:
`cat /sys/devices/system/cpu/cpu*/cpufreq/scaling_cur_freq`
I think it's 600 minimum by default. You can lower minimum clocks to 300 via dietpi-config > Performance Options to have it further clocking down on idle.

SimplyGardner, 2019-08-09:
@MichaIng 

The USB cable to the HDD is not hot or warm. Its normal - room temp.

- Disabled bluetooth
- Disabled wifi
- CPU is is ondemand

Running `cat /sys/devices/system/cpu/cpu*/cpufreq/scaling_cur_freq` I get:
```
600000
600000
600000
600000
```

MichaIng, 2019-08-09:
@SimplyGardner 
Okay, 600 Mhz default minimum clock then, like on all other RPi models. As said, you could lower to 300 Mhz which should have no downside. Default ondemand sample rate is 80 milli seconds AFAIK, which means that after max 80 milli seconds CPU usage over limit (50% by default) the clock rate will be increased, fast enough for all usual tasks.

MichaIng, 2019-08-22:
Okay, our headless mode does not work anymore, since the related `config.txt` setting has been removed from the firmware: https://www.raspberrypi.org/forums/viewtopic.php?p=1523456#p1523456

So until there is another way, `tvservice -o` has to be added again to disable HDMI/video completely. I will run some tests, if this disables the framebuffer as well, so it's dimensions can be left unchanged. At least this allows easier restoring of video capabilities (emergency case, SSH failing on boot and such things) but changing the AUTO_SETUP_HEADLESS value only. `AUTO_` is then again the wrong prefix... however I hope there will be another way soon to allow disabling things without the need to run any command on boot.

Trunkzeh, 2019-08-22:
I've added the Fan Shim from Pimoroni to my RPI4 - stable temperatures of 42c. Bear in mind the RPI4 SoC doesnt begin to throttle until the 80-85 range; over 85 it will throttle fully.

SimplyGardner, 2019-08-22:
Running with no fan and it’s running constantly at 60 degrees c.

Just so I understand the issues, is this an issue with the dietpi os of the raspberry 4 hardware?

Trunkzeh, 2019-08-22:
No the RPI4 just runs hot; same reports on the official Raspbian distro.

MichaIng, 2019-08-22:
@Trunkzeh @SimplyGardner 
Jep, hardware "issue", or lets say the expectations might have been different for users used to earlier RPi models without any heat sink/cooler. I mean I have my Pi2 here without any cooling, in closed case beside some HDD and other electronic and it idles at 36 °C (never saw temps > 50 °C even on longer load). How should I expect that I will ever need any though about cooling for on RPi 😄.

> Bear in mind the RPI4 SoC doesnt begin to throttle until the 80-85 range; over 85 it will throttle fully.

This is a matter of the `temp_limit` setting in `config.txt`, which is 65 °C by default on DietPi. On RPi1-2 this was totally okay, on RPi3 we already raise the default to 75 °C on first boot, on RPi4 with v6.26 as well.

Default on Raspbian is indeed 85 °C, but even if the hardware has no issue with this, if users can't touch certain parts of the SBC without burning their fingers, and plastic cases start to smell and things like that, this is no acceptable situation. So having 75 °C here is good IMO, which means throttling starts at 70 °C, maxxing at 75 °C. Any temperatures beyond that boarder should IMO be cooled down.

Trunkzeh, 2019-08-27:
@MichaIng I briefly took the fan shim off and the RPI4 went up and was idling around 60c. No heatsink or any manner of passive cooling. Set the throttle to 75c within dietpi-config and I havent seen it hit that high during load so this may be a good compromise.

Trunkzeh, 2019-08-29:
For those interested; running the following command below will give you the temperature:

`/opt/vc/bin/vcgencmd measure_temp`

MichaIng, 2019-08-29:
@Trunkzeh 
Alternative: `G_OBTAIN_CPU_TEMP`
This basically does `cat /sys/class/thermal/thermal_zone0/temp` and rounds it to full degree Celsius.

Note that throttling starts about 5 °C below temp limit, so when set to 75 °C it will start throttling at 70 °C. The closer temps reach the limit, the more it throttles down.

MichaIng, 2019-10-09:
Btw with v6.26 we add overclocking presets for RPi4. There is as well one "powersaving" set that simply sets overvoltage to `-2` which lowers CPU/GPU voltage from `1.2V` to `1.15V` which on our tests still runs very stable. Lower voltage = lower power = lower temp.

Aside from that there have been several enhancements done by RPi devs kernel/firmware-wise. I mark this issue hence as closed for now. If there are still ideas/enhancements to better control RPi4 temps, just mention @MichaIng here and I will reopen for investigation.

closed 2019-10-09 by MichaIng