MichaIng/DietPi #3049: Nextcloud stuck at: Step 4 is currently in process. Please reload this page later.

Borotes, 2019-08-11:
Hey guys,

I am using DietPi on my RaspberryPi 3 B+ and updated to version v6.25.3 today and after the reboot I wanted to update my nextcloud from 16.0.2 to 16.0.3.

The Updater was stuck at downloading and now I am only getting the message "Step 4 is currently in process. Please reload this page later." I already googled it, but can't find the .step file.

Could you help me with that?

MichaIng, 2019-08-11:
@Borotes 
How long is it stuck already?

Is the update really still in progress? Could you check `htop` for any webserver of PHP sub processes? Which webserver do you use?

Borotes, 2019-08-11:
@MichaIng For some hours now, I checked the internet connection immediately and everything was fine. I tried a reboot of the raspberry after ~3 hours, because I could not access it anymore and putty was really slow.

After that I could access it and log in, when I click on open Updater the message is shown.

Lighttpd is the webserver.

https://i.imgur.com/uiamky5.png
https://i.imgur.com/ZeRJdVd.png

MichaIng, 2019-08-11:
Okay, do the following:
```
rm -v /mnt/dietpi_userdata/nextcloud_data/updater-*/.step
```

Borotes, 2019-08-11:
@MichaIng It worked to stop it, I did a reboot and tried to update again.

https://i.imgur.com/JPgAT0b.png

Seems stuck again :(

MichaIng, 2019-08-11:
@Borotes 
What does the nextcloud log show:
```
tail /mnt/dietpi_userdata/nextcloud_data/nextcloud.log
```

Borotes, 2019-08-11:
It is really slow (40KB/s) I will check, what could be the problem, maybe the SD-Card has a problem.

MichaIng, 2019-08-11:
@Borotes 
Since  `/tmp` is a tmpfs (RAM disk), it cannot be related to the SDcard, but the network setup then 🤔.

With this download speed, the download would take ~30 minutes, thus the updater at this step.

Borotes, 2019-08-11:
Restarted my switch, my router and changed the cable, Hope that it works then 😄

Borotes, 2019-08-11:
Seemed to have worked now, but I don't know why it happed. Anyway thanks for your help!! 😀😀

MichaIng, 2019-08-11:
@Borotes 
Great. I mark this issue as closed then, feel free to reopen if required.

closed 2019-08-11 by MichaIng