MichaIng/DietPi #3051: Image Request | Rock64 Buster

svh1985, 2019-08-11:
### Creating an image request

#### Formal device information
- Device name | Pine Rock64
- Official product URL | https://www.pine64.org/devices/single-board-computers/rock64/
- GitHub resource URL (if available) | 
- Image download URL | https://wiki.pine64.org/index.php/ROCK64_Software_Release#Armbian_Debian_Buster_Minimal_.5BmicroSD_.2F_eMMC_Boot.5D_.5B5.91.5D

#### Is the SBC officially supported by the Debian installer?
-  

#### If not, is a reliable 3rd party Debian image available for this SBC?
- https://github.com/ayufan-rock64/linux-build

MichaIng, 2019-08-11:
@svh1985 
Thanks for your request. Jep we will update our images when we find time.

MichaIng, 2020-01-14:
@svh1985 
New Rock64 Buster image is ready for testing. It has been created within a systemd-nspawn container, hence real hardware test is required: https://dietpi.com/downloads/testing/DietPi_Rock64-ARMv8-Buster.7z

svh1985, 2020-01-21:
@MichaIng just tried but it hangs on `vcc_sd: disabling`

MichaIng, 2020-01-21:
@svh1985 
Thanks and damn, I applied the Ethernet fix the wrong way round... Will remove it, probably the new kernel has it fixed anyway.

MichaIng, 2020-01-21:
@svh1985 
If you find time, could you retest with the newly packed image: https://dietpi.com/downloads/testing/DietPi_Rock64-ARMv8-Buster.7z

If this works well, could you paste `ip a` after reboot to check if this issue is still present without fix: https://github.com/MichaIng/DietPi/issues/3066#issuecomment-523394847
It seems to have been fixed in Armbian kernel, but I am not yet sure if the current package from their APT repo already contains it.

svh1985, 2020-01-21:
@MichaIng still no luck:
[image]

MichaIng, 2020-01-21:
Okay I think I have to redo the image from scratch and skip the uboot image redo right from the beginning.

MichaIng, 2020-01-21:
@svh1985 
Lol Cloudflare cache was the issue, the old image was still in place until I manually cleaned cache just now. Regardless I redid the image with slightly newer kernel version, which, based on timestamp, might even be required to have the Ethernet fix included 😄. So I hope this works now fine as expected. As said, `ip a` check for correct `eth0` `UP` state would be great.

svh1985, 2020-01-22:
Works like a charm!

svh1985, 2020-01-22:
Only thing that doesnt work is enabling the swapfile
[image]
[image]
[image]

MichaIng, 2020-01-22:
@svh1985 
Great that it works now. Cloud you paste `ip a`, please to check Ethernet state?

Note that `1` only adds a swap file if less than 2 GiB physical RAM is available, to have 2 GiB overall, which is a certain minimum to run most software options without issues. Based on the /tmp size, your Rock64 model has 4 GiB, hence swap is disabled.

If you want or need a swap file for heavy RAM usage/peeks, add the wanted number in MiB instead of `1`, e.g. `4096` for 4 GiB swap file size.

svh1985, 2020-01-23:
`3: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 92:26:27:9c:2b:7a brd ff:ff:ff:ff:ff:ff
    inet 172.16.0.24/24 brd 172.16.0.255 scope global dynamic eth0
       valid_lft 71661sec preferred_lft 71661sec
    inet6 fe80::9026:27ff:fe9c:2b7a/64 scope link
       valid_lft forever preferred_lft forever`

Looks great!

MichaIng, 2020-01-23:
@svh1985 
Okay that looks very great.

We have btw another workaround in place for related issues: #2028
Only thing is that I cannot see the kernel fix being applied for Rock64 but only RK3399 boards: https://github.com/armbian/build/pull/1736
If you are in mood and switched your production Rock64 to the new Buster image, you could remove the following file and see if you face any network issues during normal operation:
```
rm /etc/network/if-up.d/dietpi-disable_offload
ethtool -K eth0 rx on tx on
```

svh1985, 2020-01-24:
@MichaIng its actually already on my production device. My previous install crashed 🤯

Is there a way to undo that command if it causes issues? Probably just backing up dietpi-disable_offload. But for the other command?

MichaIng, 2020-01-24:
@svh1985 
To revert:
```
wget https://raw.githubusercontent.com/MichaIng/DietPi/master/rootfs/etc/network/if-up.d/dietpi-disable_offload -O /etc/network/if-up.d/dietpi-disable_offload
ethtool -K eth0 rx off tx off
```
Note that this cannot really break things, worst case would be an unstable Ethernet connection with some hangs here and there.

svh1985, 2020-01-28:
Still have to run the command, but I found out that the USB 3 port doesnt work.
https://github.com/ayufan-rock64/linux-build/issues/391

MichaIng, 2020-01-31:
@svh1985 
> Still have to run the command, 

What you mean by this, `ethtool -K eth0 rx off tx off` is still required to have a stable Ethernet connection?

> but I found out that the USB 3 port doesnt work.

Ui, thanks for info. Damn thing that this seems to be the case with the Armbian kernel as well.

svh1985, 2020-02-01:
@MichaIng 
just ran the commands but got this:
```
root@RockPi:~# rm /etc/network/if-up.d/dietpi-disable_offload
root@RockPi:~# ethtool -K eth0 rx on tx on
Could not change any device features
```

MichaIng, 2020-02-02:
@svh1985 
Hmm, probably this cannot be done on the fly with every adapter, or meanwhile it has been something hardcoded with the latest driver update. I played a bid with virtual network adapters:
```
2020-02-02 16:45:42 root@VM-Buster:~$ ethtool -K eth1 rx off tx off
Actual changes:
rx-checksumming: off
tx-checksumming: off
        tx-checksum-ip-generic: off
tcp-segmentation-offload: off
        tx-tcp-segmentation: off [requested on]
2020-02-02 16:45:57 root@VM-Buster:~$ ethtool -K eth1 rx on tx on
Actual changes:
rx-checksumming: on
tx-checksumming: on
        tx-checksum-ip-generic: on
tcp-segmentation-offload: on
        tx-tcp-segmentation: on
2020-02-02 16:46:19 root@VM-Buster:~$ ethtool -K eth0 rx off tx off
Cannot change rx-checksumming
Actual changes:
tx-checksumming: off
        tx-checksum-ip-generic: off
tcp-segmentation-offload: off
        tx-tcp-segmentation: off [requested on]
        tx-tcp6-segmentation: off [requested on]
2020-02-02 16:46:25 root@VM-Buster:~$ ethtool -K eth0 rx on tx on
Cannot change rx-checksumming
Actual changes:
tx-checksumming: on
        tx-checksum-ip-generic: on
tcp-segmentation-offload: on
        tx-tcp-segmentation: on
        tx-tcp6-segmentation: on
```
- Both up, `eth0` for inet and virtio virtual adapter, `eth1` added for internet network only, and emulated `Intel PRO/1000 MT Desktop`.
- Changes work fine, but `eth0` cannot change rx, only tx, regardless if its up or not. So it is a bid adapter/driver dependent at least.

As you removed `/etc/network/if-up.d/dietpi-disable_offload`, there are not changes done at boot now at least. If after reboot `ip a` still shows state UP and you do not face any unstable network or such, then I think we can remove this workaround. I'll not remove it from running boards, to not break any running system, but will not apply it to our fresh images anymore, which in case of Rock64 and RK3399 boards are/will be all based on current Armbian kernel.

svh1985, 2020-02-05:
UP seems to be fine. I didn't noticed any other network issues, are there specific issues I look out for?
```
root@RockPi:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: dummy0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 5e:33:cf:0f:48:33 brd ff:ff:ff:ff:ff:ff
3: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 92:26:27:9c:2b:7a brd ff:ff:ff:ff:ff:ff
    inet 172.16.0.24/24 brd 172.16.0.255 scope global dynamic eth0
       valid_lft 86370sec preferred_lft 86370sec
    inet6 fe80::9026:27ff:fe9c:2b7a/64 scope link
       valid_lft forever preferred_lft forever
root@RockPi:~#
```

MichaIng, 2020-02-05:
@svh1985
Great, should be fixed kernel/driver-wise then. Just report if you face any Ethernet connection issues at a later date during normal operation, so we can reopen investigation.

svh1985, 2020-03-27:
Image runs great, the only issue is the USB 3.0 thing. Not sure who needs to fix the kernel.

svh1985, 2020-04-09:
@MichaIng USB 3.0 issue seems to be solved in 0.10.x. https://github.com/ayufan-rock64/linux-build/issues/391#issuecomment-610819273

MichaIng, 2020-04-14:
@svh1985 
Note that this is for Ayufans image/kernel builds, not necessarily for Armbian ones (that we use). However, probably the fix was coming from mainline kernel anyway or has been applied on Armbian as well.

MichaIng, 2020-05-17:
I updated ping defaults to 9.9.9.9/Quad9, removed the obsolete offload workaround and removed the unexpected resolvconf head file entry. Will move image to stable now 🚀!

closed 2020-05-17 by MichaIng