MichaIng/DietPi #3052: Can't connect to rpi3

Dr0bac, 2019-08-12:
It is my mistake, i removed external hdd (userdata is on ext) cable and lan cable from raspberry pi and than i removed power cable..after plugging all cables back to rpi3 now i can't connect to it.. in router status page i don't see rpi3 ip address, or any new ip address.. i can't attach hdmi because in options it is set to headless.. when i power on rpi3 ext hdd goes to sleep mode instantly.. i don't know if network is messed up or powering on is messed up..what can i do?

MichaIng, 2019-08-12:
@Dr0bac 
Generally I would always unplug the power cable first, however should not play a role after ordinary shutdown.
And re-connect power cable last, so everything else is connected on boot.

Please re-enable HDMI to investigate. Edit `config.txt` (on FAT partition) from external system, comment (or set to `0`) those two entries:
```
hdmi_ignore_hotplug
hdmi_ignore_composite
```
and comment (or set to some supported resolution):
```
framebuffer_width
framebuffer_height
```
_(Ah I think the two above are already commented, so could be left untouched.)_

Dr0bac, 2019-08-12:
when i plug micro sd into pc only boot partition is displayed (which is FAT32), i have config.txt there, and i changed

    hdmi_ignore_hotplug=1
    hdmi_ignore_composite=1

to 

    hdmi_ignore_hotplug=
    hdmi_ignore_composite=0

when i turned on rpi3 again hdmi is not detected, when i plugged micro sd back into pc, hotplug and composite are again at value 1.

I guess i need to change another config.txt. How to access that on pc?

MichaIng, 2019-08-13:
@Dr0bac 
Hmm that is strange, this one `config.txt` on FAT partition is the only and correct one. Probably some caching issue on the system where you edited the file? Did you recheck after changing it, in case unplug+replug, that the change has been indeed done?

You could as well edit `dietpi.txt` and set `AUTO_SETUP_HEADLESS=0`, however this value is only applied on first boot, so does not influence HDMI/video signal directly.

You did not move the root file system to an external drive, or even boot from USB, right?

Dr0bac, 2019-08-13:
You are correct! after i edited config.txt, and unplugged with safe to remove hardware option, and plugged again, config.txt is changed to original state, i tried again and again with notepad and notepad++. Also i moved config.txt from micro sd, and after unplug sd card, config.txt is again on sd card..
 
Also when i plug sd card i get message there is a problem with this drive, scan the drive now and fix it, i didn't clicked on message, i'm afraid that it will format sd card or similar..

MichaIng, 2019-08-14:
@Dr0bac 
Interesting. But the editors do not give any error when you safe the file? The large SDcards have a R/W slider at the side, I guess microSD => SD adapters as well. Currently not sure how Windows recognises this or not, but at least worth to have a look at it, if it is switched to R/W 😄.

Hmm, yeah I also once had a bad result with file system check+fix on Windows. It does never format anything, but the way it fixes things is not 100% UNIX compatible in every case I think. Perhaps has to do with Windows vs UNIX line ending format or something similar, not sure.

However safer would be to fsck the two partitions on another Linux system. E.g. you could setup a VM to achieve that, if you have no second native Linux system with SDcard slot there.
- You can create a `.vmdk` that links to the physical SDcard and can be mounted to the VM, e.g. I use:
```
"C:\Program Files\Oracle\VirtualBox\VBoxManage" internalcommands createrawvmdk -filename "C:\Users\Micha\VirtualMachines\RPi2\rpi2.vmdk" -rawdisk "\\.\PHYSICALDRIVE1"
```

Dr0bac, 2019-08-28:
Hey @MichaIng i still can't solve my issue :smiley: 
first i tried with virtualbox to change config.txt, i downloaded raspbian, started it, plugged miscro sd card reader, and again same result, settings are returned after replug
Than i bought second card reader, and tried again, and results are even worse.. here are few screenshots

[image]

[image]

[image]

[image]

MichaIng, 2019-08-28:
@Dr0bac 
Indeed it looks like the SDcard is dying. Not too unusual, their lifetime R/W cycles are limited. Makes sense to take care of good quality/brand and warranty.

This must have been also the reason for your initial issue with connecting to DietPi/RPi. Most likely boot failed on early stage.

Does `fdisk -l /dev/sdb` at least report the partition table? Probably it can re-create/re-write the partition table.

Dr0bac, 2019-08-31:
I tried to format it in windows with no luck.. in disk managment i got cyclic redundancy check error, also in diskpart same error.. with chkdsk i get that device RAW and can't fix it..

I think sd card have software malfunction, because problem started because i unplugged lan cable and external hard disk without power off raspberrry pi..

What sd card you recommend for raspberry pi?

MichaIng, 2019-09-02:
@Dr0bac 
A bid strange that an unclean cable removal destroys the SDcard (instead of data corruption only), but it seems to.

If there is any chance, then with `fdisk` or, if it cannot even read partitions, it could be tried to cleanly overwrite the whole drive with dd: `dd if=/dev/zero of=/dev/<sdcard>`
Perhaps the Windows tools are bad in dealing with broken partition tables/file systems only, and the SDcard itself is still intact, who knows.

closed 2019-10-02 by Dr0bac