MichaIng/DietPi #3053: OdroidC2, can't end clean-install, menu-loop

pbecks1963, 2019-08-12:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2019-08-12:
@pbecks1963 
Thanks for your report.

Can you please give some more details, which menu loops, paste of screen output/error messages and such?

pbecks1963, 2019-08-12:
Photos.zip
<https://drive.google.com/file/d/1-kIg0lB1j6nym9wnns79-27hDE27zDw_/view?usp=drive_web>
Hello Michalng,

What i did: reflashed dietpi on a new sd-card. Took a lot of pictures
during the install process (ssh). For the record: i changed the root/Unix
passwords onces, in both instances the SAME password, i don't know if that
matters, just to mention it.

See attachments for install-process..

Greetz: Patrick

MichaIng, 2019-08-13:
@pbecks1963 
Ah I know the issue. The dietpi-software whiptail menu is so small that the last entry is not visible, which need to be selected. Please us the up/down arrow buttons and navigate to the very last menu entry, which is "Install". This will finish first run setup.

No need to go on reading, as long as you're not interested in details 😉.
__________________
I also know the reason why the menu is so small, while the screen has sufficient size: https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-login#L165
- We call `dietpi-software` from login script and redirect STDOUT + STDERR to log file, to have a firstrun setup log.
- It is already known that on Debian versions before Buster, `tput` requires one of these two streams to estimate the correct console dimensions, otherwise it uses some low defaults.
- Within dietpi-update the issue was the same: https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-update#L469
  - But as can be seen we implemented some terminal dimension override system for this case.
- So we simply need to implement this for dietpi-software firstrun call as well.

Just to verify the issue:
```
# Small whiptail
G_WHIP_MSG test 2>&1 | tee
# Large whiptail
G_WHIP_MSG test
# Due to wrong dimensions
root@VM-Stretch:~# tput cols 2>&1 | tee; tput lines 2>&1 | tee
80
24
# While this shows correct dimensions
root@VM-Stretch:~# tput cols; tput lines
237
71
```

On Buster this has been solved:
```
2019-08-12 14:10:58 root@micha:/tmp# tput cols
237
2019-08-13 01:45:38 root@micha:/tmp# tput cols 2>&1 | tee
237
```
_________
Resolved for v6.26 with: https://github.com/MichaIng/DietPi/commit/2443d5a8ce8136d0df39062aaf04ed2b1b7af5c0
Changelog: https://github.com/MichaIng/DietPi/commit/5bd537d1d5ba6db85ce6113c48c18944ebbf914c

pbecks1963, 2019-08-18:
> Resolved for v6.26 with: 2443d5a
> Changelog: 5bd537d

Thnx man! (that was it)

Greetz: Patrick

MichaIng, 2019-08-27:
Okay I mark this issue as closed. Feel free to reopen if required.

closed 2019-08-27 by MichaIng