MichaIng/DietPi #3055: System | Switch from initramfs-tools to tiny-initramfs?

MichaIng, 2019-08-13:
A lightweight alternative to the full fledged initramfs implementation. Much smaller, boots faster, less features that are not required for default setups. Can be easily replaced if required.
- https://github.com/chris-se/tiny-initramfs
- https://packages.debian.org/buster/tiny-initramfs

Have it up on my testing VMs for a while now, no issues observed. Only issue is that is has a different update command: `update-tirfs`
So we would need to check which command is available, before calling it.

It does not support PARTUUID for rootfs in cmdline, however we ship RPi without any initramfs implementation and all other SBCs allow UUIDs?
But requires testing on each SBC anyway, at first should be only added to x86_64 devices without custom kernel+bootloader.
___________
So if someone is in mood, for testing:
```
apt purge initramfs-tools
apt autoremove --purge
apt install tiny-initramfs
```
- Just it. It's updated with kernel upgrades automatically, just as initramfs-tools does.
- Note that in case of an SBC, it might make it unbootable, if kernel/bootloader requires some extended feature. So have a backup of the old `initrd.img` (either in root or /boot) available to restore, or at best an SDcard or /boot partition clone.

Vansinnet, 2019-08-15:
Is it a problem that it was last updates 3 years ago?

MichaIng, 2019-08-15:
Yeah recognised this. However it works well at least on my VMs and has boot time benefits.

Also Debian is still maintaining it regularly: https://deb.debian.org/debian/pool/main/t/tiny-initramfs

We could also experiment with Dracut, with by default should be faster as well with its modular system and is actively developed within Linux itself.

Vansinnet, 2019-08-16:
As long as the intent is to switch to something faster and still being able to function to DietPis needs I’m all for switching to it!

I’m just a bit concerned about switching to something which barely is supported itself anymore. Maybe it has not been updated because it doesn’t need to! In that case, I’d say do the switch! :)

MichaIng, 2020-03-13:
Done:
- https://github.com/MichaIng/DietPi/commit/16e9a945ac0be3c659942deab25db9976f680c28
- https://github.com/MichaIng/DietPi/commit/76a373fe391cb567596cbc3d5adafa8b458060c2

Probably native PCs can follow, but requires more testing.

closed 2020-03-13 by MichaIng