MichaIng/DietPi #3056: DietPi-Software | qBittorrent - Login with default username/password not working

cstegmann, 2019-08-14:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=25
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
10.0
- Kernel version | `uname -a`
Linux DietPi 4.19.58-v7+ #1245 SMP Fri Jul 12 17:25:51 BST 2019 armv7l GNU/Linux
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
RPi 2 Model B (armv7l)

#### Additional Information (if applicable)
- Software title | qBittorrent
- Was the software title installed freshly or updated/migrated?
Freshly installed on a newly flashed install of Dietpi with the Buster image (DietPi_RPi-ARMv6-Buster)
- Can this issue be replicated on a fresh installation of DietPi?
I guess, mine is fresh.

df58bce3-0908-4d55-9894-ff250f139903
- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. Installed qBittorrent with dietpi-software
2. Can't log in to qBittorrent Web UI with the qbittorrent/dietpi username and password. Also tried combinations of the older stuff and default qbittorrent usernames/passwords and I just get locked IP banned after a few tries.

#### Expected behaviour

- Login should be working. 

#### Actual behaviour

- It isn't.

#### Extra details

- there is no qBittorrent directory in ~/.config/ or /root/.config/

cstegmann, 2019-08-19:
Any updates on this? I'm using Transmission for now, but it doesn't have rss support.

I create the ~/.config/qBittorrent directory and that didn't help. I can't find the default config file anywhere that I could paste in.

cstegmann, 2019-08-21:
@MichaIng found the problem. It's either documentation or implementation, depends how you want to look at it.

In dietpi-software line [10797](https://github.com/MichaIng/DietPi/blob/d36ad166271fead7707faf90a7fba8b8a2ae4947/dietpi/dietpi-software#L10797) it says:

`WebUI\Password_ha1=@ByteArray($(echo -n "$GLOBAL_PW" | md5sum | mawk '{print $1}'))`

In the [documentation](https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=70#p2272) it says the password should be 'dietpi' and not the Global password/$GLOBAL_PW you see in the code.

So either the documentation needs to change or the $GLOBAL_PW needs to be replaced by 'dietpi'

MichaIng, 2019-08-21:
@cstegmann 
Indeed, wherever it says password "dietpi" in the docs, it means the global software password that you chose during firstrun setup, which only defaults to "dietpi".

But you are right, we should make this more clear in the docs.

MichaIng, 2020-11-11:
This has mostly be done on our new documentation page: https://github.com/MichaIng/DietPi-Docs
`dietpi-software list` links still need to be adjusted, however marking it as closed here.

closed 2020-11-11 by MichaIng