MichaIng/DietPi #3057: Jackett keeps breaking and doesn't work

athousanddetails, 2019-08-14:
### Creating a bug report/issue

I'm on the latest version, everything updated on my DietPi Asus Thinker Board, and Jackett keeps breaking down all the time. If I unninstall and install again it will work for 1 day and then off again he goes. 

I don't know how to reproduce the bug, but this is annoying as well specially with Sonarr and Radarr. Any idea what is going on or how to fix this?

#### Required Information
- SBC device | Asus Thinker Board

MichaIng, 2019-08-16:
@athousanddetails 
Many thanks for your report.

You mean it crashed after you updated Jackett, right?
Most likely this known issue with the internal Jackett updater: https://github.com/MichaIng/DietPi/issues/2593

A solution to repair the broken Jackett instance is given in the linked issue as well.

Sad to see that this quite important bug has still not yet been fixed. It is sadly outside our possibilities to ship some workaround with DietPi, besides probably disable the updater right within our install script. I was just hoping for a soon upstream fix 😞.
_________
Sorry for copy&paste from here: https://github.com/MichaIng/DietPi/issues/3058
Just the same bug reports the same time 😉.

VasilevAndrei, 2019-08-20:
Hi, I had the same problem. The following solution worked for me - after reinstalling jackett, check the box for "Disable auto update". My current Jackett Version 0.11.589.0

MichaIng, 2019-08-27:
I mark this issue as closed in favour of the linked initial one.

closed 2019-08-27 by MichaIng