MichaIng/DietPi #3058: Jackett --NoRestart (code=exited, status=2) | Latest version after upgrade

SotirisParaskevopoulos, 2019-08-15:
OrangePi Plus 2
Armbian 
Linux orangepiplus 4.19.62-sunxi

I tried to update my box and after that, the service won't start. What can i do to provide more informations?

MichaIng, 2019-08-16:
@SotirisParaskevopoulos 
Many thanks for your report.

You mean it crashed after you updated Jackett, right?
Most likely this known issue with the internal Jackett updater: https://github.com/MichaIng/DietPi/issues/2593

A solution to repair the broken Jackett instance is given in the linked issue as well.

Sad to see that this quite important bug has still not yet been fixed. It is sadly outside our possibilities to ship some workaround with DietPi, besides probably disable the updater right within our install script. I was just hoping for a soon upstream fix 😞.

MichaIng, 2019-08-27:
I mark this issue as closed in favour of the linked initial one.

closed 2019-08-27 by MichaIng