MichaIng/DietPi #3059: Restarting php7.3-fpm using dietpi-services gives an error:

maartenlangeveld, 2019-08-16:
Restarting php7.3-fpm using dietpi-services gives an error:
> root@DietPi:/# dietpi-services restart php7.3-fpm

```
[  OK  ] DietPi-Services | Root access verified.

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: restart php7.3-fpm

/DietPi/dietpi/dietpi-services: line 408: php7.3-fpm: syntax error: invalid arithmetic operator (error token is ".3-fpm")
```

However
> root@DietPi:/# service php7.3-fpm restart

works fine

MichaIng, 2019-08-16:
@maartenlangeveld 
Many thanks for your report. Which DietPi version is this?

maartenlangeveld, 2019-08-17:
DietPi v6.25.3

```
root@DietPi:~# sed -n 408p /DietPi/dietpi/dietpi-services
                        [[ ${aSERVICE_MODE[$i]} == 'masked' ]] && { G_DIETPI-NOTIFY 2 "skip : ${aSERVICE_NAME[$i]} (due to mask)"; continue; }
```

MichaIng, 2019-08-19:
@maartenlangeveld 
Found the issue, indeed a coding bug.
Resolved with: https://github.com/MichaIng/DietPi/commit/b4cb3512b02e516af02620add81503523242b39a
Changelog: https://github.com/MichaIng/DietPi/blob/dev/CHANGELOG.txt

To apply fix now:
```
sed -i 's/Apply_Service_State $INPUT_CMD $INPUT_SERVICE/Apply_Service_State $INPUT_CMD ${INPUT_SERVICE+0}/' /DietPi/dietpi/dietpi-services
```

This passed testing since when you add any non-integer value as array index, (`echo ${array[test]}`) it will be converted/interpreted as `0`, which is correct when dietpi-services is called with single service input. But certain character types, like dots (in your case), this does not work and the error is thrown instead:
```
root@VM-Stretch:~# aTEST[0]=success
root@VM-Stretch:~# echo ${aTEST[0]}
success
root@VM-Stretch:~# echo ${aTEST[test]}
success
root@VM-Stretch:~# echo ${aTEST[test.oops]}
-bash: test.oops: syntax error: invalid arithmetic operator (error token is ".oops")
```

maartenlangeveld, 2019-08-23:
@MichaIng 
Thanks, but unfortunately the fix did not help....

MichaIng, 2019-08-25:
@maartenlangeveld 
Ay jep, I've overseen something. Fixed with: https://github.com/MichaIng/DietPi/commit/2738685269c4e33b55dfb0af2da569a89fbfe30b

closed 2019-10-15 by MichaIng