MichaIng/DietPi #3060: WAN-IP check backup

svh1985, 2019-08-17:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- ATM the dietpi.com site is down, this means that the check ip also fails with all kinds of errors

#### Describe the solution you'd like:

- Have a secondary IP check as a backup
- Ignore HTML errors like example below, probably better to show UNKNOWN, not the entire HTML

#### Describe alternatives you've considered:

- We can also include the DYN Check IP in the check: http://checkip.dyn.com/

#### Error example
[image]

MichaIng, 2019-08-17:
@svh1985 
Many thanks for your report.

Indeed we had a server downtime: https://github.com/MichaIng/DietPi/issues/3061
It is now up again, but you're right, we need to to avoid this output for WAN IP and as well if the weather data download fails: https://github.com/MichaIng/DietPi/blob/dev/dietpi/func/dietpi-banner#L199

I thing since we use Cloudflare, the problem is that curl does not report an error when dietpi.com is down, but instead one is redirected to this Cloudflare page, showing that the final host is down, and one can try to load a cached version of the static website. Easiest solution would be IMO to only print the first output line (head -1), as WAN IP and weather data should never require more.

MichaIng, 2019-08-26:
Okay I changed the way we handle curl error outputs in the banner: https://github.com/MichaIng/DietPi/commit/db47a9466ff452bf20c48bf0bd474aa85f3fd6e4

To compare on a non-existing document:
```
root@VM-Stretch:~# curl -sLm 2 https://dietpi.com/myip.phpa 2>&1
<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>404 Not Found</title>
</head><body>
<h1>Not Found</h1>
<p>The requested URL /myip.phpa was not found on this server.</p>
</body></html>
root@VM-Stretch:~# curl -sSfLm 2 https://dietpi.com/myip.phpa 2>&1
curl: (22) The requested URL returned error: 404
```
To be true, it requires testing for Cloudflare served websites, but it looks like the HTTP status code in your case was 521, a Cloudflare special for webserver downs: https://en.wikipedia.org/wiki/List_of_HTTP_status_codes#Cloudflare
If this response code is sent to clients, curl should omit the error page HTML code and instead print the informational one-line error only.

Perhaps we can find some Cloudflare served server that is down, for testing?
________
However there is still things we can improve, also to speed up the banner:
- Estimate network-based info via cron job (or systemd timer): https://github.com/MichaIng/DietPi/issues/2832
- By this we can also allow longer connections times for slow network systems. Currently it allows max 2 seconds to not delay the banner print too much.
- In case of any error, we could keep the last stored info. Probably allow this for 1 day, if until then error is not resolved, print it, so users can report/we can investigate.

closed 2019-10-15 by MichaIng