MichaIng/DietPi #3062: Netdata missing in repository

Velociraptor85, 2019-08-18:
So I wanted to install Netdata on my Dietpi Nanopi Neo2 but apparently it is missing in the repository? Or at least it did not install and gave the following error.

#### Details:
- Date           | Sun 18 Aug 12:24:08 BST 2019
- Bug report     | N/A
- DietPi version | v6.25.3 (Fourdee/master)
- Img creator    | DietPi Core Team
- Pre-image      | FriendlyARM
- SBC device     | NanoPi NEO 2 (aarch64) (index=65)
- Kernel version | #82 SMP Fri Dec 8 14:33:14 CST 2017
- Distro         | stretch (index=4)
- Command        | G_AGI netdata
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:
1. dietpi-software
2. search - netdata
3. install

#### Expected behaviour:
- It should have been installed or dietpi-software should not find/offer the package if its not available

#### Actual behaviour:
- APT is being called but apparently it cant find netdata for the stretch distro

#### Extra details:
- before the dietpi-update from 6.14 to 6.25 netdata could be installed

#### Additional logs:
```
Log file contents:
E: Package 'netdata' has no installation candidate
```

MichaIng, 2019-08-18:
@Velociraptor85 
Damn you are right. This was available via `stretch-backports` but obviously Debian removed it, why ever... https://packages.debian.org/search?keywords=netdata

The packages have already been removed from the pool server... 
Worst case is that we fallback to the older packages on our server (1.11.1) and hopefully find time to recompile to current version by times.

liepumartins, 2019-08-26:
Meanwhile, how to workaround this?
I am trying to install allo GUI package, which tries to get netdata and fails. 
`dietpi-software install 159`

MichaIng, 2019-08-26:
@liepumartins 
Okay I found the backport packages in Debian snapshots. The latest snapshot that still contains them is 2019-09-07 morning: https://snapshot.debian.org/archive/debian/20190709T042518Z/pool/main/n/netdata/
The were removed the next hours from official repo, for whatever was the reason 🤔.

Okay I packaged the packages together, lets see if this works. Try:
```
cd /tmp
rm -Rf netdata*
wget https://dietpi.com/downloads/binaries/stretch/netdata_armv7l.7z
7zr x netdata_armv7l.7z
rm netdata_armv7l.7z
G_AGUP
dpkg -i netdata-{core,plugins-bash,web,}_*.deb
G_AGF
```
Optionally you can as well install the Python and Node.js plugins, which show a bunch of additional charts for certain software titles:
- The Python plugin shows e.g. charts for Redis and MariaDB databases and some others. Most simply double with what the default bash-based plugins already show. So I would not install it as long as you don't have Python 3 installed anyway.
It requires some package version from backports. The following worked for me:
```
apt install python3-six python3-yaml python3-urllib3/stretch-backports
dpkg -i netdata-plugins-python_1.12.1-2_bpo9+1_all.deb
```
- The Node.js plugin installs fine, but this only makes sense if your have Node.js based software installed, AFAIK:
```
dpkg -i netdata-plugins-nodejs_1.12.1-2_bpo9+1_all.deb
```
#### Cleanup:
```
rm -Rf netdata*
```

liepumartins, 2019-08-27:
@MichaIng thank You for quick response!

However,  I have an armv8 (NanoPi NEO2).

And I can live without netdata, I just want to install the allo GUI. If I could customize what packages it requires, that would be awesome. Since, in this case, I care only about audio (roon,gmrender,shairport,squeezelite) and the ability for user to configure settings via web GUI.

MichaIng, 2019-09-02:
@liepumartins 
> Had to manually install `libipmimonitoring5a`

`G_AGF` did not install this package automatically?

MichaIng, 2019-09-02:
Hmm there is a separate netdata repository: https://packagecloud.io/netdata/netdata/install
Will test this and probably add as default install method.

MichaIng, 2019-09-02:
PR up: https://github.com/MichaIng/DietPi/pull/3091

Velociraptor85, 2019-09-03:
Is there a Fallback or rather how would I go back to the "default" branch?

liepumartins, 2019-09-04:
> @liepumartins
> 
> > Had to manually install `libipmimonitoring5a`
> 
> `G_AGF` did not install this package automatically?

Not 100% sure. But `dpkg -i netdata*.deb` failed without `libipmimonitoring5a` present.

MichaIng, 2019-09-04:
@liepumartins 
Jep the dpkg command is expected to fail, if not yet all dependencies are installed. `G_AGF` == `apt-get -f install` fixes this by pulling all required dependencies automatically. So I hope when running both commands in a row, afterwards you should have the netdata process running, thus ip:19999 accessible from local machine. Note that you need to edit config to allow access from remote browser, check the online docs.

closed 2019-10-15 by MichaIng